riscv-collab/riscv-gnu-toolchain #1312: Failing to build rv32em: "The riscv32 is not supported"

GlaireDaggers, 2023-08-17:
So for reference, I am attempting to build this toolchain on Linux Mint 21.1

I have followed the steps outlined in the readme. I first cloned the repository to my secondary drive, fetched the packages as given for Ubuntu, ran the following configure:

```
./configure --prefix=/opt/riscv --with-arch=rv32em --with-abi=ilp32e
```

And then did:

```
sudo make linux
```

The problem I'm running into is that this is failing in build-glibc-linux-rv32em-ilp32e, printing "checking sysdep dirs" before displaying the error "The riscv32 is not supported". The outputs of the config.log in this folder are as follows:

```
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by GNU C Library configure (see version.h), which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ /media/hazel/DATA[HDD]/Projects/riscv-gnu-toolchain/glibc/configure --host=riscv32-unknown-linux-gnu --prefix=/usr --disable-werror --enable-shared --enable-obsolete-rpc --with-headers=/media/hazel/DATA[HDD]/Projects/riscv-gnu-toolchain/linux-headers/include --disable-multilib --enable-kernel=3.0.0 --libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib

## --------- ##
## Platform. ##
## --------- ##

hostname = hazel-Nitro-AN515-54
uname -m = x86_64
uname -r = 5.15.0-73-generic
uname -s = Linux
uname -v = #80-Ubuntu SMP Mon May 15 15:18:26 UTC 2023

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /media/hazel/DATA[HDD]/Projects/riscv-gnu-toolchain/scripts/wrapper/sed
PATH: /opt/riscv/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /snap/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2250: checking build system type
configure:2264: result: x86_64-pc-linux-gnu
configure:2284: checking host system type
configure:2297: result: riscv32-unknown-linux-gnu
configure:2326: checking for riscv32-unknown-linux-gnu-gcc
configure:2353: result: riscv32-unknown-linux-gnu-gcc 
configure:2622: checking for C compiler version
configure:2631: riscv32-unknown-linux-gnu-gcc  --version >&5
riscv32-unknown-linux-gnu-gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
[... 543 lines cut ...]

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "GNU C Library"
#define PACKAGE_TARNAME "glibc"
#define PACKAGE_VERSION "(see version.h)"
#define PACKAGE_STRING "GNU C Library (see version.h)"
#define PACKAGE_BUGREPORT "https://sourceware.org/bugzilla/"
#define PACKAGE_URL "https://www.gnu.org/software/glibc/"
#define PKGVERSION "(GNU libc) "
#define REPORT_BUGS_TO "<https://www.gnu.org/software/libc/bugs.html>"
#define TIMEOUTFACTOR 1
#define HAVE_TUNABLES 1
#define USE_PPC_SCV 1
#define HAVE_CC_NO_STACK_PROTECTOR 1
#define STACK_PROTECTOR_LEVEL 0
#define HAVE_GNU_RETAIN 1

configure: exit 1
```

I'm really not sure what I'm missing here, is there some step I skipped? Any help would be appreciated.

TommyMurphyTM1234, 2023-08-17:
Anything relevant here?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/617
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/604

GlaireDaggers, 2023-08-17:
I'm seeing reference to the fact that this could happen if I somehow had the wrong glibc version checked out, but I'm not sure how that could be the case if I cloned this repo directly from master. Just in case I tried to update the glibc submodule, which had no effect. I'm currently doing a full `submodule update --init --recursive` as a last ditch effort to see if that helps, but ofc it's going to take a little while.

Possibly a red flag that according to one of those I'm supposed to be seeing some kind of `checking for sysdeps preconfigure fragments...` line, but I see nothing of the sort in the log.

TommyMurphyTM1234, 2023-08-18:
I did this on a WSL2/Ubuntu 22.04.3 installation on which I have done several toolchain builds recently:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv32em --with-abi=ilp32e
```
and I got this config.log without the error that you mention: [config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12374193/config.log)

Can you try with a new/clean Virtual Machine or other installation?

GlaireDaggers, 2023-08-18:
Hmm, strange. I guess I can try deleting it and checking it out from scratch, and then trying the same steps on my Windows box in WSL. If the latter works I guess I'll chalk it up to Linux Mint being weird.

TommyMurphyTM1234, 2023-08-18:
I can try a Linux Mint installation/build but it might be tomorrow morning (Irish time) at this stage...

kito-cheng, 2023-08-18:
Linux toolchain does not support rv32e, only bare metal toolchain support that

TommyMurphyTM1234, 2023-08-18:
> Linux toolchain does not support rv32e, only bare metal toolchain support that

Ok - but the "riscv32 is not supported" issue is arising at configuration and not make/build time.

TommyMurphyTM1234, 2023-08-18:
Ah - I misunderstood this:

> The problem I'm running into is that this is failing in build-glibc-linux-rv32em-ilp32e, printing "checking sysdep dirs" before displaying the error "The riscv32 is not supported". The outputs of the config.log in this folder are as follows:

I assumed that this meant the "top level" `config.log`.
But it seems to a lower level `config.log` for glibc?
In that case @kito-cheng's comment is apposite.
The Linux toolchain does not suport rv32e and also requires the A (Atomic) extension.

GlaireDaggers, 2023-08-18:
oooh, I see! I think I misunderstood the instructions, I do believe what I actually want is a baremetal toolchain. Sorry for the misunderstanding!

TommyMurphyTM1234, 2023-08-18:
I tried a bare-metal toolchain build and it succeeded.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1312
cd issue-1312
./configure --prefix=`pwd`/installed-tools --with-arch=rv32em --with-abi=ilp32e
make 2>&1 | tee build.log
```

closed 2023-08-18 by TommyMurphyTM1234