riscv-collab/riscv-gnu-toolchain #1313: Can't build toolchain on Arm Linux machine

howjmay, 2023-08-18:
I am installing it on a ARM machine which runs Linux. And the following error returned after I built it by `make`
command.

```
/usr/bin/ld: ../libbacktrace/.libs/libbacktrace.a(elf.o): in function `elf_add_syminfo_data':
/home/howjmay/risc-v/riscv-gnu-toolchain/build-gcc-newlib-stage1/libbacktrace/../.././gcc/libbacktrace/elf.c:746: undefined reference to `__aarch64_cas8_sync'
/usr/bin/ld: ../libbacktrace/.libs/libbacktrace.a(elf.o): in function `backtrace_initialize':
/home/howjmay/risc-v/riscv-gnu-toolchain/build-gcc-newlib-stage1/libbacktrace/../.././gcc/libbacktrace/elf.c:4906: undefined reference to `__aarch64_cas8_sync'
/usr/bin/ld: ../libbacktrace/.libs/libbacktrace.a(dwarf.o): in function `backtrace_dwarf_add':
/home/howjmay/risc-v/riscv-gnu-toolchain/build-gcc-newlib-stage1/libbacktrace/../.././gcc/libbacktrace/dwarf.c:4060: undefined reference to `__aarch64_cas8_sync'
```

TommyMurphyTM1234, 2023-08-18:
You mean that you are trying to build the `riscv-gnu-toolchain` on the Arm target running Linux itself?
What are your `clone`, `configure` and `make` commands and your build log (please compress and attach it rather than copying/pasting it directly)?

If you need a prebuilt RISC-V bare-metal/newlib toolchain that runs on Arm/Linux then you could have a look at the @ilg-ul's  xPack project:

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
  * https://xpack.github.io/dev-tools/riscv-none-elf-gcc/releases/

howjmay, 2023-08-19:
Hi @TommyMurphyTM1234 here is the log I ran `make`
https://gist.github.com/howjmay/58946243a0ef32fc5a6e715b19317761

And the xpack project you provided is really useful. If I still can't build the toolchain successfully, then I will directly use the xpack project. Thanks

TommyMurphyTM1234, 2023-08-19:
While this relates to Gentoo, maybe it's a more general issue and relevant here?
I can't tell from your log what Arm Linux distro or GCC version that you're using.

* https://bugs.gentoo.org/868018

howjmay, 2023-08-19:
The gcc version is 
gcc (Ubuntu 10.5.0-1ubuntu1~20.04) 10.5.0

TommyMurphyTM1234, 2023-08-19:
I presume that you can get a suitable toolchain from here?

* https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads
* https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads#:~:text=AArch64%20Linux%20hosted%20cross%20toolchains

howjmay, 2023-08-19:
Let me try thank you!!

TommyMurphyTM1234, 2023-08-21:
Any update @howjmay?

ilg-ul, 2023-08-22:
> I am still figuring out how to use qemu to run the compiled riscv binary

You can use the [Hello World QEMU](https://github.com/micro-os-plus/hello-world-qemu-template-xpack/) template to generate functional RISC-V projects, and take a look at how it is done there.

TommyMurphyTM1234, 2023-08-22:
> Hi @TommyMurphyTM1234 I think as you said it is the fault of gcc, and I am using xPack now. 

OK - I'll close  this issue so. Thanks for the update.