riscv-collab/riscv-gnu-toolchain #1316: Does binutils support zvfh/zvfhmin extensions?

lhtin, 2023-08-24:
It appears that binutils does not yet support the zvfh/zvfhmin extension since GCC already supports zvfh/zvfhmin already. Is there any plan about this? Thanks.

TommyMurphyTM1234, 2023-08-24:
> It appears that binutils does not yet support the zvfh/zvfhmin extension

Can you clarify how exactly you are concluding this please?
A self-contained reproducible test case and instructions on how you are invoking the tools would be ideal.

> since GCC already supports zvfh/zvfhmin already

Ditto.

TommyMurphyTM1234, 2023-08-24:
Oh - was this you adding (enhancing?) support for those extensions to GCC?

* https://gcc.gnu.org/pipermail/gcc-patches/2023-August/628236.html
* https://gcc.gnu.org/pipermail/gcc-patches/2023-August/628235.html

Are you asking if somebody else is going to add the complementary support for these to binutils?
If so then you probably need to ask over on the binutils mailing lists:

* https://www.gnu.org/software/binutils/#:~:text=org/bugzilla/.-,Mailing%20lists,-There%20are%20three

lhtin, 2023-08-25:
> Are you asking if somebody else is going to add the complementary support for these to binutils?
> If so then you probably need to ask over on the binutils mailing lists:

Yes, thanks anyway.

TommyMurphyTM1234, 2023-08-25:
Just to clarify, the majority of maintenance and enhancement of the various `riscv-gnu-tools` components is now done in the relevant upstream projects (e.g. GCC, Binutils, Newlib, GLIBC etc.). This repo primarily deals with the integration of those tools into usable toolchain bundles. The main exception where development continues here is the prototyping/development of draft extension support in the `riscv-gcc` repo and here because GCC has strict rules about only accepting ratified/stable extension support upstream.

TommyMurphyTM1234, 2023-08-25:
Is this of any relevance?

* https://marc.info/?l=binutils-cvs&m=169104906413151&w=2
* https://marc.info/?l=binutils-cvs&m=169208189614247&w=2
*

lhtin, 2023-08-25:
> Is this of any relevance?
> 
> * https://marc.info/?l=binutils-cvs&m=169104906413151&w=2
> * https://marc.info/?l=binutils-cvs&m=169208189614247&w=2

Yes, that's it. Thank you so much.

TommyMurphyTM1234, 2023-08-25:
I see the change here

*https://sourceware.org/git/?p=binutils-gdb.git;a=blob;f=bfd/elfxx-riscv.c;h=cb65024beaffbe60b51e303c058f2bb573d72f38;hb=HEAD#l1113

But it doesn't seem to appear in the GitHub mirror repo at the moment even though the mirror is supposed to sync daily.

* https://github.com/bminor/binutils-gdb/blob/a89e364b45a93acd20f48abd787ef5cb7c07f683/bfd/elfxx-riscv.c#L1113

lhtin, 2023-08-25:
> I see the change here
> 
> * https://sourceware.org/git/?p=binutils-gdb.git;a=blob;f=bfd/elfxx-riscv.c;h=cb65024beaffbe60b51e303c058f2bb573d72f38;hb=HEAD#l1113
> 
> But it doesn't seem to appear in the GitHub mirror repo at the moment even though the mirror is supposed to sync daily.
> 
> * https://github.com/bminor/binutils-gdb/blob/a89e364b45a93acd20f48abd787ef5cb7c07f683/bfd/elfxx-riscv.c#L1113
> 
> As such, you'll probably need to build the latest from the master repo rather than the mirror to ensure that you get the very latest changes.

Yes, I just didn't search for the name of the zvfh extension on top of Github Mirror, so I assumed it wasn't supported.

TommyMurphyTM1234, 2023-08-25:
Is it ok to close this issue so?

lhtin, 2023-08-25:
Totally fine, thanks again.

closed 2023-08-25 by lhtin