riscv-collab/riscv-gnu-toolchain #1317: Internal compiler error with zbs and constant 2048/0x800

juliusbaxter, 2023-08-30:
I'm playing with the bitmanip extensions and noticed the codebase I'm compiling triggering internal compiler errors always with a particular constant: 2048 or 0x800.

I'm able to recreate it with the following code in a file called `rvzbs.c`:

```
unsigned int function(unsigned int arg1)
{
  unsigned int rv = 0;
  if (arg1 & 0x800) {
    rv = 0x800;
  } else {
    rv = arg1 - 0x800;
  }
  return rv;
}
```

Compiled like so:

`riscv64-unknown-elf-gcc -c -O2 -march=rv32imac_zbs -mabi=ilp32 -o rvzbs.o  rvzbs.c`

Gives the following error:

```
rvzbs.c: In function 'function':
rvzbs.c:10:1: error: unrecognizable insn:
   10 | }
      | ^
(insn 7 3 8 2 (set (reg:SI 75)
        (const_int 2048 [0x800])) "rvzbs.c":4:12 -1
     (expr_list:REG_EQUAL (const_int 2048 [0x800])
        (nil)))
during RTL pass: vregs
rvzbs.c:10:1: internal compiler error: in extract_insn, at recog.cc:2791
Please submit a full bug report, with preprocessed source (by using -freport-bug).
See <https://gcc.gnu.org/bugs/> for instructions.
```

Changing each 0x800 to something else will result in successful compilation.

My GCC is on an OS X Ventura machine, installed via brew:

```
riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (g2ee5e430018-dirty) 12.2.0
```

juliusbaxter, 2023-08-30:
Oops, this has already been reported several times and looks to be getting fixed in GCC 13?

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=106532

closed 2023-08-30 by juliusbaxter

kito-cheng, 2023-08-31:
Seems like I should spend some time to bump GCC to 13.