riscv-collab/riscv-gnu-toolchain #1318: [gdb] Printing Rust enum values on function return sometimes crashes gdb

jeffy1009, 2023-08-31:
## Environment

rustc 1.73.0-nightly (08d00b40a 2023-08-09), stable-x86_64-unknown-linux-gnu
qemu-riscv64 version 8.1.0 (v8.1.0)
GNU gdb (GDB) 12.1

## Minimal reproducible example:

`.cargo/config`:
```toml
[target.riscv64gc-unknown-linux-gnu]
linker = "riscv64-unknown-linux-gnu-gcc"

[build]
target = "riscv64gc-unknown-linux-gnu"
```

`main.rs`:
```rust
fn main() {
    let num = (1..2).next().unwrap();
    println!("{}", num);
}

QEMU
```console
qemu-riscv64 -L ~riscv-gnu-toolchain/install/sysroot -g 1234target/riscv64gc-unknown-linux-gnu/debug/temp
```

GDB
```
riscv64-unknown-linux-gnu-gdb target/riscv64gc-unknown-linux-gnu/debug/temp
(gdb) target remote:1234
(gdb) b main
(gdb) c
(gdb) c
Breakpoint 1, temp::main () at src/main.rs:41
41          let num = (1..2).next().unwrap();
(gdb) s
core::iter::range::{impl#3}::next<i32> (self=0x2aaaab2aae60) at /rustc/08d00b40aef2017fe6dba3ff7d6476efa0c10888/library/core/src/iter/range.rs:712
712     /rustc/08d00b40aef2017fe6dba3ff7d6476efa0c10888/library/core/src/iter/range.rs: No such file or directory.
(gdb) fin
```

backtrace
```
Value returned is $1 = /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/gdbtypes.h:1064: internal-error: field: Assertion `idx >= 0 && idx < num_fields ()' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
----- Backtrace -----
0x56125cab3556 gdb_internal_backtrace_1
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/bt-utils.c:122
0x56125cab3556 _Z22gdb_internal_backtracev
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/bt-utils.c:168
0x56125cdc50e4 internal_vproblem
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/utils.c:394
0x56125cdc5310 _Z15internal_verrorPKciS0_P13__va_list_tag
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/utils.c:471
0x56125cf087b4 _Z14internal_errorPKciS0_z
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdbsupport/errors.cc:55
0x56125cdd7ecd _ZNK4type5fieldEi
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/gdbtypes.h:1064
0x56125cddd946 _ZNK4type5fieldEi
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/value.c:3129
0x56125cddd946 _Z21value_primitive_fieldP5valueliP4type
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/value.c:3037
0x56125ccfa6f9 _ZNK13rust_language10print_enumEP5valueP7ui_fileiPK19value_print_options
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/rust-lang.c:469
0x56125ccfab43 _ZNK13rust_language17value_print_innerEP5valueP7ui_fileiPK19value_print_options
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/rust-lang.c:590
0x56125cdd3aba _Z16common_val_printP5valueP7ui_fileiPK19value_print_optionsPK13language_defn
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/valprint.c:1065
0x56125cad7eda _Z13c_value_printP5valueP7ui_filePK19value_print_options
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/c-valprint.c:588
0x56125cdd3c52 _Z11value_printP5valueP7ui_filePK19value_print_options
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/valprint.c:1182
0x56125cbf79c7 print_return_value_1
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/infcmd.c:1533
0x56125cbf79c7 _Z18print_return_valueP6ui_outP17return_value_info
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/infcmd.c:1567
0x56125cc063c8 _Z16print_stop_eventP6ui_outb
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/infrun.c:8343
0x56125cda9178 tui_on_normal_stop
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/tui/tui-interp.c:102
0x56125cc083a2 _ZNKSt8functionIFvP6bpstatiEEclES1_i
        /usr/include/c++/11/bits/std_function.h:590
0x56125cc083a2 _ZNK3gdb9observers10observableIJP6bpstatiEE6notifyES3_i
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/../gdbsupport/observable.h:150
0x56125cc083a2 _Z11normal_stopv
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/infrun.c:8593
0x56125cc14f02 _Z20fetch_inferior_eventv
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/infrun.c:4117
0x56125cd0686d run_async_handler_and_reschedule
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/ser-base.c:138
0x56125cf08b85 gdb_wait_for_event
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdbsupport/event-loop.cc:700
0x56125cf0932a gdb_wait_for_event
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdbsupport/event-loop.cc:596
0x56125cf0932a _Z16gdb_do_one_eventv
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdbsupport/event-loop.cc:237
0x56125cc49014 start_event_loop
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/main.c:421
0x56125cc49014 captured_command_loop
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/main.c:481
0x56125cc4ab14 captured_main
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/main.c:1351
0x56125cc4ab14 _Z8gdb_mainP18captured_main_args
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/main.c:1366
0x56125ca19baf main
        /home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/gdb.c:32
---------------------
/home/xxx/riscv-gnu-toolchain/build/../gdb/gdb/gdbtypes.h:1064: internal-error: field: Assertion `idx >= 0 && idx < num_fields ()' failed.

```

jeffy1009, 2023-08-31:
Opps.. it happens on x86 too, so I'll look for the proper place for this issue..

TommyMurphyTM1234, 2023-08-31:
You should log the issue upstream:

* https://www.sourceware.org/gdb/bugs/

closed 2023-08-31 by TommyMurphyTM1234

jeffy1009, 2023-08-31:
Seems like the issue is gone in the latest binutils-gdb. https://github.com/bminor/binutils-gdb
I'll see when exactly the issue was fixed.

jeffy1009, 2023-08-31:
The issue was fixed in https://github.com/bminor/binutils-gdb/commit/debd0556e519c3d258299cf5f14a44cc01c795da.
The patch is really simple so those who have the same issue can change that one line and you'd be good to go.
(The related API's been changed in https://github.com/bminor/binutils-gdb/commit/debd0556e519c3d258299cf5f14a44cc01c795da also, so the patch doesn't apply directly though)

jeffy1009, 2023-08-31:
@TommyMurphyTM1234 If you don't mind, I actually have one question.
It seems that I can just use the upstream gdb using `set arch riscv:rv64`
Does gdb in riscv-gnu-toolchain exist for any other purposes?

TommyMurphyTM1234, 2023-08-31:
> @TommyMurphyTM1234 If you don't mind, I actually have one question. It seems that I can just use the upstream gdb using `set arch riscv:rv64` Does gdb in riscv-gnu-toolchain exist for any other purposes?

Sorry, I don't understand your question.

jeffy1009, 2023-08-31:
Oh `gdb` is just a link to `binutils-gdb`. That was dumb :( 
I thought it was something different, with updates pulled from binutils-gdb.

TommyMurphyTM1234, 2023-08-31:
> Oh `gdb` is just a link to `binutils-gdb`. That was dumb :( I thought it was something different, with updates pulled from binutils-gdb.

Sorry, I still don't really understand your point.
Do you have a question?

jeffy1009, 2023-08-31:
> Do you have a question?

Nope, I just figured out that I asked a dump question.
Thanks for your reply :)