riscv-collab/riscv-gnu-toolchain #1321: make[1] : *** [stage1-bubble'] Error 2

LGRwx, 2023-09-04:
I got this error when I try to upgrade my virtual machine's gcc to version 10.4.0, I have succeed to upgrade the gcc, but the first time I didn't got this error , my machine's OS is  neokylin 7 , a system's distribution based on RHEL, CPU's arch is aarch 64, the error information as list:
make[5]: Leaving directory `/usr/local/src/gcc-10.4.0/build/isl'
Making all in doc
make[5]: Entering directory `/usr/local/src/gcc-10.4.0/build/isl/doc'
make[5]: Nothing to be done for `all'.
make[5]: Leaving directory `/usr/local/src/gcc-10.4.0/build/isl/doc'
make[4]: Leaving directory `/usr/local/src/gcc-10.4.0/build/isl'
make[3]: Leaving directory `/usr/local/src/gcc-10.4.0/build/isl'
make[2]: Leaving directory `/usr/local/src/gcc-10.4.0/build'
make[1]: *** [stage1-bubble] Error 2
make[1]: Leaving directory `/usr/local/src/gcc-10.4.0/build'
make: *** [all] Error 2
the comman I have used to configure is: ../configure --prefix=/usr/local/gcc-10.4.0/ --mandir=/usr/share/man --infodir=/usr/share/info --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-linker-hash-style=gnu --enable-languages=c,c++,lto --enable-plugin --enable-initfini-array --disable-libgcj --enable-gnu-indirect-function --with-tune=generic --with-arch=armv8-a --disable-multilib
I run the ../configure ...... and generate the makefile, then I run  make -j`nproc`  and I got this error, hoping to get your advice and your experience about build gcc. thank you!

TommyMurphyTM1234, 2023-09-04:
> I got this error when I try to upgrade my virtual machine's gcc to version 10.4.0, 

What exactly do you mean by "upgrading" here?
What are the exact steps taken to arrive at this situation?
For example the exact `git clone`, `configure` and `make` commands?
What is the full build log (for a non-parallel build - see below)?

> I have succeed to upgrade the gcc, but the first time I didn't got this error 

This is confusing - so there is a problem or there is not?

> I run the ../configure ...... and generate the makefile, then I run make -j`nproc` and I got this error, hoping to get your advice and your experience about build gcc. thank you!

Did you try a non-parallel build - with no `-j` option?

TommyMurphyTM1234, 2023-09-04:
Also, are you sure that you're doing a clean build by doing `make distclean` first and maybe also manually deleting `/usr/local/gcc-10.4.0/` etc.?

LGRwx, 2023-09-04:
> What exactly do you mean by "upgrading" here? What are the exact steps taken to arrive at this situation? For example the exact `git clone`, `configure` and `make` commands? What is the full build [...]
I have upgraded the gcc from version 4.8.5 to version 10.4.0 on another machine, I didn't try to build it without the '-j', I asked  an experienced engineer, he execute some commands like you said ,'make clean' and without '-j' just 'make', then the log report   some other information 'configure: error: C++ preprocessor "/lib/cpp" fails sanity check' then we install some dependencies and due to the system and arch are not recognized by the script，and now it is making status, it seems nice ,thank you! If there are any error happened ,I will continue to comment at this issue. thank you very much! But it is still exists some problem, today I delete the gcc-10.4.0 directory both generate by the build and the source code directory, I unzip the *.tar.gz and rebuild it ,but it still just report like the above information, if it could report    'configure: error: C++ preprocessor "/lib/cpp" fails sanity check ' I could solve it quickly.....last, thank you for your quick reply !

TommyMurphyTM1234, 2023-09-04:
Sorry, your post is confusing and difficult to read/understand.
Are you still having a problem or not?
If you are then please provide all information needed to reproduce it and a full configure/build log file (compressed and attached - not copy and pasted into a post).

LGRwx, 2023-09-04:
> Sorry, your post is confusing and difficult to read/understand. Are you still having a problem or not? If you are then please provide all information needed to reproduce it and a full configure/build log file (compressed and attached - not copy and pasted into a post). If your issue has been resolved then please close the issue. If you have issues with the GCC or related build scripts then please log an issue upstream in the relevant project.
en, the building just finish a minute ago , it's healthy , thank you for spending time to give me your suggestions!

closed 2023-09-04 by LGRwx

TommyMurphyTM1234, 2023-09-04:
> if it could report 'configure: error: C++ preprocessor "/lib/cpp" fails sanity check ' I could solve it quickly..

But it does, doesn't it - **if** you use a non-parallel build?

> I asked an experienced engineer, he execute some commands like you said ,'make clean' and without '-j' just 'make', then the log report some other information 'configure: error: C++ preprocessor "/lib/cpp" fails sanity check' then we install some dependencies 

It's always good practice if a parallel build fails to do the equivalent non-parallel build because parallel builds interleave output from the different jobs making the timing of and reason for errors more difficult to spot.