riscv-collab/riscv-gnu-toolchain #1322: Does GCC compiler for RISCV support asm goto?

jerryhethatday, 2023-09-04:
I am following the [Extended Asm](https://gcc.gnu.org/onlinedocs/gcc/Extended-Asm.html) to create a code sequence like below:

```

int main(int argc, char *argv[])
{
    int i;
    printf("Hello world\n");

    int qq = 0;
    asm goto(".align 2 \n\taddi %0,%0, 1 \n\t.align 1 \n\taddi	%0,%0,1552 \n\t.align 1 \n\taddi %0,%0, 1553 \n\t"
            :"+r"(qq)
            :
            :
            :carry
            );
    carry:
    return 0;
}

```
But when I compile the code with the following option: 
```
riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -g -O2  -pedantic -Wall -specs=../util/htif.specs  -L ./rv32imac/ilp32/ -T ../util/htif.ld -Wl,-Map=hello.riscv.map -o hello.riscv ../tests/hello.c

```

The compile complains the following:
[image]

I checked the format many times but still can not make it right. The` asm volatile()` work as expected, but the` asm goto` just 
keeps giving wired error depicted above.
I wonder if there is a version requirement for asm goto to work? I tried gcc version 9.2 and 10.2, none of them work. 
Any help will be appriciated, thanks.

TommyMurphyTM1234, 2023-09-04:
Can you post the error messages in text form please?
The image is impossible to read (for me at least).

TommyMurphyTM1234, 2023-09-04:
I just tried the latest CI build of the tools here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases

namely:

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2023.07.07

with this source file:
```
// test.c

#include <stdio.h>

int main(int argc, char *argv[])
{
    // int i;
    printf("Hello world\n");

    int tt = 1;
    asm goto(".align 2 \n\t"
             "addi %0,%0, 1 \n\t"
             ".align 1 \n\t"
             "j %l[carry] \n\t"
             :"+r"(tt)
             :
             :
             :carry);
    carry:
    return 0;
}
```
and it compiled fine:
```
./riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc () 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

./riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -g -O2  -pedantic -Wall -c test.c

./riscv64-unknown-elf-objdump -dS test.o 

test.o:     file format elf32-littleriscv

Disassembly of section .text.startup:

00000000 <main>:
#include <stdio.h>

int main(int argc, char *argv[])
{
    // int i;
    printf("Hello world\n");
   0:	00000537          	lui	a0,0x0

00000004 <.LVL1>:
{
   4:	1141                	add	sp,sp,-16
    printf("Hello world\n");
   6:	00050513          	mv	a0,a0
{
   a:	c606                	sw	ra,12(sp)
    printf("Hello world\n");
   c:	00000097          	auipc	ra,0x0
  10:	000080e7          	jalr	ra # c <.LVL1+0x8>

00000014 <.LVL2>:
             :
             :
             :carry);
    carry:
    return 0;
}
  14:	40b2                	lw	ra,12(sp)
  16:	4501                	li	a0,0
  18:	0141                	add	sp,sp,16
  1a:	8082                	ret
```

jerryhethatday, 2023-09-04:
> I just tried the latest CI build of the tools here:
> 
> * https://github.com/riscv-collab/riscv-gnu-toolchain/releases
> 
> namely:
> 
> * https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2023.07.07
> 
> with this source file (unused variable definition commented out to avoid compilation error with `-Wall`):
> 
> ```
> // test.c
> 
> #include <stdio.h>
> 
> int main(int argc, char *argv[])
> {
>     // int i;
>     printf("Hello world\n");
> 
>     int tt = 1;
>     asm goto(".align 2 \n\t"
>              "addi %0,%0, 1 \n\t"
>              ".align 1 \n\t"
>              "j %l[carry] \n\t"
>              :"+r"(tt)
>              :
>              :
>              :carry);
>     carry:
>     return 0;
> }
> ```
> 
> and it compiled fine:
> 
> ```
> ./riscv64-unknown-elf-gcc --version
> riscv64-unknown-elf-gcc () 12.2.0
> Copyright (C) 2022 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> ./riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -g -O2  -pedantic -Wall -c test.c
> 
> ./riscv64-unknown-elf-objdump -dS test.o 
> 
> test.o:     file format elf32-littleriscv
> 
> Disassembly of section .text.startup:
> 
> 00000000 <main>:
> #include <stdio.h>
> 
> int main(int argc, char *argv[])
> {
>     // int i;
>     printf("Hello world\n");
>    0:	00000537          	lui	a0,0x0
> 
> 00000004 <.LVL1>:
> {
>    4:	1141                	add	sp,sp,-16
>     printf("Hello world\n");
>    6:	00050513          	mv	a0,a0
> {
>    a:	c606                	sw	ra,12(sp)
>     printf("Hello world\n");
>    c:	00000097          	auipc	ra,0x0
>   10:	000080e7          	jalr	ra # c <.LVL1+0x8>
> 
> 00000014 <.LVL2>:
>              :
>              :
>              :carry);
>     carry:
>     return 0;
> }
>   14:	40b2                	lw	ra,12(sp)
>   16:	4501                	li	a0,0
>   18:	0141                	add	sp,sp,16
>   1a:	8082                	ret
> ```

Thanks for your help, then it seems there is a min version requirement for this `asm goto` :)

TommyMurphyTM1234, 2023-09-04:
I tried the latest xPack RISC-V GCC 10, 11, 12 and 13 builds from here:

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/releases/

and got this:

* GCC 10:
```
/riscv-none-embed-gcc --version
riscv-none-embed-gcc (xPack GNU RISC-V Embedded GCC x86_64) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

./riscv-none-embed-gcc -march=rv32imac -mabi=ilp32 -g -O2  -Wall -c ~/Downloads/test.c
/home/user/Downloads/test.c: In function 'main':
/home/user/Downloads/test.c:15:15: error: expected ':' before string constant
   15 |              :"+r"(tt)
      |               ^~~~
      |               :
/home/user/Downloads/test.c:19:5: warning: label 'carry' defined but not used [-Wunused-label]
   19 |     carry:
      |     ^~~~~
/home/user/Downloads/test.c:10:9: warning: unused variable 'tt' [-Wunused-variable]
   10 |     int tt = 1;
      |         ^~
```
* GCC 11:
```
./riscv-none-elf-gcc --version
riscv-none-elf-gcc (xPack GNU RISC-V Embedded GCC x86_64) 11.3.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

./riscv-none-elf-gcc -march=rv32imac -mabi=ilp32 -g -O2  -Wall -c ~/Downloads/test.c
```
* GCC 12:
```
./riscv-none-elf-gcc --version
riscv-none-elf-gcc (xPack GNU RISC-V Embedded GCC x86_64) 12.3.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

./riscv-none-elf-gcc -march=rv32imac -mabi=ilp32 -g -O2  -Wall -c ~/Downloads/test.c
```
* GCC 13:
```
./riscv-none-elf-gcc --version
riscv-none-elf-gcc (xPack GNU RISC-V Embedded GCC x86_64) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

./riscv-none-elf-gcc -march=rv32imac -mabi=ilp32 -g -O2  -Wall -c ~/Downloads/test.c
```
So it looks like the issue that you were hitting was probably fixed in GCC 11.

closed 2023-09-04 by TommyMurphyTM1234