riscv-collab/riscv-gnu-toolchain #1323: Error building RISC-V toolchain for RV32i only

TJ1999, 2023-09-04:
Hello everyone,

I tried to build the riscv toolchain for rv32i ISA. Unfortunately, some errors occurred.
I configured the project with the following command:
`./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32`
But I get following error message:
```
...
echo timestamp > stmp-int-hdrs
/home/tjung/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/xgcc -B/home/tjung/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/  -xc -nostdinc /dev/null -S -o /dev/null -fself-test=../.././gcc/gcc/testsuite/selftests
/home/tjung/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/xgcc -B/home/tjung/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/  -xc++ -nostdinc /dev/null -S -o /dev/null -fself-test=../.././gcc/gcc/testsuite/selftests
cc1: error: '-mdiv' requires '-march' to subsume the 'M' extension
cc1plus: error: '-mdiv' requires '-march' to subsume the 'M' extension
...
```
It seems like hardware instructions for divisions are used whereas I configured the project with rv32i spec and ilp32 abi

TommyMurphyTM1234, 2023-09-04:
Please provide full `git clone`, `configure` and `make` commands need to reproduce your issue.

TJ1999, 2023-09-05:
Thanks for your response,
after re-cloning the repository the toolchain build without any errors. 
I don't know what went wrong. Seems like I checked out the repo in a wrong way. I can remember that I had issues with that. As far as I remember, the submodule checkout did not work properly. But it seems like the build script is doing that for me.
Thanks for providing your commands.

closed 2023-09-05 by TJ1999