riscv-collab/riscv-gnu-toolchain #1324: How to make the name of the generated `clang` consistent with `riscv64-unknown-linux-gnu-gcc`?

sunshaoce, 2023-09-05:
When `--enable-llvm` is enabled, how to make the name of the generated `clang` consistent with `riscv64-unknown-linux-gnu-gcc` and become `riscv64-unknown-linux-gnu-clang`? 

Is there an option to control this?

TommyMurphyTM1234, 2023-09-05:
I don't understand.
The CI builds here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases

already use the following prefixes:
```
riscv64-unknown-linux-gnu-gcc
riscv64-unknown-linux-gnu-g++
riscv64-unknown-linux-gnu-clang
riscv64-unknown-linux-gnu-clang++
etc.
```
Specifically:
```
cd ~/Downloads
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2023.07.07/riscv64-glibc-ubuntu-20.04-llvm-nightly-2023.07.07-nightly.tar.gz
tar xvf riscv64-glibc-ubuntu-20.04-llvm-nightly-2023.07.07-nightly.tar.gz
cd riscv/bin/
ls
amdgpu-arch             clang-rename      llvm-dwp       llvm-symbolizer                      riscv64-unknown-linux-gnu-gcc            riscv64-unknown-linux-gnu-lto-dump
analyze-build           clang-repl        llvm-lib       nvptx-arch                           riscv64-unknown-linux-gnu-gcc-12.2.0     riscv64-unknown-linux-gnu-nm
clang                   clang-scan-deps   llvm-ml        qemu-riscv32                         riscv64-unknown-linux-gnu-gcc-ar         riscv64-unknown-linux-gnu-objcopy
clang++                 diagtool          llvm-nm        qemu-riscv64                         riscv64-unknown-linux-gnu-gcc-nm         riscv64-unknown-linux-gnu-objdump
clang-16                git-clang-format  llvm-objcopy   riscv64-unknown-linux-gnu-addr2line  riscv64-unknown-linux-gnu-gcc-ranlib     riscv64-unknown-linux-gnu-ranlib
clang-check             hmaptool          llvm-objdump   riscv64-unknown-linux-gnu-ar         riscv64-unknown-linux-gnu-gcov           riscv64-unknown-linux-gnu-readelf
clang-cl                intercept-build   llvm-pdbutil   riscv64-unknown-linux-gnu-as         riscv64-unknown-linux-gnu-gcov-dump      riscv64-unknown-linux-gnu-run
clang-cpp               ld.lld            llvm-profdata  riscv64-unknown-linux-gnu-c++        riscv64-unknown-linux-gnu-gcov-tool      riscv64-unknown-linux-gnu-size
clang-extdef-mapping    ld64.lld          llvm-ranlib    riscv64-unknown-linux-gnu-c++filt    riscv64-unknown-linux-gnu-gdb            riscv64-unknown-linux-gnu-strings
clang-format            lld               llvm-rc        riscv64-unknown-linux-gnu-clang      riscv64-unknown-linux-gnu-gdb-add-index  riscv64-unknown-linux-gnu-strip
clang-linker-wrapper    lld-link          llvm-readobj   riscv64-unknown-linux-gnu-clang++    riscv64-unknown-linux-gnu-gfortran       scan-build
clang-offload-bundler   llvm-ar           llvm-size      riscv64-unknown-linux-gnu-cpp        riscv64-unknown-linux-gnu-gprof          scan-build-py
clang-offload-packager  llvm-cov          llvm-strings   riscv64-unknown-linux-gnu-elfedit    riscv64-unknown-linux-gnu-ld             scan-view
clang-refactor          llvm-cxxfilt      llvm-strip     riscv64-unknown-linux-gnu-g++        riscv64-unknown-linux-gnu-ld.bfd         wasm-ld
```
What exactly are you trying to change here?

TommyMurphyTM1234, 2023-09-05:
Any update @sunshaoce?

sunshaoce, 2023-09-06:
I didn't notice that besides `clang` there is also `riscv64-unknown-linux-gnu-clang`. Thanks for your reply!

sunshaoce, 2023-09-06:
One more question, is there a way to change `clang` to `riscv64-unknown-linux-gnu-clang` using `mv` instead of `ln -s`?

TommyMurphyTM1234, 2023-09-06:
> One more question, is there a way to change `clang` to `riscv64-unknown-linux-gnu-clang` using `mv` instead of `ln -s`?

You could try changing these lines in the `riscv-gnu-toolchain` `Makefile.in`:

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/da18f1cc2411099737df8845d3ec8945d94cc08f/Makefile.in#L942
* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/da18f1cc2411099737df8845d3ec8945d94cc08f/Makefile.in#L959

but I don't know if anything in the toolchain may depend on the existence of the RISC-V `clang/clang++` such that, if they don't exist then things may go wrong? Probably not but you would need to check.

If the issue is to avoid picking up the RISC-V `clang/clang++` when using the native `clang/clang++` then why not just put the RISC-V toolchain directory at the end of the regular `$PATH` and always use the `riscv64-unknown-linux-gnu-` prefix when calling the RISC-V tools?