riscv-collab/riscv-gnu-toolchain #1325: Question about the use of intrinsic versions in rvv-next branch

rvvstuding, 2023-09-06:
Hi guys, I manually compiled the `rvv-next` branch of this project a few months ago. Recently I found that the intrinsic used in this branch is different from the latest ones in `riscv-non-isa/rvv-intrinsic-doc`. I would like to know what version of intrinsic used in the `rvv-next` branch, and  what is different from the v0.11 version of intrinsic used by gcc13.

TommyMurphyTM1234, 2023-09-06:
See here: 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1308

The `rvv-next` branch is stale and redundant now that Intrinsic support is integrated and maintained upstream.

Until `riscv-gnu-toolchain` is bumped to GCC 13:

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1239

you should build the toolchain using upstream GCC 13 sources.
For example:
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain issue-1325
cd issue-1325
git clone https://github.com/gcc-mirror/gcc -b releases/gcc-13 gcc-13
./configure --prefix=... --with-arch=rv64gcv --with-abi=lp64d --with-gcc-src=`pwd`/gcc-13
make
```

TommyMurphyTM1234, 2023-09-07:
Did that address your question @rvvstuding?

rvvstuding, 2023-09-07:
> Did that address your question @rvvstuding?

Perhaps not, I know the `rvv-next` branch may be outdated, but what I want to know is what version of intrinsic is implemented in `rvv-next`, I tried to use the `__riscv_v_intrinsic` macro to get its version, but it seems that this macro is not defined in `rvv-next` branch.

TommyMurphyTM1234, 2023-09-07:
You may need to go through the `riscv-gnu-toolchain` and `riscv-gcc` commits and PRs to figure out what non-ratified version of the RVV intrinsics is implemented. But there's really no reason to use this outdated stuff instead of the more up to date upstream implementation.

TommyMurphyTM1234, 2023-09-07:
Maybe @kito-cheng might know the answer to your question?

rvvstuding, 2023-09-07:
> You may need to go through the `riscv-gnu-toolchain` and `riscv-gcc` commits and PRs to figure out what non-ratified version of the RVV intrinsics is implemented. But there's really no reason to use this outdated stuff instead of the more up to date upstream implementation.

Thanks for your answer.
 This is just a small question that puzzles me. I'll go with the more up to date upstream implementation.

cmuellner, 2023-10-05:
> Perhaps not, I know the `rvv-next` branch may be outdated, but what I want to know is what version of intrinsic is implemented in `rvv-next`, I tried to use the `__riscv_v_intrinsic` macro to get its version, but it seems that this macro is not defined in `rvv-next` branch.

The efforts to standardize the vector intrinsics have formed after the rvv-next branch has been created.
My assumption is that they have been mainly inspired by what was implemented in LLVM back then.
However, one should not use these intrinsics, as they are not supported by any compiler.

closed 2023-10-05 by TommyMurphyTM1234