riscv-collab/riscv-gnu-toolchain #1326: I am trying to adding packed SIMD instruction to riscv-gnu-toolchain I am getting this error

manasabatchu, 2023-09-06:
I am trying to adding packed SIMD instruction to riscv-gnu-toolchain I am getting this error 
 add8.c: Assembler messages:
add8.c:8: Error: unrecognized opcode `add8 a5,a5,a4'
Binutils modification
Now, binutils need to be aware of the new instruction. riscv-gnu-toolchain/riscv-binutils/include/opcode/riscv-opc.h is updated as follows:

/* Instruction opcode macros.  */
+ #define MATCH_ADD8 0x48000077
+ #define MASK_ADD8  0xfe00707f

#endif /* RISCV_ENCODING_H */
#ifdef DECLARE_INSN
+ DECLARE_INSN(add8, MATCH_ADD8, MASK_ADD8)

The related C file (riscv-gnu-toolchain/riscv-binutils/opcodes/riscv-opc.c) has to be modified as well:

/* name, xlen, isa, operands, match, mask, match_func, pinfo.  */
+ {"add8",         0, INSN_CLASS_P, "d,s,t",     MATCH_ADD8 MASK_ADD8, match_opcode, 0 },
same files I modified in gdb folders also

_Originally posted by @manasabatchu in https://github.com/riscv-collab/riscv-gnu-toolchain/issues/772#issuecomment-1708181817_

TommyMurphyTM1234, 2023-09-06:
Explaining how to add custom instructions to the GCC tools is outside the scope of this repo which is focused on integrating the (now mostly upstream) tools into usable toolchains. There are several existing closed issues about this topic which might help. Otherwise you should seek assistance in another more appropriate forum - e.g. the GCC/binutils etc. mailing lists etc.

closed 2023-09-06 by TommyMurphyTM1234

TommyMurphyTM1234, 2023-09-06:
> There are several existing closed issues about this topic which might help. 

For example:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+riscv-opc.h