riscv-collab/riscv-gnu-toolchain #1330: Having error while make command

m-ertem, 2023-09-13:
After I have installed the prerequisites, I have tried to install the toolchain via make command. However after couple of minutes, the below error was given.

[image]

TommyMurphyTM1234, 2023-09-13:
You haven't provided even the basic information needed for somebody to try to understand our reproduce your issue.
For example, what OS? The full list of commands that you used (`git clone`, `configure`, `make`). A full build log. Etc.

Please also attach logs rather than screenshots as the latter are not convenient to read for many users.

TommyMurphyTM1234, 2023-09-13:
You still haven't installed `python[2]` as per this:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1251

TommyMurphyTM1234, 2023-09-13:
This failed:
```
cd ~/Downloads/issue-1330
sudo apt remove python-is-python3
make distclean
rm -rf installed-tools
./configure --prefix=`pwd`/installed-tools --with-multilib-generator="rv32gc-ilp32d--;rv32i-ilp32--"
make 2>&1 | tee build.log

...

checking if mkdir takes one argument... no
/usr/bin/env: 'python': No such file or directory
invalid option for --with-multilib-generator
make[1]: *** [Makefile:4549: configure-gcc] Error 1
make[1]: Leaving directory '/home/user/Downloads/issue-1330/build-gcc-newlib-stage1'
make: *** [Makefile:589: stamps/build-gcc-newlib-stage1] Error 2
```

closed 2023-09-13 by TommyMurphyTM1234

m-ertem, 2023-09-14:
I think it has worked. How can I check if the make is successful? Thanks for your effort.

m-ertem, 2023-09-14:
Unfortunately I couldn't be able to use riscv64-unknown-elf-gcc command, as seen below log file. I added /opt/riscv/bin to path by sudo nano /etc/environment and by writing :/opt/riscv/bin

[Compile.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12603519/Compile.log)

TommyMurphyTM1234, 2023-09-14:
> Unfortunately I couldn't be able to use riscv64-unknown-elf-gcc command, as seen below log file. I added /opt/riscv/bin to path by sudo nano /etc/environment and by writing :/opt/riscv/bin
> 
> [Compile.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12603519/Compile.log)

If your toolchain built successfully when you installed `python-is-python3` then this has nothing to do with `riscv-gnu-toolchain` but is an issue with your local environment/configuration.

You don't clarify if/how your toolchain built successfully (e.g. your full build log), where it was installed to, what your `$PATH` environment variable is set to after your changes etc. If you set your `$PATH` to precisely `:/opt/riscv/bin` then that's probably incorrect as it would remove any default search directories that are required for normal system behaviour.

TommyMurphyTM1234, 2023-09-14:
> I added /opt/riscv/bin to path by sudo nano /etc/environment and by writing :/opt/riscv/bin

Why are you doing this rather than editing your local account's `.bashrc` or equivalent shell configuration file?

m-ertem, 2023-09-15:
I have edited bashrc file and /opt/riscv/bin can be seen in the path. So I don't think that it is OK.
echo $PATH
:/opt/riscv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/opt/riscv/bin

However, riscv64-unknown-elf-gcc command can't find -lc and -lgloss. 

mertem@mertem:~/Downloads/RISC-V/test/bubble_sort$ riscv64-unknown-elf-gcc bubble_sort.c ../crt0.s -march=rv32i -mabi=ilp32 -T ../linksc.ld -nostartfiles -ffunction-sections -fdata-sections -Wl,--gc-sections -o bubble_sort.elf
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lc
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lgloss
collect2: error: ld returned 1 exit status

TommyMurphyTM1234, 2023-09-15:
If you installed your toolchain in `/opt/riscv`:

> ```
> echo $PATH :/opt/riscv/bin:...
> ```

then why are you picking up a RISC-V linker from `/usr/lib/`?

> ```
> /usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lc 
> /usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lgloss collect2: 
> error: ld returned 1 exit status
> ```

It sounds like your setup is messed up.

You would be better off starting with a clean slate - e.g. a clean installation of the OS on a physical or virtual machine.

TommyMurphyTM1234, 2023-09-15:
> > ```
> > /usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lc 
> > /usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lgloss collect2: 
> > error: ld returned 1 exit status
> > ```

This suggests that you previously installed a RISC-V toolchain from the Ubuntu software repos - e.g.:

`sudo apt install gcc-riscv64-unknown-elf`

which will put things in "system" directories including `/usr/lib/riscv64-unknown-elf`.

If you are using your own build of the toolchain then you need to take care to ensure that only it is used and not bits of some other toolchain.

I would suggest that you at least remove this toolchain and try again:

`sudo apt remove gcc-riscv64-unknown-elf`

mskertem, 2023-09-22:
Thanks for all your effort. As you said the problem is turned out to be the PATH. Hence I added the wrong directory to PATH, riscv64-unknown-elf-gcc command couldn't be found. Then I checked the exact directory of gcc-riscv64-unknown-elf, then added to PATH. Eventually everything works correctly!

TommyMurphyTM1234, 2023-09-22:
> As you said the problem is turned out to be the PATH. Hence I added the wrong directory to PATH, riscv64-unknown-elf-gcc command couldn't be found. Then I checked the exact directory of gcc-riscv64-unknown-elf, then added to PATH. Eventually everything works correctly!

That's why I asked for this information earlier:

> > Unfortunately I couldn't be able to use riscv64-unknown-elf-gcc command, as seen below log file. I added /opt/riscv/bin to path by sudo nano /etc/environment and by writing :/opt/riscv/bin
> > [Compile.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12603519/Compile.log)
> 
> If your toolchain built successfully after you installed `python-is-python3` then this has nothing to do with `riscv-gnu-toolchain` but is an issue with your local environment/configuration.
> 
> You don't clarify if/how your toolchain built successfully (e.g. your full build log), where it was installed to, what your `$PATH` environment variable is set to after your changes etc. If you set your `$PATH` to precisely `:/opt/riscv/bin` then that's probably incorrect as it would remove any default search directories that are required for normal system behaviour.
> 
> Even the following basic information would help.
> 
> ```
> echo $PATH
> ls /opt/riscv/bin
> ```