riscv-collab/riscv-gnu-toolchain #1331: [support]  output file generation with respect to change in memory view

dayanajayaraman, 2023-09-14:
Hi,

Below is the script used to generate hex file for memory dimension [8:0] mem [depth] [bytes] but now memory dimension changes to  [63:0]  mem [depth]

[image]

please let me know what modifications required in script to fix the hex image for preload using readmem , binary output file would be more helpful as the preloader already has readmemb task defined.

dayanajayaraman, 2023-09-14:
Thank you for your inputs and references ! Sure It will be more appropriate to raise this support under riscv tool chain .
That's correct its [7:0] ,  readmemb -> $readmemb fixed typo..

TommyMurphyTM1234, 2023-09-14:
> Thank you for your inputs and references ! GCC_PREFIX = riscv64-unknown-elf. and ABI = -mabi=ilp32 -march=rv32imc_zicsr_zifencei 

I don't understand what the point of this is?

> That's correct its [7:0] , readmemb -> $readmemb fixed typo..

What is `$readmemb`?
Some Verilog feature?

I don't see how this is a RISC-V/`riscv-gnu-toolchain` issue.
Seems more like a question about binutils/`objdump`/BFD and how to convert an ELF file into some desired output format (for use by Verilog `$readmemb`?) to me?

TommyMurphyTM1234, 2023-09-14:
Maybe this might help or give you somewhere to start from?

* https://github.com/sifive/elf2hex

TommyMurphyTM1234, 2023-09-15:
> Maybe this might help or give you somewhere to start from?
> 
> * https://github.com/sifive/elf2hex

If that doesn't help then perhaps the `objcopy --verilog-data-width=bytes` option will:

* https://sourceware.org/binutils/docs/binutils/objcopy.html#:~:text=%2D%2Dverilog%2Ddata%2Dwidth%3Dbytes

Otherwise, from a few quick searches, there seem to be lots of discussions about using `objcopy` to convert files into formats suitable for use with Verilog and simulations so I suspect that this issue is covered elswhere already.

Closing this issue since it's fundamentally not a RISC-V/`riscv-gnu-toolchain` issue.

closed 2023-09-15 by TommyMurphyTM1234