riscv-collab/riscv-gnu-toolchain #1332: Missing build instructions for MacOS running MacPorts in `README.md`

pilonsi, 2023-09-18:
The `README.md` file is missing instructions for building on a MacPorts MacOS system. These would be:

To fetch required dependencies:
```
$ sudo port install python311 gawk gsed gmp mpfr libmpc isl zlib expat texinfo flock
$ sudo ln -s /opt/local/bin/gmake /opt/local/libexec/gnubin/gnumake
```

Before running `./configure`:

```
$ export PATH=/opt/local/libexec/gnubin:$PATH 
$ export CPPFLAGS=-I/opt/local/include 
$ export LDFLAGS=-L/opt/local/lib
```

I have added `gmake` to the install list because when building the Linux toolchaing `glibc` complains that the system `make` is too old:

```
$ make linux
[...]
configure: error:                                                                                                                                                                 
*** These critical programs are missing or too old: make                 
*** Check the INSTALL file for required versions.
make: *** [stamps/build-glibc-linux-headers] Error 1
[...]

$ cat glibc/INSTALL | grep make
[...]
   * GNU 'make' 4.0 or newer
[...]

$ make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
```

Maybe this issue also happens when building from Homebrew but I cannot test it.

I have tested these building both the Newlib and Linux toolchains on a MacOS 12.6.8 host.

I have added these changes to the `README.md` file [here](https://github.com/pilonsi/riscv-gnu-toolchain/blob/readme-macports/README.md). If it looks OK I can submit a PR.

cmuellner, 2024-04-10:
Sorry for not responding earlier. Please go ahead and file a PR to adjust the MacOS build instructions as needed.

TommyMurphyTM1234, 2024-10-12:
Unfortunately it seems like none of the active contributors here have access to macOS hardware (Intel or Arm based) in order to analyse/investigate such issues. Until somebody with access to and knowledge of these platforms steps up to help macOS issues will remain in abeyance. In this case the issue was not followed through with a PR to address the specific issue in question. But even such issues seem piecemeal and ideally it would be better if somebody took a systematic approach to teasing out and explaining all of the issues with building the `riscv-gnu-toolchain` based tools using the relevant tooling (e.g. macOS Intel and Arm, Brew?, XCode, MacPorts, etc.). In the absence of such contributions here the only option that I can think of is to point people in the direction of the xPack Project where I know that @ilg-ul has done a lot of work to establish a stable/reliable build system for macOS (and many other OS targets) and doeument it extensively.

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/

I am inclined to close this issue due to lack of activity and, in particular, the contribution of a suitable PR.