riscv-collab/riscv-gnu-toolchain #1333: Make issue

Krishnakumarmohanraj, 2023-09-22:
Hi Team,

  we unable to clone newlibs in git while giving make, i attached the image. 

[image]
[image]

Can you guide me on this issue

TommyMurphyTM1234, 2023-09-22:
What happens if you try a recursive clone?
```
git clone --verbose --recursive https://github.com/riscv-collab/riscv-gnu-toolchain issue-1333
cd issue-1333
./configure --prefix=`pwd`/installed-tools
make 2>&1 | tee build.log
```
Is this StackOverflow thread of any use on the specific error that you're encountering?

* https://stackoverflow.com/questions/60262230/fatal-unable-to-access-gnutls-handshake-failed-handshake-failed

TommyMurphyTM1234, 2023-09-22:
This worked fine for me using a WSL2 Ubuntu with all available updates installed:
```
git clone --recursive --verbose https://github.com/riscv-collab/riscv-gnu-toolchain issue-1333 2>&1 | tee clone.log
Cloning into 'issue-1333'...
POST git-upload-pack (175 bytes)
POST git-upload-pack (gzip 4017 to 1972 bytes)
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/user/Downloads/issue-1333/binutils'...
Cloning into '/home/user/Downloads/issue-1333/dejagnu'...
Cloning into '/home/user/Downloads/issue-1333/gcc'...
Cloning into '/home/user/Downloads/issue-1333/gdb'...
Cloning into '/home/user/Downloads/issue-1333/glibc'...
Cloning into '/home/user/Downloads/issue-1333/llvm'...
Cloning into '/home/user/Downloads/issue-1333/musl'...
Cloning into '/home/user/Downloads/issue-1333/newlib'...
Cloning into '/home/user/Downloads/issue-1333/pk'...
Cloning into '/home/user/Downloads/issue-1333/qemu'...
Cloning into '/home/user/Downloads/issue-1333/spike'...
Submodule path 'binutils': checked out 'b2bc62b7b4e7638c3a249d2d2728ceb4d5f2b22c'
Submodule path 'dejagnu': checked out 'c298959ef991b389b64a825f70094738c6a48780'
Submodule path 'gcc': checked out '2ee5e4300186a92ad73f1a1a64cb918dc76c8d67'
Submodule path 'gdb': checked out 'e53a8e8685685c97588f8319d993ea6cd5635e47'
Submodule path 'glibc': checked out 'a704fd9a133bfb10510e18702f48a6a9c88dbbd5'
Submodule path 'llvm': checked out '185b81e034ba60081023b6e59504dfffb560f3e3'
Submodule path 'musl': checked out '85e0e3519655220688e757b9d5bfd314923548bd'
Submodule path 'newlib': checked out '415fdd4279b85eeec9d54775ce13c5c412451e08'
Submodule path 'pk': checked out '2efabd3e6604b8a9e8f70baf52f57696680c7855'
Submodule path 'qemu': checked out '621da7789083b80d6f1ff1c0fb499334007b4f51'
Submodule 'dtc' (https://gitlab.com/qemu-project/dtc.git) registered for path 'qemu/dtc'
Submodule 'meson' (https://gitlab.com/qemu-project/meson.git) registered for path 'qemu/meson'
Submodule 'roms/QemuMacDrivers' (https://gitlab.com/qemu-project/QemuMacDrivers.git) registered for path 'qemu/roms/QemuMacDrivers'
Submodule 'roms/SLOF' (https://gitlab.com/qemu-project/SLOF.git) registered for path 'qemu/roms/SLOF'
Submodule 'roms/edk2' (https://gitlab.com/qemu-project/edk2.git) registered for path 'qemu/roms/edk2'
Submodule 'roms/ipxe' (https://gitlab.com/qemu-project/ipxe.git) registered for path 'qemu/roms/ipxe'
Submodule 'roms/openbios' (https://gitlab.com/qemu-project/openbios.git) registered for path 'qemu/roms/openbios'
Submodule 'roms/opensbi' (https://gitlab.com/qemu-project/opensbi.git) registered for path 'qemu/roms/opensbi'
Submodule 'roms/qboot' (https://gitlab.com/qemu-project/qboot.git) registered for path 'qemu/roms/qboot'
Submodule 'roms/qemu-palcode' (https://gitlab.com/qemu-project/qemu-palcode.git) registered for path 'qemu/roms/qemu-palcode'
Submodule 'roms/seabios' (https://gitlab.com/qemu-project/seabios.git/) registered for path 'qemu/roms/seabios'
Submodule 'roms/seabios-hppa' (https://gitlab.com/qemu-project/seabios-hppa.git) registered for path 'qemu/roms/seabios-hppa'
Submodule 'roms/sgabios' (https://gitlab.com/qemu-project/sgabios.git) registered for path 'qemu/roms/sgabios'
Submodule 'roms/skiboot' (https://gitlab.com/qemu-project/skiboot.git) registered for path 'qemu/roms/skiboot'
Submodule 'roms/u-boot' (https://gitlab.com/qemu-project/u-boot.git) registered for path 'qemu/roms/u-boot'
Submodule 'roms/u-boot-sam460ex' (https://gitlab.com/qemu-project/u-boot-sam460ex.git) registered for path 'qemu/roms/u-boot-sam460ex'
Submodule 'roms/vbootrom' (https://gitlab.com/qemu-project/vbootrom.git) registered for path 'qemu/roms/vbootrom'
Submodule 'slirp' (https://gitlab.com/qemu-project/libslirp.git) registered for path 'qemu/slirp'
Submodule 'subprojects/libvfio-user' (https://gitlab.com/qemu-project/libvfio-user.git) registered for path 'qemu/subprojects/libvfio-user'
Submodule 'tests/fp/berkeley-softfloat-3' (https://gitlab.com/qemu-project/berkeley-softfloat-3.git) registered for path 'qemu/tests/fp/berkeley-softfloat-3'
Submodule 'tests/fp/berkeley-testfloat-3' (https://gitlab.com/qemu-project/berkeley-testfloat-3.git) registered for path 'qemu/tests/fp/berkeley-testfloat-3'
Submodule 'tests/lcitool/libvirt-ci' (https://gitlab.com/libvirt/libvirt-ci.git) registered for path 'qemu/tests/lcitool/libvirt-ci'
Submodule 'ui/keycodemapdb' (https://gitlab.com/qemu-project/keycodemapdb.git) registered for path 'qemu/ui/keycodemapdb'
Cloning into '/home/user/Downloads/issue-1333/qemu/dtc'...
[... cut ...]
Submodule path 'qemu/subprojects/libvfio-user': checked out '0b28d205572c80b568a1003db2c8f37ca333e4d7'
Submodule path 'qemu/tests/fp/berkeley-softfloat-3': checked out 'b64af41c3276f97f0e181920400ee056b9c88037'
Submodule path 'qemu/tests/fp/berkeley-testfloat-3': checked out '5a59dcec19327396a011a17fd924aed4fec416b3'
Submodule path 'qemu/tests/lcitool/libvirt-ci': checked out '324355cf62e86fb551408575afb123bac989ac37'
Submodule path 'qemu/ui/keycodemapdb': checked out 'd21009b1c9f94b740ea66be8e48a1d8ad8124023'
Submodule path 'spike': checked out '71f5a8fd1b1185b7c15151f116aa3fef058bf210'
```

TommyMurphyTM1234, 2023-09-22:
`fatal: unable to access ...: gnutls_handshake() failed: Error in the pull function.` seems to be a common enough `git` error with various causes and fixes. You might want to check out other relevant threads on StackOverflow and other forums that may help. I'm pretty sure that your problem is not due to any issue with the `riscv-gnu-toolchain` repository itself but is more likely a local system configuration issue. Maybe you can try a test clone from a clean virtual machine installation?

Krishnakumarmohanraj, 2023-09-22:
Hi Mr.Tommy Murphy, Problem in https ssl issue in network. its works fine now. Thanks for your support

closed 2023-09-22 by Krishnakumarmohanraj