riscv-collab/riscv-gnu-toolchain #1336: Non matching information in debug data can output binary

KelvinChung2000, 2023-10-02:
On the left-hand side is the disassembled binary and on the right-hand side is the DWARF debug information. 

On the left,  the order of operation is SP change -> Value Store -> SP change. 
However, the debug information is showing SP change -> SP change -> Value Store. 

If I understand the DWARF correctly, the calculations of the address of the stored variable will be based on the current CFA, which is the changed address by the `DW_CFA_def_cfa_offset`, as a result, we cannot find the stored value based on the debug information. 

[image]

TommyMurphyTM1234, 2023-10-02:
Please provide a reproducible test case - i.e. the code that you're using and the commands used to generate the assembly code and DWARF data. Also, what toolchain are you using and where did you get it or how did you build it?

KelvinChung2000, 2023-10-02:
[missMatchExample.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12783048/missMatchExample.zip)

To generate the binary: `riscv64-unknown-linux-gnu-gcc -static -O3 -Wall -Wno-unused-label -march=rv64imafd -g -fno-reorder-blocks -o bfs bfs.c local_support.c support.c harness.c`

To generate the dump file: `riscv64-unknown-linux-gnu-objdump -d bfs > bfs.dump`

To generate the DWARF dump file: `riscv64-unknown-linux-gnu-objdump -g bfs > bfs.dwarf`

Compiler version: riscv64-unknown-linux-gnu-gcc (g2ee5e430018) 12.2.0

Please let me know if any further information is needed.

TommyMurphyTM1234, 2023-10-02:
Thanks - unfortunately I cannot really read/interpret the DWARF dump.
For what it's worth, I did compare the RISC-V results to the x86-64 results and the latter does not seem to exhibit the "SP Change -> SP Change -> Value Store" pattern that you mention.
The following are snippets from the disassembly dump and the DWARF dump from the x86-64 `bfs.o` file (not the linked ELF file):
```
// bfs.dump
bfs.o:     file format elf64-x86-64

Disassembly of section .text:

0000000000000000 <bfs>:
   0:   f3 0f 1e fa             endbr64
   4:   41 56                   push   %r14
   6:   49 89 c9                mov    %rcx,%r9
   9:   41 be ff 00 00 00       mov    $0xff,%r14d
   f:   41 55                   push   %r13
  11:   41 bd 00 01 00 00       mov    $0x100,%r13d
  17:   41 54                   push   %r12
  19:   45 31 e4                xor    %r12d,%r12d
  1c:   55                      push   %rbp
  1d:   48 89 f5                mov    %rsi,%rbp
  20:   53                      push   %rbx
  21:   48 89 fb                mov    %rdi,%rbx
  ...
```
```
// bfs.dwarf

00000018 0000000000000044 0000001c FDE cie=00000000 pc=0000000000000000..0000000000000125
  DW_CFA_advance_loc: 6 to 0000000000000006
  DW_CFA_def_cfa_offset: 16
  DW_CFA_offset: r14 (r14) at cfa-16
  DW_CFA_advance_loc: 11 to 0000000000000011
  DW_CFA_def_cfa_offset: 24
  DW_CFA_offset: r13 (r13) at cfa-24
  DW_CFA_advance_loc: 8 to 0000000000000019
  DW_CFA_def_cfa_offset: 32
  DW_CFA_offset: r12 (r12) at cfa-32
  DW_CFA_advance_loc: 4 to 000000000000001d
  DW_CFA_def_cfa_offset: 40
  DW_CFA_offset: r6 (rbp) at cfa-40
  DW_CFA_advance_loc: 4 to 0000000000000021
  DW_CFA_def_cfa_offset: 48
  DW_CFA_offset: r3 (rbx) at cfa-48
  DW_CFA_advance_loc: 15 to 0000000000000030
  DW_CFA_def_cfa_offset: 2112
  DW_CFA_advance_loc1: 203 to 00000000000000fb
  DW_CFA_remember_state
  DW_CFA_def_cfa_offset: 48
  DW_CFA_advance_loc: 1 to 00000000000000fc
  DW_CFA_def_cfa_offset: 40
  DW_CFA_advance_loc: 1 to 00000000000000fd
  DW_CFA_def_cfa_offset: 32
  DW_CFA_advance_loc: 2 to 00000000000000ff
  DW_CFA_def_cfa_offset: 24
  DW_CFA_advance_loc: 2 to 0000000000000101
  DW_CFA_def_cfa_offset: 16
  DW_CFA_advance_loc: 2 to 0000000000000103
  DW_CFA_def_cfa_offset: 8
  DW_CFA_advance_loc: 5 to 0000000000000108
  DW_CFA_restore_state
  DW_CFA_nop
  DW_CFA_nop
  DW_CFA_nop
  DW_CFA_nop
```

KelvinChung2000, 2023-10-02:
I think you might not be able to see the same problem in x86. I have 2 hypotheses:

1. x86 has fewer registers available, the compiler might be spilling or init the variable midway, which makes the SP change not appear in the call frame address (CFA) table.  
2. The variable associated with the second SP change is init the pointer of an array. Since x86 can mix fold load into the instructions, the use of that variable might be folded into the instruction when needed.

pinskia, 2023-10-03:
Just FYI:
> Fair points - would it make more sense to compare it to Arm maybe?
aarch64 does not do 2 adds to sp rather it creates an offset to register and does an add to the sp that way (if it is above the limit of the intermediate. And then the dwarf2 pass does not get confused with the 2 increments to sp ...

TommyMurphyTM1234, 2023-10-11:
@KelvinChung2000 - did you raise this upstream and get any feedback?

KelvinChung2000, 2023-10-11:
I have reported it as a bug. I have also found and submitted a few more problems related to the debug information, but they might not be RISC-V specific. The problem raised here will require either a fix in the code generation, debug information generation, or both. So this shouldn't be resolved any time soon.

Links to the bug: [bug111680](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=111680)

KelvinChung2000, 2023-10-12:
@TommyMurphyTM1234, do you know how long, i.e. months or years, it takes for this sort of stuff to get fixed?

TommyMurphyTM1234, 2023-10-12:
> @TommyMurphyTM1234, do you know how long, i.e. months or years, it takes for this sort of stuff to get fixed?

It really depends on the level of interest and engagement on the specific issue - e.g. how much of an issue it is for whom and what resources they are willing to expend on investigating and resolving it.

TommyMurphyTM1234, 2024-03-31:
Closing this as the issue has been raised upstream.

* https://gcc.gnu.org/bugzilla/show_bug.cgi?id=111680

closed 2024-03-31 by TommyMurphyTM1234