riscv-collab/riscv-gnu-toolchain #1339: Update Spike submodule to avoid test suite failures?

TommyMurphyTM1234, 2023-10-05:
See here:

* https://github.com/riscv/riscv-openocd/issues/932#issuecomment-1747538804

Any reason why the Spike submodule should not be updated to the latest commit on the `master` branch?

TommyMurphyTM1234, 2023-10-05:
Thanks @cmuellner 

> So I am fine with every PR that bumps the Spike reference to the current latest commit on their master branch. I just ask this PR to be tested.

I can take a look but can you clarify (in as much detail as you can) the extent of the expected/requested testing, please?

cmuellner, 2023-10-05:
> I can take a look but can you clarify (in as much detail as you can) the extent of the expected/requested testing, please?

The CI currently tests "newlib, rv64gc-lp64d, spike" (`make report-newlib`).
The README.md says "spike only support rv64* bare-metal/elf toolchain".

So I think testing "newlib, rv64gc-lp64d, spike" such that it does not introduce any regressions should be all that is needed.

TommyMurphyTM1234, 2023-10-05:
> The CI currently tests "newlib, rv64gc-lp64d, spike" (`make report-newlib`). The README.md says "spike only support rv64* bare-metal/elf toolchain".
>
> So I think testing "newlib, rv64gc-lp64d, spike" [...]

Thanks @cmuellner - so, run `make report-newlib` before and after bumping the Spike submodule and compare the results?

cmuellner, 2023-10-05:
Yes, that should be enough.

TommyMurphyTM1234, 2023-10-07:
> `make report-newlib` 

I have tried this but it seems to take an inordinate amount of time - it's been running for over 24 hours and still no sign of it completing! Am I doing something wrong or is there some less resource hungry way of running the necessary tests?

patrick-rivos, 2023-10-07:
That doesn't sound good - in the CI the spike runs typically finish before qemu runs.
I'll take a look on Monday if you're ok with waiting till then.

TommyMurphyTM1234, 2023-10-07:
Thanks @patrick-rivos - I can wait. 
Just to clarify, this is a run of the tests **before** bumping the Spike commit just for a baseline reference.
The machine that I'm using builds the toolchain in an hour or two in case that gives some sort of gauge of its capabilities.
Every time I glance at the console it just seems to report FAIL for tests.
Maybe I'm missing some step here?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=... --enable-multilib --with-sim=spike
make
make build-sim
make report-newlib
```

patrick-rivos, 2023-10-07:
Ah - the CI uses non multilib spike. I'm not sure if spike has multilib support.
I successfully built (baseline and bumping spike to `d1efcdff`):
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
mkdir build
cd build
../configure --prefix=$(pwd) --with-sim=spike
make report-newlib -j32
```
`d1efcdff` had no new failures.

TommyMurphyTM1234, 2023-10-07:
Hi @patrick-rivos - thanks. I assumed that the test suite might need a multilib toolchain. But if it doesn't then does building one matter? Or does the test suite detect a multilib toolchain and run the tests for multiple `arch/abi` targets or something? I find the test suite stuff a bit confusing and lacking in any clear guidance here - unless there is generic guidance upstream that suffices?

Anyway, sounds from your tests that bumping Spike to the latest master branch commit might be safe so should I go ahead and submit a PR for that or do you want to do it?

Thanks again for the feedback.

TommyMurphyTM1234, 2023-10-08:
OK - with a non-multilib toolchain I get these results both before and after bumping Spike to the latest commit on the master branch:

* https://github.com/riscv-software-src/riscv-isa-sim/commit/d1efcdffffee57bab0fdbd2b377c6132b37556fd
```
                 === gcc Summary ===

# of expected passes            122872
# of unexpected failures        38
# of unexpected successes       4
# of expected failures          786
# of unresolved testcases       4
# of unsupported tests          2804

                === g++ Summary ===

# of expected passes            184651
# of unexpected failures        29
# of expected failures          1018
# of unsupported tests          10457

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |      - |
```
I don't really know what the unexpected failures/successes mean but since there is no change after bumping Spike I guess that they are not relevant here?

cmuellner, 2023-10-08:
> OK - with a non-multilib toolchain I get these results both before and after bumping Spike to the latest commit on the master branch:
> 
> * [riscv-software-src/riscv-isa-sim@d1efcdf](https://github.com/riscv-software-src/riscv-isa-sim/commit/d1efcdffffee57bab0fdbd2b377c6132b37556fd)
> 
> ```
>                  === gcc Summary ===
> 
> # of expected passes            122872
> # of unexpected failures        38
> # of unexpected successes       4
> # of expected failures          786
> # of unresolved testcases       4
> # of unsupported tests          2804
> 
>                 === g++ Summary ===
> 
> # of expected passes            184651
> # of unexpected failures        29
> # of expected failures          1018
> # of unsupported tests          10457
> 
>                ========= Summary of gcc testsuite =========
>                             | # of unexpected case / # of unique unexpected case
>                             |          gcc |          g++ |     gfortran |
>  rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |      - |
> ```
> 
> I don't really know what the unexpected failures/successes mean but since there is no change after bumping Spike I guess that they are not relevant here?

If both runs show 0 unexpected issues, then we can say, that the version bump does not introduce any regressions. With this I am fine to merge an update of Spike.

Is there a reason why no gfortran tests were run?

TommyMurphyTM1234, 2023-10-08:
> Is there a reason why no gfortran tests were run?

No idea.
I just did this...
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=$HOME/Downloads/riscv-tools --with-sim=spike
export PATH=$HOME/Downloads/riscv-tools/bin:$PATH
make && make build-sim && make report-newlib
```

cmuellner, 2023-10-08:
I would press the merge button, but this is not a PR... ;)

TommyMurphyTM1234, 2023-10-08:
I looked at the generated `Makefile` and it looks like only C and C++ are enabled for the bare metal/newlib toolchain while C, C++ and Fortran are enabled for the Linux toolchain?
```
stamps/build-gcc-linux-stage1: $(GCC_SRCDIR) $(GCC_SRC_GIT) stamps/build-binutils-linux \
                               stamps/build-linux-headers
        if test -f $</contrib/download_prerequisites && test "false" = "true"; then cd $< && ./contrib/download_prerequisites; fi
        rm -rf $@ $(notdir $@)
        mkdir $(notdir $@)
        cd $(notdir $@) && $</configure \
                --target=$(LINUX_TUPLE) \
                $(CONFIGURE_HOST) \
                --prefix=$(INSTALL_DIR) \
                --with-sysroot=$(SYSROOT) \
                --with-newlib \
                --without-headers \
                --disable-shared \
                --disable-threads \
                --with-system-zlib \
                --enable-tls \
                --enable-languages=c \
                ...

stamps/build-gcc-linux-stage2: $(GCC_SRCDIR) $(GCC_SRC_GIT) $(addprefix stamps/build-glibc-linux-,$(GLIBC_MULTILIB_NAMES)) \
                               stamps/build-glibc-linux-headers
        rm -rf $@ $(notdir $@)
        mkdir $(notdir $@)
        cd $(notdir $@) && $</configure \
                --target=$(LINUX_TUPLE) \
                $(CONFIGURE_HOST) \
                --prefix=$(INSTALL_DIR) \
                --with-sysroot=$(SYSROOT) \
                --with-pkgversion="$(GCCPKGVER)" \
                --with-system-zlib \
                --enable-shared \
                --enable-tls \
                --enable-languages=c,c++,fortran \
                ...

stamps/build-gcc-linux-native: $(GCC_SRCDIR) $(GCC_SRC_GIT) stamps/build-gcc-linux-stage2 stamps/build-binutils-linux-native
        if test -f $</contrib/download_prerequisites; then cd $< && ./contrib/download_prerequisites; fi
        rm -rf $@ $(notdir $@)
        mkdir $(notdir $@)
        cd $(notdir $@) && $</configure \
                --host=$(LINUX_TUPLE) \
                --target=$(LINUX_TUPLE) \
                $(CONFIGURE_HOST) \
                --prefix=$(INSTALL_DIR)/native \
                --without-system-zlib \
                --enable-shared \
                --enable-tls \
                --enable-languages=c,c++,fortran \
                ...

stamps/build-gcc-newlib-stage1: $(GCC_SRCDIR) $(GCC_SRC_GIT) stamps/build-binutils-newlib
        if test -f $</contrib/download_prerequisites && test "false" = "true"; then cd $< && ./contrib/download_prerequisites; fi
        rm -rf $@ $(notdir $@)
        mkdir $(notdir $@)
        cd $(notdir $@) && $</configure \
                --target=$(NEWLIB_TUPLE) \
                $(CONFIGURE_HOST) \
                --prefix=$(INSTALL_DIR) \
                --disable-shared \
                --disable-threads \
                --disable-tls \
                --enable-languages=c,c++ \
                ...

stamps/build-gcc-newlib-stage2: $(GCC_SRCDIR) $(GCC_SRC_GIT) stamps/build-newlib \
                stamps/merge-newlib-nano
        rm -rf $@ $(notdir $@)
        mkdir $(notdir $@)
        cd $(notdir $@) && $</configure \
                --target=$(NEWLIB_TUPLE) \
                $(CONFIGURE_HOST) \
                --prefix=$(INSTALL_DIR) \
                --disable-shared \
                --disable-threads \
                --enable-languages=c,c++ \
                ...

stamps/build-gcc-musl-stage1: $(GCC_SRCDIR) $(GCC_SRC_GIT) stamps/build-binutils-musl \
                               stamps/build-linux-headers
        if test -f $</contrib/download_prerequisites && test "false" = "true"; then cd $< && ./contrib/download_prerequisites; fi
        rm -rf $@ $(notdir $@)
        mkdir $(notdir $@)
        cd $(notdir $@) && $</configure \
                --target=$(MUSL_TUPLE) \
                $(CONFIGURE_HOST) \
                --prefix=$(INSTALL_DIR) \
                --without-headers \
                --disable-shared \
                --disable-threads \
                --with-system-zlib \
                --enable-tls \
                --enable-languages=c \
                ...

stamps/build-gcc-musl-stage2: $(GCC_SRCDIR) $(GCC_SRC_GIT) stamps/build-musl-linux \
                               stamps/build-musl-linux-headers
        rm -rf $@ $(notdir $@)
        mkdir $(notdir $@)
        # Disable libsanitizer for now
        # https://github.com/google/sanitizers/issues/1080
        cd $(notdir $@) && $</configure \
                --target=$(MUSL_TUPLE) \
                $(CONFIGURE_HOST) \
                --prefix=$(INSTALL_DIR) \
                --with-sysroot=$(SYSROOT) \
                --with-system-zlib \
                --enable-shared \
                --enable-tls \
                --enable-languages=c,c++ \
                ...
```

cmuellner, 2023-10-08:
> I looked at the generated `Makefile` and it looks like only C and C++ are enabled for the bare metal/newlib toolchain while C, C++ and Fortran are enabled for the Linux toolchain?

Ah, right. I forgot that we don't run glibc tests.

Results look good. Waiting for the PR!

TommyMurphyTM1234, 2023-10-08:
> Results look good. Waiting for the PR!

Hopefully this is OK?

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1340

TommyMurphyTM1234, 2023-10-08:
> Or does the test suite detect a multilib toolchain and run the tests for multiple `arch/abi` targets or something?

Just FYI - that does seem to be the case alright and explains why my test suite was taking so long when I ran it with the multilib toolchain. When run with a non-multilib toolchain it took a few hours which was OK.
```
                === gcc tests ===

Schedule of variations:
    riscv-sim/-march=rv32i/-mabi=ilp32/-mcmodel=medlow
    riscv-sim/-march=rv32iac/-mabi=ilp32/-mcmodel=medlow
    riscv-sim/-march=rv32im/-mabi=ilp32/-mcmodel=medlow
    riscv-sim/-march=rv32imac/-mabi=ilp32/-mcmodel=medlow
    riscv-sim/-march=rv32imafc/-mabi=ilp32f/-mcmodel=medlow
    riscv-sim/-march=rv64imac/-mabi=lp64/-mcmodel=medlow
    riscv-sim/-march=rv64imafdc/-mabi=lp64d/-mcmodel=medlow
```

cmuellner, 2023-10-09:
The corresponding PR (#1340) has been merged.
Let's close this ticket.

closed 2023-10-09 by cmuellner