riscv-collab/riscv-gnu-toolchain #1343: cannot install the whole toolchain

wangzhiwei18, 2023-10-10:
I have run such commands:
./configure --prefix=/home/wzw/toolchain --enable-multilib
make -j8
make linux -j8

but there are only few tools installed in the floder(without riscv64-unknown-elf-gcc and many other tools):
[image]

TommyMurphyTM1234, 2023-10-10:
I'm not sure that it makes sense to build both the bare-metal/Newlib and Linux toolchains such that they get installed into the same target directory.

You need to check your build log for errors/issues.
If you can't see them then compress/zip and attach it for analysis by others here.
For such analysis it makes sense to do a non-parallel build by omitting any `-jn` options.
For example:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --enable-multilib
make 2>&1 | tee build.log
make linux 2>&1 | tee -a build.log
```

TommyMurphyTM1234, 2023-10-10:
I used a new/clean install of Ubuntu on WSL2 on Windows 10 to do a similar build and it worked fine.
```
# In Windows PowerShell
# Ensure clean install of Ubuntu on WSL2
wsl --unregister Ubuntu
wsl --install Ubuntu

# In WSL2/Ubuntu command shell
# Update
sudo apt update
sudo apt upgrade

# Install riscv-gnu-tools prerequisites - https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites
sudo apt install autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev

# Clone, configure and build riscv-gnu-toolchain
git clone [https://github.com/riscv-collab/riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites)
cd riscv-gnu-toolchain
./configure --prefix=$HOME/riscv-tools
make 2>&1 | tee build.log 
make linux 2>&1 | tee -a build.log 
```

wangzhiwei18, 2023-10-11:
> I used a new/clean install of Ubuntu on WSL2 on Windows 10 to do a similar build and it worked fine.
> 
> ```
> # In Windows PowerShell
> # Ensure clean install of Ubuntu on WSL2
> wsl --unregister Ubuntu
> wsl --install Ubuntu
> 
> # In WSL2/Ubuntu command shell
> cat /etc/os-release
> PRETTY_NAME="Ubuntu 22.04.3 LTS"
> NAME="Ubuntu"
> VERSION_ID="22.04"
> VERSION="22.04.3 LTS (Jammy Jellyfish)"
> VERSION_CODENAME=jammy
> ID=ubuntu
> ID_LIKE=debian
> HOME_URL="https://www.ubuntu.com/"
> SUPPORT_URL="https://help.ubuntu.com/"
> BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
> PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
> UBUNTU_CODENAME=jammy
> 
> # Update
> sudo apt update
> sudo apt upgrade
> 
> # Install riscv-gnu-tools prerequisites - https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites
> sudo apt install autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev
> 
> # Clone, configure and build riscv-gnu-toolchain
> git clone [https://github.com/riscv-collab/riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites)
> cd riscv-gnu-toolchain
> ./configure --prefix=$HOME/riscv-tools
> make 2>&1 | tee build.log 
> make linux 2>&1 | tee -a build.log 
> ```
> 
> I'm still not sure that it makes sense or is safe to build a "merged" bare-metal/newlib and Linux toolchain like this or if any files common but different between the two might clash. To avoid potential problems I would generally keep the two toolchain flavours separate.

Thanks a lot!
I also successfully installed it in a brand new Ubuntu environment. It might have been other tools in the previous environment that caused the installation to fail.

closed 2023-10-11 by TommyMurphyTM1234

TommyMurphyTM1234, 2023-10-11:
> there seem to be a minimal number of files common between the two which have no differences 

Correction - there **are** some files in both toolchain builds that **are** different.
As such, I would exercise extreme caution in building a "merged" newlib/Linux toolchain - in fact I would generally avoid it.
```
.\lib\libcc1.la                            different (\\wsl.localhost\Ubuntu\home\user\issue-1343\riscv-tools-linux is more recent)
.\lib\libcc1.so                            Both files unreadable (\\wsl.localhost\Ubuntu\home\user\issue-1343\riscv-tools-linux is more recent)
.\lib\libcc1.so.0                          Both files unreadable (\\wsl.localhost\Ubuntu\home\user\issue-1343\riscv-tools-linux is more recent)
.\lib\libcc1.so.0.0.0                      different (\\wsl.localhost\Ubuntu\home\user\issue-1343\riscv-tools-linux is more recent)
.\lib\bfd-plugins\libdep.so                different (\\wsl.localhost\Ubuntu\home\user\issue-1343\riscv-tools-linux is more recent)
.\share\gdb\python\gdb\framedecorator.py   Both files unreadable (\\wsl.localhost\Ubuntu\home\user\issue-1343\riscv-tools-linux is more recent)
.\share\gdb\python\gdb\frameiterator.py    Both files unreadable (\\wsl.localhost\Ubuntu\home\user\issue-1343\riscv-tools-linux is more recent)
.\share\info\dir                           different (\\wsl.localhost\Ubuntu\home\user\issue-1343\riscv-tools-linux is more recent)
```
[image]