riscv-collab/riscv-gnu-toolchain #1350: riscv64-unknown-elf-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory

Golnaz-Korkian, 2023-10-14:
Hello all,
I am trying to install riscv-toolchan. When I run make command, showing the error: "riscv64-unknown-elf-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory". 

**whereis gcc:** gcc: /usr/bin/gcc /usr/lib/gcc /usr/share/gcc /usr/share/man/man1/gcc.1.gz
**whereis cc1:** cc1: /usr/lib/gcc/x86_64-linux-gnu/11/cc1 /usr/lib/gcc/riscv64-unknown-elf/10.2.0/cc1 /usr/lib/gcc/avr/5.4.0/cc1 /usr/lib/gcc/x86_64-linux-gnu/12/cc1

**find /usr/ -name "*cc1*"**
/usr/lib/gcc/riscv64-unknown-elf/10.2.0/plugin/libcc1plugin.la
/usr/lib/gcc/riscv64-unknown-elf/10.2.0/plugin/libcc1plugin.so.0.0.0
/usr/lib/gcc/riscv64-unknown-elf/10.2.0/plugin/libcc1plugin.so.0
/usr/lib/gcc/riscv64-unknown-elf/10.2.0/plugin/libcc1plugin.so
/usr/lib/gcc/riscv64-unknown-elf/10.2.0/cc1
/usr/lib/gcc/riscv64-unknown-elf/10.2.0/cc1plus
/usr/lib/gcc/x86_64-linux-gnu/12/libcc1.so
/usr/lib/gcc/x86_64-linux-gnu/12/plugin/libcc1plugin.so.0.0.0
/usr/lib/gcc/x86_64-linux-gnu/12/plugin/libcc1plugin.so.0
/usr/lib/gcc/x86_64-linux-gnu/12/plugin/libcc1plugin.so
/usr/lib/gcc/x86_64-linux-gnu/12/cc1
/usr/lib/gcc/x86_64-linux-gnu/11/libcc1.so
/usr/lib/gcc/x86_64-linux-gnu/11/plugin/libcc1plugin.so.0.0.0
/usr/lib/gcc/x86_64-linux-gnu/11/plugin/libcc1plugin.so.0
/usr/lib/gcc/x86_64-linux-gnu/11/plugin/libcc1plugin.so
/usr/lib/gcc/x86_64-linux-gnu/11/cc1
/usr/lib/gcc/x86_64-linux-gnu/11/cc1plus
/usr/lib/gcc/avr/5.4.0/cc1
/usr/lib/gcc/avr/5.4.0/cc1plus
/usr/lib/modules/6.2.0-33-generic/kernel/drivers/iio/adc/cc10001_adc.ko
/usr/lib/modules/6.2.0-34-generic/kernel/drivers/iio/adc/cc10001_adc.ko
/usr/lib/x86_64-linux-gnu/libcc1.so.0
/usr/lib/x86_64-linux-gnu/libcc1.so.0.0.0
/usr/lib/debug/.build-id/92/d2e3d35fbf88f0d2533cc1691e3ee1045a554b.debug
/usr/share/doc/libcc1-0
/usr/share/doc/avr-libc/avr-libc-user-manual/dir_eed8f1d749f716b40e52eb2f4ccc1142.html
/usr/share/doc/avr-libc/avr-libc-user-manual-xml/dir_eed8f1d749f716b40e52eb2f4ccc1142.xml
/usr/bin/riscv64-unknown-elf-gcc1

Also, I added cc1 to path with modifying ./bashrc: 
**export PATH=$PATH:/usr/lib/gcc/x86_64-linux-gnu/11/
export PATH=$PATH:/usr/lib/gcc/riscv64-unknown-elf/10.2.0/**

Could you please help me to solve this issue?

TommyMurphyTM1234, 2023-10-14:
The steps outlined here have worked reliably for me for a long time on WSL2/Ubuntu (as here) and many other Linux distros:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1343#issuecomment-1755446261

Please provide more details including the OS on which you are trying to build the toolchain, your `git clone`, `configure` and `make` commands, your build log (captured using `make 2>&1 | tee build.log`) etc.

TommyMurphyTM1234, 2023-10-14:
> Also, I added cc1 to path with modifying ./bashrc: **export PATH=$PATH:/usr/lib/gcc/x86_64-linux-gnu/11/ export PATH=$PATH:/usr/lib/gcc/riscv64-unknown-elf/10.2.0/**

This seems very old - RISC-V GCC is on 13.2 now.

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1346

Golnaz-Korkian, 2023-10-14:
**lsb_release -a:**
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

**I have used these commands for riscv:** 

$git clone  https://github.com/riscv/riscv-gnu-toolchain
$git checkout 3e8b333
$git submodule update --init --recursive
$mkdir [path/to/repo]/riscv-gnu-toolchain/build
$ cd [path/to/repo]/riscv-gnu-toolchain/build
$ ../configure --prefix=$RISCV --enable-multilib --with-cmodel=medany --with-arch=rv64imafd --with-abi=lp64d
$sudo make

TommyMurphyTM1234, 2023-10-14:
> `$git checkout 3e8b333`

Why this commit?
It's very old:

* https://github.com/riscv-collab/riscv-gnu-toolchain/commit/3e8b333f85eb3378e346318f277805513053c6e8

However, otherwise the build commands look OK at a glance so I can only assume that the issue is with your environment.
I would suggest trying a clean virtual or physical machine installation along the lines of the WSL2/Ubuntu instructions that I linked to earlier and see if the problem persists there.

Golnaz-Korkian, 2023-10-14:
[config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12904963/config.log)
[build.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12904964/build.log)

Please kindly find the attached logs. I will try your solutions and since I am new in Linux, it might take time.Then coming back again. Thanks!

TommyMurphyTM1234, 2023-10-14:
> [config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12904963/config.log) [build.log]

I'm not sure that it's a good idea to have one RISC-V toolchain on your `PATH` when trying to build another one?
> ```
> Line  43: PATH: /usr/lib/gcc/riscv64-unknown-elf/10.2.0/
> ```

There also seem to be other toolchains on the `PATH` which may or may not cause issues (obviously you need at least one x86_64 toolchain though):

> ```
> PATH: /usr/lib/gcc/x86_64-linux-gnu/11/
> PATH: /usr/lib/gcc/riscv64-unknown-elf/10.2.0/
> PATH: /usr/lib/gcc/avr/5.4.0/
> PATH: /usr/lib/gcc/x86_64-linux-gnu/12/
> ```

I would be inclined to try with just the bare minimum default user `PATH` configured for your OS - e.g. create a new user account and try building under that. 

But the best approach might still be what I suggested - a clean virtual/physical machine install so that you can start with a clean slate and see if the tools build there OK using the sort of instructions that I linked to earlier.

TommyMurphyTM1234, 2023-10-14:
I did the following on a clean WSL2/Ubuntu installation after installing all available updates and [the `riscv-gnu-toolchain` prerequisites](https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites):
```
git clone https://github.com/riscv/riscv-gnu-toolchain issue-1350
cd issue-1350/
git checkout 3e8b333
git submodule update --init --recursive
mkdir build
cd build
../configure --enable-multilib --with-cmodel=medany --with-arch=rv64imafd --with-abi=lp64d --prefix=`pwd`/installed-tools
make 2>&1 | tee build.log
```
and the build completed without any issues.

`config.log`: [config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12905823/config.log)
`build.log` (compressed): [build.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/12905824/build.zip)

TommyMurphyTM1234, 2023-10-17:
Any update @Golnaz-Korkian?
I'm pretty sure that this is not an issue with `riscv-gnu-toolchain` itself and is almost certainly a local environment/setup issue.

Golnaz-Korkian, 2023-10-17:
Thanks Tommy for your time. I installed another branch of riscv, however, there were some problems again that as you mentioned it belonged to environmental variables. Now, it was solver. Thanks again!

closed 2023-10-17 by TommyMurphyTM1234