riscv-collab/riscv-gnu-toolchain #1355: Help needed to understand how data alignment works

DaHuoKolmostar, 2023-10-17:
I'm trying to understand how data alignment works with riscv build toolchain.
I know I can use `__attribute__((aligned))` to manually set how a variable or a type should align. However I believe there are some basic rules of padding for structs. For example as I experimented:
If I code like this
```c
struct stu_b {
    char c;
    int i[4];
    short s;
};
```

and the compiled data would be like:
```c
struct stu_b {
    int i[4];
    char c; // reordered by compiler
    char padding; // added by compiler
    short s;
};
```

So is there any document I can refer to how data alignment works with riscv compilers? How paddings are added and how data reorder is performed?

Thanks

TommyMurphyTM1234, 2023-10-17:
Alignment of data is going to depend on many factors - the specific toolchain, the compilation (in particular optimisation) options used, the ABI, the linker script etc. You really need to look at the documentation for the toolchain that you use (e.g. bare metal GCC, Linux GCC, LLVM/Clang, the linker etc.).

cmuellner, 2023-10-17:
> So is there any document I can refer to how data alignment works with riscv compilers?

C/C++ type details can be found here:
https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-cc.adoc

> How paddings are added and how data reorder is performed?

Struct memory layout is not standardized and compiler-specific.
This means, that good code should not assume anything here (`offsetof` is a good building block).
Differences in type memory layouts is also a reason to avoid linking object files generated by different compilers.

TommyMurphyTM1234, 2023-10-17:
Not an issue with `riscv-gnu-toolchain` so I am closing this.
General support on stuff like this is better done in other forums such as the mailing lists, upstream project forums etc.

closed 2023-10-17 by TommyMurphyTM1234