riscv-collab/riscv-gnu-toolchain #1356: cannot create directory 'libnosys': No such file or directory

nidhinmohancheriyan, 2023-10-17:
Getting error like this:

=== configuring in libnosys (/home/nidhin/tenstorrent/build/build-newlib/riscv64-unknown-elf/libgloss/libnosys)
mkdir: cannot create directory 'libnosys': No such file or directory
sh: 0: getcwd() failed: No such file or directory
mkdir: cannot create directory 'libnosys': No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
/home/nidhin/tenstorrent/riscv-gnu-toolchain/riscv-newlib/libgloss/configure: line 4589: cd: libnosys: No such file or directory
configure: running /bin/bash /home/nidhin/tenstorrent/build/../riscv-gnu-toolchain/riscv-newlib/libgloss/libnosys/configure --disable-option-checking '--prefix=/home/nidhin/tenstorrent/riscv-tools'  '--with-newlib' '--enable-multilib' '--with-cross-host=x86_64-pc-linux-gnu' '--enable-newlib-io-long-double' '--enable-newlib-io-long-long' '--enable-newlib-io-c99-formats' '--enable-newlib-register-fini' '--program-transform-name=s&^&riscv64-unknown-elf-&' '--with-target-subdir=riscv64-unknown-elf' '--build=x86_64-pc-linux-gnu' '--host=riscv64-unknown-elf' '--target=riscv64-unknown-elf' 'build_alias=x86_64-pc-linux-gnu' 'host_alias=riscv64-unknown-elf' 'target_alias=riscv64-unknown-elf' --cache-file=.././config.cache --srcdir=/home/nidhin/tenstorrent/build/../riscv-gnu-toolchain/riscv-newlib/libgloss/libnosys
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
pwd: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
configure: error: working directory cannot be determined
configure: error: /home/nidhin/tenstorrent/build/../riscv-gnu-toolchain/riscv-newlib/libgloss/riscv/configure failed for riscv
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
/home/nidhin/tenstorrent/build/../riscv-gnu-toolchain/riscv-newlib/newlib/libc/configure: line 13219: ./config.status: No such file or directory
/home/nidhin/tenstorrent/build/../riscv-gnu-toolchain/riscv-newlib/newlib/libc/configure: line 13322: conf99149subs.sh: No such file or directory
configure: error: could not make ./config.status
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory

TommyMurphyTM1234, 2023-10-17:
This is almost certainly a local environment/configuration issue.
I have been doing dozens of builds lately on different Linux systems and have never encountered this problem.
You need to clarify more details - for example:

* What host OS/version
* The `git clone`, `configure` and `make` commands used
* Your `config.log` and (compressed) build log (e.g. `make 2>&1 | tee build.log`)

Are you sure that some other task isn't deleting stuff in the build directories while the build is in progress?
Some of the errors in your log snippet certainly suggest that something like that could be happening.

TommyMurphyTM1234, 2023-10-17:
Another suggestion - install a clean machine (virtual or physical - with virtual being the easier in most cases)and then try to build the `riscv-gnu-toolchain` tools from scratch. For example - using Ubuntu or another Debian based distro...
```
# Install clean OS

# Install any available updates
sudo apt update
sudo apt upgrade

# Install riscv-gnu-toolchain prerequisites: https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites
sudo apt install autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev

# Clone riscv-gnu-toolchain: https://github.com/riscv-collab/riscv-gnu-toolchain#getting-the-sources
cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1356
cd issue-1356

# Configure: https://github.com/riscv-collab/riscv-gnu-toolchain#installation-newliblinux-multilib
./configure --prefix=`pwd`/installed-tools --enable-multilib

# Build
make 2>&1 | tee build.log
```

nidhinmohancheriyan, 2023-10-18:
Im using Ubunti 20.04 and run the following cmd:
./as-new -I../../../riscv-gnu-toolchain/riscv-gcc/include -I../../../riscv-gnu-toolchain/riscv-gcc/include/blackhole --traditional-format -march=rv32il -mabi=ilp32 -mblackhole -misa-spec="2.2" -o t.o t.s

TommyMurphyTM1234, 2023-10-18:
> and run the following cmd: ./as-new -I../../../riscv-gnu-toolchain/riscv-gcc/include -I../../../riscv-gnu-toolchain/riscv-gcc/include/blackhole --traditional-format -march=rv32il -mabi=ilp32 -mblackhole -misa-spec="2.2" -o t.o t.s

This command makes no sense to me and isn't part of the usual `git clone`, `configure` and `make` command sequence normally used to build the `riscv-gnu-toolchain` tools as outlined in my previous post.
You haven't clarified the details that I requested in my first post
You presumably haven't tried the suggestion in my second post.

Building on Ubuntu 20.04 works fine for me and also for the `riscv-gnu-toolchain` CI workflow actions:

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/b86b2b37d0acc607156ff56ff17ee105a9b48897/.github/workflows/build.yaml#L16
* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/b86b2b37d0acc607156ff56ff17ee105a9b48897/.github/workflows/nightly-release.yaml#L51
* https://github.com/riscv-collab/riscv-gnu-toolchain/actions

Your issue is almost certainly some local host configuration/setup issue but without more detailed info it's impossible to speculate or comment further.

TommyMurphyTM1234, 2023-10-18:
> `-march=rv32il`

This is either an invalid RISC-V architecture value or involves some non-standard extension (`L`) not supported by `riscv-gnu-toolchain`.

TommyMurphyTM1234, 2023-10-22:
Closing as invalid.

closed 2023-10-22 by TommyMurphyTM1234