riscv-collab/riscv-gnu-toolchain #1357: Run command is getting stuck

nidhinmohancheriyan, 2023-10-19:
Im running this command and its getting stuck. 
 A new class called blackhole is added and the directory contaiuns the headerfile. 

./as-new -I../../../riscv-gnu-toolchain/riscv-gcc/include -I../../../riscv-gnu-toolchain/riscv-gcc/include/blackhole --traditional-format -march=rv32il -mabi=ilp32 -mblackhole -misa-spec="2.2" -o t.o t.s

t.s contains:

``.file   "smoke.cc"
        .option nopic
        .attribute arch, "rv32il"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  1
        .globl  _Z5smokev
        .type   _Z5smokev, @function
_Z5smokev:
        SPFLOAD 1, 2, 3, 4
        .size   _Z5smokev, .-_Z5smokev
        .align  1
main:
        addi    sp,sp,-16
        sw      ra,12(sp)
        call    _Z5smokev
        lw      ra,12(sp)
        li      a0,0
        addi    sp,sp,16
        jr      ra
        .size   main, .-main
        .ident  "GCC: (g2ee5e430018-dirty) 12.2.0"
~                                                    
Thanks

TommyMurphyTM1234, 2023-10-19:
I don't understand why you are opening a new issue about this when it's already being dealt with here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1356#issuecomment-1767544128

And, as I said there, you have not clarified precisely what commands you are using to build the toolchain or provided logs etc.

I have no idea what the `as-new` command is, what `blackhole` is, what the non-standard (?) architecture `rv32il` is etc. 

Also this Issue tracker is for issues with the `riscv-gnu-toolchain` and not for issues with other upstream projects or for general support. Please clarify what issue with `riscv-gnu-toolchain` you are claiming, provide the aforementioned information and ideally a reproducible test case.

Otherwise these issues will be closed as invalid.

TommyMurphyTM1234, 2023-10-19:
Original post deleted.
Closing as invalid.

closed 2023-10-19 by TommyMurphyTM1234