riscv-collab/riscv-gnu-toolchain #1358: Mac: Undefined symbols for architecture arm64

howjmay, 2023-10-22:
When compiling on M1 Mac. The following error is returned when I ran `make`
```
Undefined symbols for architecture arm64:
  "___aarch64_cas8_sync", referenced from:
      _macho_add in libbacktrace.a(macho.o)
      _backtrace_initialize in libbacktrace.a(macho.o)
      _backtrace_dwarf_add in libbacktrace.a(dwarf.o)
  "___aarch64_swp4_sync", referenced from:
      _backtrace_free in libbacktrace.a(mmap.o)
      _backtrace_alloc in libbacktrace.a(mmap.o)
ld: symbol(s) not found for architecture arm64
```

TommyMurphyTM1234, 2023-10-22:
When compiling what?
You need to provide a lot more detail and context.

howjmay, 2023-10-22:
Hi @TommyMurphyTM1234 thanks for answering.
I have provided my command to generate makefile. Please check!

TommyMurphyTM1234, 2023-10-22:
Sorry, I've no experience with Mac builds of the `riscv-gnu-toolchain` or ability to try it.

Is this of any relevance?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1332

Is this build approach of any use?

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
  * https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#build-details
    * https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/xpack/README-MAINTAINER.md

howjmay, 2023-10-22:
xpack may help, but it is too huge to understand. I am still trying to figure it out.
I am so curious what is the possible reason that I meet undefined symbols there.

TommyMurphyTM1234, 2023-10-23:
> xpack may help, but it is too huge to understand. 

Maybe @ilg-ul might be able to comment or offer some advice?

ilg-ul, 2023-10-23:
I'm not familiar with the MacPorts environment, but I would say that `--prefix ... --with-cmodel=medany --enable-multilib` is not enough for `configure` to prepare a correct build environment, you have to provide several libraries, including `libisl`.

TommyMurphyTM1234, 2023-10-23:
> I'm not familiar with the MacPorts environment

I think that @howjmay is actually using Homebrew not MacPorts:

* https://github.com/riscv-software-src/homebrew-riscv/issues/111

I linked to the MacPorts issue in case it had any info common to both - there isn't a lot of info here on the `riscv-gnu-toolchain` repo about using Mac to build the toolchain.

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1332

> you have to provide several libraries, including `libisl`.

The MacPorts issue above does mention `isl` alright.
Maybe installing the list of prerequisites mentioned there might help even with Homebrew?

ilg-ul, 2023-10-23:
> what are the libraries

They're quite a lot, see the build script:

- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/xpack/scripts/versioning.sh

Please note that installing those libraries in HomeBrew/MacPorts specific folders does not guarantee that `configure` will identify them, and you might have to explicitly provide the paths to those folders.

howjmay, 2023-10-23:
> They're quite a lot, see the build script:
>
> * https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/xpack/scripts/versioning.sh
>
> Please note that installing those libraries in [...]

Yes that was exactly what I met. I have checked out I installed libisl, but configure can't identify it. I need to figure it out how to explicitly identify them too.
But thank you!!

howjmay, 2023-10-24:
@ilg-ul I have a question about xPack project. I didn't see you call `configure` file. Are all the building steps in the package.json file?

ilg-ul, 2023-10-24:
- https://github.com/xpack-dev-tools/xbb-helper-xpack/blob/918109e955b484dcab3e3e21ff9d30067b17ee42/dependencies/gcc-cross.sh#L426
- https://github.com/xpack-dev-tools/xbb-helper-xpack/blob/918109e955b484dcab3e3e21ff9d30067b17ee42/dependencies/gcc-cross.sh#L778

howjmay, 2023-10-24:
Update:
The error seems to be an error happened when building gcc on M1. Another people met the same problem. Not sure how can I assign the correct setting or compiler to solve it

https://stackoverflow.com/questions/72922055/how-do-i-fix-the-linker-error-when-building-gcc-on-mac-m1

TommyMurphyTM1234, 2023-10-24:
> Update: The error seems to be an error happened when building gcc on M1. Another people met the same problem. Not sure how can I assign the correct setting or compiler to solve it
> 
> https://stackoverflow.com/questions/72922055/how-do-i-fix-the-linker-error-when-building-gcc-on-mac-m1

Are you sure that it's the same issue?
The unresolved symbols in your original report:
```
Undefined symbols for architecture arm64:
  "___aarch64_cas8_sync", referenced from:
      _macho_add in libbacktrace.a(macho.o)
      _backtrace_initialize in libbacktrace.a(macho.o)
      _backtrace_dwarf_add in libbacktrace.a(dwarf.o)
  "___aarch64_swp4_sync", referenced from:
      _backtrace_free in libbacktrace.a(mmap.o)
      _backtrace_alloc in libbacktrace.a(mmap.o)
ld: symbol(s) not found for architecture arm64
```
are different to the ones in the StackOverflow thread:
```
Undefined symbols for architecture arm64:
  "___aarch64_cas8_acq_rel", referenced from:
      _macho_add in libbacktrace.a(macho.o)
      _backtrace_initialize in libbacktrace.a(macho.o)
      _backtrace_dwarf_add in libbacktrace.a(dwarf.o)
  "___aarch64_swp4_acq", referenced from:
      _backtrace_free in libbacktrace.a(mmap.o)
      _backtrace_alloc in libbacktrace.a(mmap.o)
  "_host_hooks", referenced from:
      c_common_no_more_pch() in c-pch.o
      toplev::main(int, char**) in libbackend.a(toplev.o)
      gt_pch_save(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_restore(__sFILE*) in libbackend.a(ggc-common.o)
ld: symbol(s) not found for architecture arm64
```

howjmay, 2023-10-24:
@TommyMurphyTM1234 you may be right.
But I just made some progress by removing the gcc on my machine. Now it is totally using clang, and now I process until build-gdb-newlib

I pretty sure I have installed libgmp already. How should I solve this? Can I pass `--with-gmp=/opt/homebrew/Cellar/gmp` directly in to root `./configure`?
```
12 warnings generated.
  CXX      event-pipe.o
  AR       libgdbsupport.a
mkdir ./gdb
Configuring in ./gdb
configure: creating cache ./config.cache
checking whether to enable maintainer-specific portions of Makefiles... no
checking for aarch64-apple-darwin22.3.0-gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... rm: conftest.dSYM: is a directory
rm: conftest.dSYM: is a directory
yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking build system type... aarch64-apple-darwin22.3.0
checking host system type... aarch64-apple-darwin22.3.0
checking target system type... riscv64-unknown-elf
checking for dlfcn.h... yes
checking for windows.h... no
checking for library containing dlsym... none required
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking how to print strings... printf
checking for a sed that does not truncate output... /opt/homebrew/bin/gsed
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for aarch64-apple-darwin22.3.0-objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for aarch64-apple-darwin22.3.0-ar... ar 
checking for aarch64-apple-darwin22.3.0-strip... no
checking for strip... strip
checking for aarch64-apple-darwin22.3.0-ranlib... ranlib 
checking for gawk... /opt/homebrew/bin/gawk
checking command to parse /usr/bin/nm -B output from gcc object... ok
[... 72 lines cut ...]
checking for library containing socketpair... none required
checking for libzstd >= 1.4.0... yes
checking for ld used by GCC... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for shared library run path origin... done
checking for iconv... yes
checking how to link with libiconv... -liconv
checking for iconv declaration... 
         extern size_t iconv (iconv_t cd, char * *inbuf, size_t *inbytesleft, char * *outbuf, size_t *outbytesleft);
checking for library containing waddstr... -lncurses
checking for library containing tgetent... none required
checking size of unsigned long long... 8
checking size of unsigned long... 8
checking size of unsigned __int128... 16
checking for library containing dlopen... none required
checking whether to use expat... yes
checking for libexpat... yes
checking how to link with libexpat... -lexpat
checking for XML_StopParser... yes
checking for libgmp... no
configure: error: GMP is missing or unusable
make[2]: *** [configure-gdb] Error 1
make[1]: *** [all] Error 2
make: *** [stamps/build-gdb-newlib] Error 2
```

howjmay, 2023-10-25:
I add `--disable-gdb` to bypass the aformentioned error. No error showed this time. But riscv64-unknown-elf is not installed  too.

```
Making install in c++98
make[10]: Nothing to be done for `install-exec-am'.
make[10]: Nothing to be done for `install-data-am'.
Making install in c++11
make[10]: Nothing to be done for `install-exec-am'.
make[10]: Nothing to be done for `install-data-am'.
Making install in c++17
make[10]: Nothing to be done for `install-exec-am'.
make[10]: Nothing to be done for `install-data-am'.
Making install in c++20
make[10]: Nothing to be done for `install-exec-am'.
make[10]: Nothing to be done for `install-data-am'.
Making install in c++23
make[10]: Nothing to be done for `install-exec-am'.
make[10]: Nothing to be done for `install-data-am'.
Making install in libbacktrace
make[10]: Nothing to be done for `install-exec-am'.
make[10]: Nothing to be done for `install-data-am'.
Making install in experimental
 /Users/hauyang/Work/riscv/riscv-gnu-toolchain/gcc-master/libstdc++-v3/../install-sh -c -d '/Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64'
 /bin/sh ../../libtool   --mode=install /usr/bin/install -c   libstdc++exp.la '/Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64'
libtool: install: /usr/bin/install -c .libs/libstdc++exp.lai /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64/libstdc++exp.la
libtool: install: /usr/bin/install -c .libs/libstdc++exp.a /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64/libstdc++exp.a
libtool: install: chmod 644 /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64/libstdc++exp.a
libtool: install: /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/bin/ranlib --plugin /Users/hauyang/Work/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/liblto_plugin.so --plugin /Users/hauyang/Work/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/liblto_plugin.so /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64/libstdc++exp.a
----------------------------------------------------------------------
Libraries have been installed in:
   /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the `-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the `LD_RUN_PATH' environment variable
     during linking
   - use the `-Wl,-rpath -Wl,LIBDIR' linker flag

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
make[10]: Nothing to be done for `install-data-am'.
 /Users/hauyang/Work/riscv/riscv-gnu-toolchain/gcc-master/libstdc++-v3/../install-sh -c -d '/Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64'
 /bin/sh ../libtool   --mode=install /usr/bin/install -c   libstdc++.la '/Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64'
libtool: install: /usr/bin/install -c .libs/libstdc++.lai /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64/libstdc++.la
libtool: install: /usr/bin/install -c .libs/libstdc++.a /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64/libstdc++.a
libtool: install: chmod 644 /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64/libstdc++.a
libtool: install: /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/bin/ranlib --plugin /Users/hauyang/Work/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/liblto_plugin.so --plugin /Users/hauyang/Work/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/liblto_plugin.so /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64/libstdc++.a
----------------------------------------------------------------------
Libraries have been installed in:
   /Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/lib/rv64imac/lp64

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the `-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the `LD_RUN_PATH' environment variable
     during linking
   - use the `-Wl,-rpath -Wl,LIBDIR' linker flag

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
make[10]: Nothing to be done for `install-data-am'.
Making install in doc
make[9]: Nothing to be done for `install-exec-am'.
make[9]: Nothing to be done for `install-data-am'.
Making install in po
[... 202 lines cut ...]
----------------------------------------------------------------------
make[10]: Nothing to be done for `install-data-am'.
Making install in doc
make[9]: Nothing to be done for `install-exec-am'.
make[9]: Nothing to be done for `install-data-am'.
Making install in po
make[9]: Nothing to be done for `install-exec-am'.
[... cut ...] iscv-gnu-toolchain/build/riscv64-unknown-elf/bin/ranlib" "NM=/Users/hauyang/Work/riscv-gnu-toolchain/build-gcc-newlib-stage2/./gcc/nm" "NM_FOR_BUILD=" "NM_FOR_TARGET=/Users/hauyang/Work/riscv/riscv-gnu-toolchain/build/riscv64-unknown-elf/bin/nm" "DESTDIR=" "WERROR=" DO=install multi-do # /Library/Developer/CommandLineTools/usr/bin/make
make[9]: Nothing to be done for `install-data-am'.
make[5]: Nothing to be done for `install-data-am'.
mkdir -p stamps/ && touch stamps/build-gcc-newlib-stage2
```

TommyMurphyTM1234, 2023-10-26:
To do a clean build/install you should do `make distclean` and remove any previously installed toolchain (e.g. `rm -rf <directory>` assuming that `<directory>` contains only the installed RISC-V toolchain and not also other stuff.

howjmay, 2023-10-27:
Thank you for the advice. I finally solved the problem. I need to disable-gdb to bypass the libgmp error, which I have no idea to solve. And the compilation will be correct, just I need to export PATH

closed 2023-10-27 by howjmay