riscv-collab/riscv-gnu-toolchain #1360: No permission to write into directory

Aaronyap2002, 2023-10-26:
I have created a new folder in the home directory name `new`, but it still says it doesnt have permission to write. How do i solve this?
These are the commands i ran.
`cd new
sudo apt-get install autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev
git clone https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain/
/configure --prefix=$home/new
make linux
`
Error:
`mkdir -p /.test || \
	(echo "Sorry, you don't have permission to write to" \
	 "''. Please make sure that the location is " \
	 "writable or use --prefix to specify another destination.'" \
	 && exit 1)
mkdir: cannot create directory ‘/.test’: Permission denied
Sorry, you don't have permission to write to ''. Please make sure that the location is  writable or use --prefix to specify another destination.'
make: *** [Makefile:221: stamps/check-write-permission] Error 1
`

TommyMurphyTM1234, 2023-10-26:
> I have created a new folder in the home directory name `new`, but it still says it doesnt have permission to write. How do i solve this? 

As I already said, you need to take this up with the lowRISC community. It has nothing to do with `riscv-gnu-toolchain`.

TommyMurphyTM1234, 2023-10-26:
> Error:
> 
> ```
> mkdir -p /.test || \

You're working in the root directory.
That seems very odd and probably inadvisable.
But if you really want to work there then you need `root/sudo` privileges.

closed 2023-10-26 by TommyMurphyTM1234

Aaronyap2002, 2023-10-27:
But I am indeed in the home directory. But for some reason, that code ran at root directory. 
[image]

Aaronyap2002, 2023-10-27:
But I am indeed in the home directory. But for some reason, that code ran at root directory. And after I ran `sudo make linux` as you suggested, it produces a lot of these lines of code, which i dont understand
[image]

TommyMurphyTM1234, 2023-10-27:
> ```
> /configure --prefix=$home/new
> ```

This is running `configure` from root `/`.
What does `echo $home` return?
If it is blank then this resolves to `/new` which is a directory off root requiring `root/sudo` permissions to modify.
I presume that you actually meant `--prefix=$HOME/new` which would resolve to `/home/aaron/new`.

* https://afni.nimh.nih.gov/pub/dist/edu/data/CD.expanded/AFNI_data6/unix_tutorial/basic/basic_01.ls.html

If that doesn't help then you need to clearly state all of the commands that you're using (`git clone`, `configure`, `make` etc.) and capture, compress and attach the full build log (e.g. `make linux 2>&1 | tee build.log` then compress and attach it).

Aaronyap2002, 2023-12-06:
Hi sorry for the late reply. I was busy with examinations. 
Running `echo $HOME` gives me `/home/aaron`. So I changed the prefix from  `/configure --prefix=$home/new` to `./configure --prefix=/home/aaron/new` and it seems like there's no error. 
However, it seems like it is stuck cloning into a directory. It has attempting to clone for hours and i am not sure if it is normal or not.

`Libraries have been installed in:
   /home/aaron/new/riscv64-unknown-linux-gnu/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the `-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the `LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the `LD_RUN_PATH' environment variable
     during linking
   - use the `-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to `/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
make[5]: Nothing to be done for 'install-data-am'.
make[5]: Leaving directory '/home/aaron/new/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic'
make[4]: Leaving directory '/home/aaron/new/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic'
make[3]: Leaving directory '/home/aaron/new/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic'
make[2]: Leaving directory '/home/aaron/new/riscv-gnu-toolchain/build-gcc-linux-stage2'
make[1]: Leaving directory '/home/aaron/new/riscv-gnu-toolchain/build-gcc-linux-stage2'
cp -a /home/aaron/new/riscv64-unknown-linux-gnu/lib* /home/aaron/new/sysroot
mkdir -p stamps/ && touch stamps/build-gcc-linux-stage2
cd /home/aaron/new/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /home/aaron/new/riscv-gnu-toolchain/gdb/ && \
flock `git rev-parse --git-dir`/config git submodule update /home/aaron/new/riscv-gnu-toolchain/gdb/
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Cloning into '/home/aaron/new/riscv-gnu-toolchain/gdb'...
`

TommyMurphyTM1234, 2023-12-06:
> However, it seems like it is stuck cloning into a directory. It has been attempting to clone for hours and i am not sure if it is normal or not.

When that happens it's often a connectivity issue from your machine.
Try a recursive clone into a new directory to see what happens:
```
cd ~/Downloads
 git clone --recursive --verbose --progress https://github.com/riscv-collab/riscv-gnu-toolchain clone-test
```

Aaronyap2002, 2023-12-06:
It seems fine. Not sure if i should let it download completly or not.

```
aaron@aaron-TravelMate-P238-M:~$ git clone --recursive --verbose --progress https://github.com/riscv-collab/riscv-gnu-toolchain clone-test
Cloning into 'clone-test'...
POST git-upload-pack (gzip 4265 to 2126 bytes)
remote: Enumerating objects: 9099, done.
remote: Counting objects: 100% (605/605), done.
remote: Compressing objects: 100% (547/547), done.
remote: Total 9099 (delta 64), reused 562 (delta 51), pack-reused 8494
Receiving objects: 100% (9099/9099), 6.66 MiB | 11.69 MiB/s, done.
Resolving deltas: 100% (4307/4307), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/aaron/clone-test/binutils'...
remote: Enumerating objects: 1194146, done.
remote: Counting objects: 100% (51138/51138), done.
remote: Compressing objects: 100% (43682/43682), done.
Receiving objects:   1% (23854/1194146), 10.17 MiB | 22.00 KiB/s
```

TommyMurphyTM1234, 2023-12-06:
This is what I get - I'm accessing the repos from Dublin, Ireland via 500Mbps home cable broadband:
```
git --version
git version 2.34.1

cat /etc/lsb-release /etc/os-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.3 LTS"
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

uname -a
Linux magpie 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

time git clone --recursive --verbose --progress https://github.com/riscv-collab/riscv-gnu-toolchain clone-test
Cloning into 'clone-test'...
POST git-upload-pack (175 bytes)
POST git-upload-pack (gzip 4302 to 2107 bytes)
remote: Enumerating objects: 9099, done.
remote: Counting objects: 100% (605/605), done.
remote: Compressing objects: 100% (547/547), done.
remote: Total 9099 (delta 64), reused 561 (delta 51), pack-reused 8494
Receiving objects: 100% (9099/9099), 6.64 MiB | 1.30 MiB/s, done.
Resolving deltas: 100% (4308/4308), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/user/Downloads/clone-test/binutils'...
remote: Enumerating objects: 1194146, done.
remote: Counting objects: 100% (51138/51138), done.
remote: Compressing objects: 100% (43682/43682), done.
remote: Total 1194146 (delta 12832), reused 10137 (delta 7352), pack-reused 1143008
Receiving objects: 100% (1194146/1194146), 446.40 MiB | 9.20 MiB/s, done.
Resolving deltas: 100% (990204/990204), done.
Cloning into '/home/user/Downloads/clone-test/dejagnu'...
remote: Counting objects: 9027, done.
remote: Compressing objects: 100% (2009/2009), done.
remote: Total 9027 (delta 6898), reused 9019 (delta 6890)
Receiving objects: 100% (9027/9027), 4.66 MiB | 5.16 MiB/s, done.
Resolving deltas: 100% (6898/6898), done.
Cloning into '/home/user/Downloads/clone-test/gcc'...
remote: Enumerating objects: 2953964, done.
remote: Counting objects: 100% (213771/213771), done.
remote: Compressing objects: 100% (17497/17497), done.
remote: Total 2953964 (delta 206423), reused 198784 (delta 196215), pack-reused 2740193
Receiving objects: 100% (2953964/2953964), 1.13 GiB | 9.52 MiB/s, done.
Resolving deltas: 100% (2429100/2429100), done.
Cloning into '/home/user/Downloads/clone-test/gdb'...
remote: Enumerating objects: 1194146, done.
remote: Counting objects: 100% (51138/51138), done.
[... 262 lines cut ...]
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/wycheproof': checked out '2196000605e45d91097147c9c71f26b72af58003'
Submodule path 'qemu/roms/edk2/MdeModulePkg/Library/BrotliCustomDecompressLib/brotli': checked out 'f4153a09f87cbb9c826d8fc12c74642bb2d879ea'
Submodule path 'qemu/roms/edk2/MdeModulePkg/Universal/RegularExpressionDxe/oniguruma': checked out 'abfc8ff81df4067f309032467785e06975678f0d'
Submodule path 'qemu/roms/edk2/RedfishPkg/Library/JsonLib/jansson': checked out 'e9ebfa7e77a6bee77df44e096b100e7131044059'
Submodule path 'qemu/roms/edk2/UnitTestFrameworkPkg/Library/CmockaLib/cmocka': checked out '1cc9cde3448cdd2e000886a26acf1caac2db7cf1'
Submodule path 'qemu/roms/edk2/UnitTestFrameworkPkg/Library/GoogleTestLib/googletest': checked out '86add13493e5c881d7e4ba77fb91c1f57752b3a4'
Submodule path 'qemu/roms/ipxe': checked out '4bd064de239dab2426b31c9789a1f4d78087dc63'
Submodule path 'qemu/roms/openbios': checked out 'af97fd7af5e7c18f591a7b987291d3db4ffb28b5'
Submodule path 'qemu/roms/opensbi': checked out '057eb10b6d523540012e6947d5c9f63e95244e94'
Submodule path 'qemu/roms/qboot': checked out '8ca302e86d685fa05b16e2b208888243da319941'
Submodule path 'qemu/roms/qemu-palcode': checked out '99d9b4dcf27d7fbcbadab71bdc88ef6531baf6bf'
Submodule path 'qemu/roms/seabios': checked out 'ea1b7a0733906b8425d948ae94fba63c32b1d425'
Submodule path 'qemu/roms/seabios-hppa': checked out '673d2595d4f773cc266cbf8dbaf2f475a6adb949'
Submodule path 'qemu/roms/skiboot': checked out '24a7eb35966d93455520bc2debdd7954314b638b'
Submodule path 'qemu/roms/u-boot': checked out '840658b093976390e9537724f802281c9c8439f5'
Submodule path 'qemu/roms/u-boot-sam460ex': checked out '60b3916f33e617a815973c5a6df77055b2e3a588'
Submodule path 'qemu/roms/vbootrom': checked out '0c37a43527f0ee2b9584e7fb2fdc805e902635ac'
Submodule path 'qemu/tests/lcitool/libvirt-ci': checked out '9bff3b763b5531a1490e238bfbf77306dc3a6dbb'
Submodule path 'spike': checked out 'd1efcdffffee57bab0fdbd2b377c6132b37556fd'

real    39m57.041s
user    45m11.156s
sys     5m15.917s

```

TommyMurphyTM1234, 2023-12-06:
Your checkout:
```
Cloning into '/home/aaron/clone-test/binutils'...
remote: Enumerating objects: 1194146, done.
remote: Counting objects: 100% (51138/51138), done.
remote: Compressing objects: 100% (43682/43682), done.
Receiving objects:   1% (23854/1194146), 10.17 MiB | 22.00 KiB/s
```
seems a lot slower than mine:
```
Cloning into '/home/user/Downloads/clone-test/binutils'...
remote: Enumerating objects: 1194146, done.
remote: Counting objects: 100% (51138/51138), done.
remote: Compressing objects: 100% (43682/43682), done.
remote: Total 1194146 (delta 12832), reused 10137 (delta 7352), pack-reused 1143008
Receiving objects: 100% (1194146/1194146), 446.40 MiB | 9.20 MiB/s, done.
Resolving deltas: 100% (990204/990204), done.
```

Aaronyap2002, 2023-12-06:
I am currently in my school's lab. The wifi speed here is actually quite fast for general uses with speeds reaching 500mbps and its good enough for normal usage. Should i stop the cloning process and run `make linux` again?

TommyMurphyTM1234, 2023-12-06:
> The wifi speed here is actually quite fast for general uses 

I would always prefer wired ethernet over wifi.

> Should i stop the cloning process and run `make linux` again?

Probably - it looks like your connectivity to the repos is simply very slow regardless of the LAN speed.

cmuellner, 2023-12-06:
> > If access to the official Binutils git repo does not work for whatever reason, you may consider a mirror of that. E.g. clone `https://github.com/bminor/binutils-gdb` locally, checkout the version that you want to build, configure this repo with `--with-binutils-src=path/to/binutils` and `--with-gdb-src=path/to/binutils`.
> 
> I thought that it wasn't just one repo that was causing issues here though?

I saw the following in one of the logs above when cloning our repo:

> 6.66 MiB | 11.69 MiB/s

This could be a sign that GitHub repos can be cloned faster.

Aaronyap2002, 2023-12-17:
[terminal output.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13695171/terminal.output.log)

Here's the terminal output file on the computer.
Before I resort to the solution you suggested, I tried running `sudo make linux` again and this time it seems it went successful this time. The terminal ended cleanly, and running the command gives off `make: Nothing to be done for 'linux'` 

Does this mean I have successfully built the compiler?

TommyMurphyTM1234, 2023-12-17:
> Does this mean I have successfully built the compiler?

Not necessarily.
You need to check the prefix directory to see what's in it.
To be sure you need to do a clean build:
```
rm -rf /home/aaron/new
make distclean
make linux
```

TommyMurphyTM1234, 2023-12-17:
BTW - I have also been finding the upstream repos to be very slow lately.
I'll post actual speeds later.

Aaronyap2002, 2023-12-17:
```rm -rf /home/aaron/new
make distclean
make linux
```
I felt like I did something wrong with these commands. I ran "make linux" in" /home/aaron/new/ riscv-gnu-toolchain". If i were to run "rm -rf /home/aaron/new", wont that remove my installed files too?

TommyMurphyTM1234, 2023-12-17:
I was assuming that `/home/aaron/new` was the directory that you specified at configuration time - i.e. `./configure --prefix=/home/aaron/new ...`. If it's not then you need to delete whatever is your `prefix` directory and then do `make distclean && make linux` to do a fresh build of the toolchain and check that this succeeds.

Aaronyap2002, 2023-12-17:
That means if i am going to built it in `/home/aaron/new/`, i will run `/configure --prefix=/home/aaron/new/(a-new-directory)` ?

TommyMurphyTM1234, 2023-12-17:
I don't really understand your question.

The bottom line is that if you want to do a clean build then you need to delete the previously built toolchain directory or run `configure` again with a different `--prefix=` directory and then run `make distclean && make linux`.

TommyMurphyTM1234, 2023-12-18:
> BTW - I have also been finding the upstream repos to be very slow lately. I'll post actual speeds later.

Took most of the day to do a recursive clone even though it's a 500Mbps cable broadband connection and all other connectivity works fine... :-|
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain
Cloning into 'riscv-gnu-toolchain'...
remote: Enumerating objects: 9127, done.
remote: Counting objects: 100% (633/633), done.
remote: Compressing objects: 100% (563/563), done.
remote: Total 9127 (delta 83), reused 580 (delta 63), pack-reused 8494
Receiving objects: 100% (9127/9127), 6.67 MiB | 1.31 MiB/s, done.
Resolving deltas: 100% (4326/4326), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/binutils'...
remote: Enumerating objects: 1195611, done.
remote: Counting objects: 100% (52603/52603), done.
remote: Compressing objects: 100% (44865/44865), done.
remote: Total 1195611 (delta 13896), reused 10479 (delta 7626), pack-reused 1143008
Receiving objects: 100% (1195611/1195611), 447.71 MiB | 46.00 KiB/s, done.
Resolving deltas: 100% (991268/991268), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/dejagnu'...
remote: Counting objects: 9058, done.
remote: Compressing objects: 100% (2040/2040), done.
remote: Total 9058 (delta 6922), reused 9013 (delta 6890)
Receiving objects: 100% (9058/9058), 4.70 MiB | 4.95 MiB/s, done.
Resolving deltas: 100% (6922/6922), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/gcc'...
remote: Enumerating objects: 2959703, done.
remote: Counting objects: 100% (209866/209866), done.
remote: Compressing objects: 100% (13771/13771), done.
remote: Total 2959703 (delta 205074), reused 196911 (delta 196057), pack-reused 2749837
Receiving objects: 100% (2959703/2959703), 1.12 GiB | 45.00 KiB/s, done.
Resolving deltas: 100% (2434987/2434987), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/gdb'...
remote: Enumerating objects: 1195611, done.
remote: Counting objects: 100% (52603/52603), done.
remote: Compressing objects: 100% (44865/44865), done.
remote: Total 1195611 (delta 13894), reused 10480 (delta 7626), pack-reused 1143008
Receiving objects: 100% (1195611/1195611), 447.74 MiB | 9.21 MiB/s, done.
Resolving deltas: 100% (991266/991266), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/glibc'...
remote: Enumerating objects: 706552, done.
remote: Counting objects: 100% (28222/28222), done.
remote: Compressing objects: 100% (9847/9847), done.
remote: Total 706552 (delta 21013), reused 20865 (delta 17989), pack-reused 678330
Receiving objects: 100% (706552/706552), 231.96 MiB | 46.00 KiB/s, done.
Resolving deltas: 100% (601738/601738), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/llvm'...
remote: Enumerating objects: 5724621, done.
remote: Counting objects: 100% (346/346), done.
remote: Compressing objects: 100% (98/98), done.
remote: Total 5724621 (delta 263), reused 263 (delta 248), pack-reused 5724275
Receiving objects: 100% (5724621/5724621), 2.29 GiB | 10.31 MiB/s, done.
Resolving deltas: 100% (4679662/4679662), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/musl'...
remote: warning: unable to access '/root/.config/git/attributes': Permission denied
remote: Enumerating objects: 37317, done.
remote: Counting objects: 100% (37317/37317), done.
remote: Compressing objects: 100% (16197/16197), done.
remote: Total 37317 (delta 27297), reused 28543 (delta 20715), pack-reused 0
[... 233 lines cut ...]
Receiving objects: 100% (4401/4401), 35.56 MiB | 3.15 MiB/s, done.
Resolving deltas: 100% (2663/2663), done.
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/boringssl': checked out '2070f8ad9151dc8f3a73bffaa146b5e6937a583f'
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/krb5': checked out '890ca2f401924cdcb88f493950b04700bbe52db3'
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/pyca-cryptography': checked out 'fa84d185c0fefa7d612759e02a26def2f948fe34'
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/wycheproof': checked out '2196000605e45d91097147c9c71f26b72af58003'
Submodule path 'qemu/roms/edk2/MdeModulePkg/Library/BrotliCustomDecompressLib/brotli': checked out 'f4153a09f87cbb9c826d8fc12c74642bb2d879ea'
Submodule path 'qemu/roms/edk2/MdeModulePkg/Universal/RegularExpressionDxe/oniguruma': checked out 'abfc8ff81df4067f309032467785e06975678f0d'
[... cut ...]
Submodule path 'qemu/roms/skiboot': checked out '24a7eb35966d93455520bc2debdd7954314b638b'
Submodule path 'qemu/roms/u-boot': checked out '840658b093976390e9537724f802281c9c8439f5'
Submodule path 'qemu/roms/u-boot-sam460ex': checked out '60b3916f33e617a815973c5a6df77055b2e3a588'
Submodule path 'qemu/roms/vbootrom': checked out '0c37a43527f0ee2b9584e7fb2fdc805e902635ac'
Submodule path 'qemu/tests/lcitool/libvirt-ci': checked out '9bff3b763b5531a1490e238bfbf77306dc3a6dbb'
Submodule path 'spike': checked out 'd1efcdffffee57bab0fdbd2b377c6132b37556fd'
```

Aaronyap2002, 2023-12-18:
I will proceed to use the built compiler to build the IBEX simulation, for now, and will rebuild it before i sleep.

TommyMurphyTM1234, 2023-12-19:
> Took most of the day to do a recursive clone even though it's a 500Mbps cable broadband connection and all other connectivity works fine... :-|
>
> ```
> git clone --recursive [...]

Tried this again today - c. 4 hours wall clock time to do a recursive clone! :-(
```
time git clone --verbose --progress --recursive https://github.com/riscv-collab/riscv-gnu-toolchain test-clone
Cloning into 'test-clone'...
POST git-upload-pack (175 bytes)
POST git-upload-pack (gzip 4402 to 2155 bytes)
remote: Enumerating objects: 9135, done.
remote: Counting objects: 100% (641/641), done.
remote: Compressing objects: 100% (569/569), done.
remote: Total 9135 (delta 86), reused 584 (delta 65), pack-reused 8494
Receiving objects: 100% (9135/9135), 6.65 MiB | 1.28 MiB/s, done.
Resolving deltas: 100% (4329/4329), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/user/Downloads/test-clone/binutils'...
remote: Enumerating objects: 1195731, done.
remote: Counting objects: 100% (52723/52723), done.
remote: Compressing objects: 100% (44985/44985), done.
Receiving objects: 100% (1195731/1195731), 447.79 MiB | 48.00 KiB/s, done.
remote: Total 1195731 (delta 13988), reused 10478 (delta 7626), pack-reused 1143008
Resolving deltas: 100% (991360/991360), done.
Cloning into '/home/user/Downloads/test-clone/dejagnu'...
remote: Counting objects: 9058, done.
remote: Compressing objects: 100% (2040/2040), done.
remote: Total 9058 (delta 6922), reused 9013 (delta 6890)
Receiving objects: 100% (9058/9058), 4.70 MiB | 4.57 MiB/s, done.
Resolving deltas: 100% (6922/6922), done.
Cloning into '/home/user/Downloads/test-clone/gcc'...
remote: Enumerating objects: 2960602, done.
remote: Counting objects: 100% (210765/210765), done.
remote: Compressing objects: 100% (14641/14641), done.
remote: Total 2960602 (delta 205557), reused 196947 (delta 196082), pack-reused 2749837
Receiving objects: 100% (2960602/2960602), 1.13 GiB | 10.76 MiB/s, done.
Resolving deltas: 100% (2435470/2435470), done.
Cloning into '/home/user/Downloads/test-clone/gdb'...
remote: Enumerating objects: 1195870, done.
remote: Counting objects: 100% (52862/52862), done.
remote: Compressing objects: 100% (45046/45046), done.
remote: Total 1195870 (delta 14074), reused 10603 (delta 7704), pack-reused 1143008
Receiving objects: 100% (1195870/1195870), 448.00 MiB | 9.02 MiB/s, done.
Resolving deltas: 100% (991446/991446), done.
Cloning into '/home/user/Downloads/test-clone/glibc'...
remote: Enumerating objects: 706596, done.
remote: Counting objects: 100% (28266/28266), done.
remote: Compressing objects: 100% (9036/9036), done.
remote: Total 706596 (delta 21067), reused 22207 (delta 18843), pack-reused 678330
Receiving objects: 100% (706596/706596), 231.97 MiB | 7.14 MiB/s, done.
Resolving deltas: 100% (601792/601792), done.
Cloning into '/home/user/Downloads/test-clone/llvm'...
remote: Enumerating objects: 5727457, done.
remote: Counting objects: 100% (1514/1514), done.
remote: Compressing objects: 100% (663/663), done.
remote: Total 5727457 (delta 1047), reused 1034 (delta 849), pack-reused 5725943
Receiving objects: 100% (5727457/5727457), 2.02 GiB | 10.99 MiB/s, done.
Resolving deltas: 100% (4667528/4667528), done.
Cloning into '/home/user/Downloads/test-clone/musl'...
[... 247 lines cut ...]
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/pyca-cryptography': checked out 'fa84d185c0fefa7d612759e02a26def2f948fe34'
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/wycheproof': checked out '2196000605e45d91097147c9c71f26b72af58003'
Submodule path 'qemu/roms/edk2/MdeModulePkg/Library/BrotliCustomDecompressLib/brotli': checked out 'f4153a09f87cbb9c826d8fc12c74642bb2d879ea'
Submodule path 'qemu/roms/edk2/MdeModulePkg/Universal/RegularExpressionDxe/oniguruma': checked out 'abfc8ff81df4067f309032467785e06975678f0d'
Submodule path 'qemu/roms/edk2/RedfishPkg/Library/JsonLib/jansson': checked out 'e9ebfa7e77a6bee77df44e096b100e7131044059'
Submodule path 'qemu/roms/edk2/UnitTestFrameworkPkg/Library/CmockaLib/cmocka': checked out '1cc9cde3448cdd2e000886a26acf1caac2db7cf1'
Submodule path 'qemu/roms/edk2/UnitTestFrameworkPkg/Library/GoogleTestLib/googletest': checked out '86add13493e5c881d7e4ba77fb91c1f57752b3a4'
Submodule path 'qemu/roms/ipxe': checked out '4bd064de239dab2426b31c9789a1f4d78087dc63'
Submodule path 'qemu/roms/openbios': checked out 'af97fd7af5e7c18f591a7b987291d3db4ffb28b5'
Submodule path 'qemu/roms/opensbi': checked out '057eb10b6d523540012e6947d5c9f63e95244e94'
[... cut ...]
Submodule path 'qemu/roms/u-boot': checked out '840658b093976390e9537724f802281c9c8439f5'
Submodule path 'qemu/roms/u-boot-sam460ex': checked out '60b3916f33e617a815973c5a6df77055b2e3a588'
Submodule path 'qemu/roms/vbootrom': checked out '0c37a43527f0ee2b9584e7fb2fdc805e902635ac'
Submodule path 'qemu/tests/lcitool/libvirt-ci': checked out '9bff3b763b5531a1490e238bfbf77306dc3a6dbb'
Submodule path 'spike': checked out 'd1efcdffffee57bab0fdbd2b377c6132b37556fd'

real    232m52.783s
user    43m34.960s
sys     3m50.597s
```