riscv-collab/riscv-gnu-toolchain #1361: How to build sanitizer lib for riscv

zqb-all, 2023-11-02:
There is a configuration item -`-enable-libsanitizer`  in the configure file.
when I try to use it, `make linux` fail with
```
./../../../../gcc/libsanitizer/sanitizer_common/sanitizer_platform_limits_posix.cpp:180:10: fatal error: crypt.h: No such file or directory
  180 | #include <crypt.h>
      |          ^~~~~~~~~
compilation terminated.
```
Should I fix it by adding -I to specify glibc's directory?
When I tried to do this, there were more problems with the later compilation.
So I thought maybe I missed some steps, can someone tell me the correct way to compile libsanitizer for riscv?

TommyMurphyTM1234, 2023-11-02:
Please clarify your `git clone`, `configure` and `make` commands used to build the toolchain and capture, compress and attach your build log (e.g. `make ... 2>&1 | tee build.log`) and `config.log`.

TommyMurphyTM1234, 2023-11-02:
This is not specifically RISC-V related (it's logged against Arm) but perhaps it's relevant?

* https://gcc.gnu.org/bugzilla//show_bug.cgi?id=111057

TommyMurphyTM1234, 2023-11-02:
> This is not specifically RISC-V related (it's logged against Arm) but perhaps it's relevant?
> 
> * https://gcc.gnu.org/bugzilla//show_bug.cgi?id=111057

Perhaps this will work?
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain issue-1361
cd issue-1361
git clone https://github.com/bminor/glibc glibc-master -b master
./configure --prefix=`pwd`/installed-tools --enable-libsanitizer --with-glibc-src=`pwd`/glibc-master
make linux 2>&1 | tee build.log
```

TommyMurphyTM1234, 2023-11-02:
> Perhaps this will work?
>
> ```
> git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain issue-1361
> cd issue-1361
> git clone https://github.com/bminor/glibc glibc-master -b master [...]

OK - that didn't help:
```
libtool: compile:  /home/user/Downloads/issue-1361/build-gcc-linux-stage2/./gcc/xgcc -shared-libgcc -B/home/user/Downloads/issue-1361/build-gcc-linux-stage2/./gcc -nostdinc++ -L/home/user/Downloads/issue-1361/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libstdc++-v3/src -L/home/user/Downloads/issue-1361/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libstdc++-v3/src/.libs -L/home/user/Downloads/issue-1361/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libstdc++-v3/libsupc++/.libs -B/home/user/Downloads/issue-1361/installed-tools/riscv64-unknown-linux-gnu/bin/ -B/home/user/Downloads/issue-1361/installed-tools/riscv64-unknown-linux-gnu/lib/ -isystem /home/user/Downloads/issue-1361/installed-tools/riscv64-unknown-linux-gnu/include -isystem /home/user/Downloads/issue-1361/installed-tools/riscv64-unknown-linux-gnu/sys-include -D_GNU_SOURCE -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -DHAVE_RPC_XDR_H=0 -DHAVE_TIRPC_RPC_XDR_H=0 -I. -I../../../.././gcc/libsanitizer/sanitizer_common -I.. -I ../../../.././gcc/libsanitizer/include -I ../../../.././gcc/libsanitizer -isystem ../../../.././gcc/libsanitizer/include/system -Wall -W -Wno-unused-parameter -Wwrite-strings -pedantic -Wno-long-long -fPIC -fno-builtin -fno-exceptions -fno-rtti -fomit-frame-pointer -funwind-tables -fvisibility=hidden -Wno-variadic-macros -I../../libstdc++-v3/include -I../../libstdc++-v3/include/riscv64-unknown-linux-gnu -I../../../.././gcc/libsanitizer/../libstdc++-v3/libsupc++ -std=gnu++14 -DSANITIZER_LIBBACKTRACE -DSANITIZER_CP_DEMANGLE -I ../../../.././gcc/libsanitizer/../libbacktrace -I ../libbacktrace -I ../../../.././gcc/libsanitizer/../include -include ../../../.././gcc/libsanitizer/libbacktrace/backtrace-rename.h -O2 -mcmodel=medlow -D_GNU_SOURCE -MT sanitizer_platform_limits_posix.lo -MD -MP -MF .deps/sanitizer_platform_limits_posix.Tpo -c ../../../.././gcc/libsanitizer/sanitizer_common/sanitizer_platform_limits_posix.cpp  -fPIC -DPIC -o .libs/sanitizer_platform_limits_posix.o
../../../.././gcc/libsanitizer/sanitizer_common/sanitizer_platform_limits_posix.cpp:180:10: fatal error: crypt.h: No such file or directory
  180 | #include <crypt.h>
      |          ^~~~~~~~~
compilation terminated.
make[5]: *** [Makefile:614: sanitizer_platform_limits_posix.lo] Error 1
make[5]: Leaving directory '/home/user/Downloads/issue-1361/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libsanitizer/sanitizer_common'
make[4]: *** [Makefile:532: all-recursive] Error 1
make[4]: Leaving directory '/home/user/Downloads/issue-1361/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libsanitizer'
make[3]: *** [Makefile:419: all] Error 2
make[3]: Leaving directory '/home/user/Downloads/issue-1361/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libsanitizer'
make[2]: *** [Makefile:12822: all-target-libsanitizer] Error 2
make[2]: Leaving directory '/home/user/Downloads/issue-1361/build-gcc-linux-stage2'
make[1]: *** [Makefile:1040: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/issue-1361/build-gcc-linux-stage2'
make: *** [Makefile:458: stamps/build-gcc-linux-stage2] Error 2
```
I guess that this patch needs to be applied manually for now?

* https://github.com/llvm/llvm-project/commit/d7bead833631486e337e541e692d9b4a1ca14edd

and the issue will presumably be addressed in glibc 2.39 in 2024?

* https://www.gnu.org/software/libc/#:~:text=The%20current%20development%20version%20of%20glibc%20is%202.39%2C%20releasing%20on%20or%20around%20February%201st%2C%202024.

The crosstool-NG Issue about this is probably also relevant:

* https://github.com/crosstool-ng/crosstool-ng/issues/2010

zqb-all, 2023-11-03:
> I guess that this patch needs to be applied manually for now?
> 
> https://github.com/llvm/llvm-project/commit/d7bead833631486e337e541e692d9b4a1ca14edd

Thank you for your quick reply. I will try it.

zqb-all, 2023-11-03:
@TommyMurphyTM1234  Thanks, with https://github.com/llvm/llvm-project/commit/d7bead833631486e337e541e692d9b4a1ca14edd  , now I can build libsanitizer.

TommyMurphyTM1234, 2023-11-03:
> @TommyMurphyTM1234 Thanks, with [llvm/llvm-project@d7bead8](https://github.com/llvm/llvm-project/commit/d7bead833631486e337e541e692d9b4a1ca14edd) , now I can build libsanitizer.

Thanks for the update. Since the issue is with the upstream glibc and is will presumably be addressed there in the 2.39 release, I'll close this issue here.