riscv-collab/riscv-gnu-toolchain #1362: Latest version fails on Linux build

drmpeg, 2023-11-07:
The latest version (tag 2023.10.18) fails when building Linux 6.1.61. The error is:

/tmp/cc21w69r.s: Assembler messages:
/tmp/cc21w69r.s: Fatal error: Infinite loop encountered whilst attempting to compute the addresses of symbols in section .text
make[3]: *** [scripts/Makefile.build:250: drivers/target/target_core_pr.o] Error 1
make[2]: *** [scripts/Makefile.build:500: drivers/target] Error 2
make[1]: *** [scripts/Makefile.build:500: drivers] Error 2
make: *** [Makefile:2014: .] Error 2

This does not occur with the previous tag 2023.10.06.

Steps to reproduce:

```git clone https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git```
```cd linux```
```git checkout linux-6.1.y```
```cp ../config-5.19.0-1004-generic-sifive ./.config```
```make CROSS_COMPILE=riscv64-unknown-linux-gnu- ARCH=riscv "KCFLAGS=-mtune=sifive-7-series -falign-functions=8 -falign-jumps=8 -falign-loops=8" LOCALVERSION= olddefconfig```
```make -s CROSS_COMPILE=riscv64-unknown-linux-gnu- ARCH=riscv "KCFLAGS=-mtune=sifive-7-series -falign-functions=8 -falign-jumps=8 -falign-loops=8" LOCALVERSION= -j10```

The config file can be downloaded here:

https://www.w6rz.net/config-5.19.0-1004-generic-sifive

TommyMurphyTM1234, 2023-11-07:
> The latest version (tag 2023.10.18) fails when building Linux 6.1.61. The error is:
> This does not occur with the previous tag 2023.10.06.

Just to note:

* [2023.10.18](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2023.10.18) is the latest build of the GCC 13.2.0 based toolchain
* [2023.10.06](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2023.10.06) is the last build of a GCC 12.2.0 based toolchain

drmpeg, 2023-11-07:
If I remove the ```falign-*``` flags, the problem goes away. I was using those since they work well with the Coremark benchmark, but they're not necessary.

I'll leave this open for a day or two in case someone wants to chase it down.

cmuellner, 2023-11-08:
The error is reported by GAS in `relax_segment()`.
So the more recent GCC release could emit some weird constraints that cause a too high computation overhead in the linker's address calculation code.

It would be great if you could report this in a ticket in the [GCC Bugzilla](https://gcc.gnu.org/bugzilla/).

Further steps would probably be:
* reproduce with GCC upstream/master
* minimize the source base to get a minimal reproducer

On the other hand, using these `-falign` flags on a big project like the Linux kernel clearly represents an extreme case. So this issue is likely to get stalled because it won't get a high priority.

drmpeg, 2023-11-08:
@cmuellner Thanks for the response. I agree it's pretty esoteric. Closing this issue here.

closed 2023-11-08 by drmpeg

TommyMurphyTM1234, 2023-11-08:
> The error is reported by GAS in `relax_segment()`. So the more recent GCC release could emit some weird constraints that cause a too high computation overhead in the linker's address calculation [...]

If it's an assembler issue then shouldn't it be logged against binutils rather than gcc?

* https://www.gnu.org/software/binutils/
* https://sourceware.org/bugzilla/

cmuellner, 2023-11-08:
> If it's an assembler issue then shouldn't it be logged against binutils rather than gcc?

Binutils has an upper bound of iterations in which it calculates the addresses.
So it is not a bug in Binutils, but the code generated by GCC triggers a limit that is intended to avoid run-time explosions or infinite loops.

Since this issue was introduced by a change in GCC, there is a chance that GCC could generate better code to prevent this from happening.
Therefore, a GCC ticket seems more reasonable to me.