riscv-collab/riscv-gnu-toolchain #1363: unknown relocation type 57

wangzhankun, 2023-11-09:
## description
When loading the driver, everything was normal when I compiled it with version 12.2.0 of Linux, but when I used version 13.2.0 to compile and load the driver, I encountered the following error:
```log
driver: unknown relocation type 57
```

## make
```
$(MAKE) CFLAGS_MODULE=-fno-pic -C ${linux_wrkdir} O=${linux_wrkdir} CROSS_COMPILE=${cross_compile} ARCH=riscv
    M=${driver_wrkdir} modules
```

```makefile
DRIVER = keystone-driver.ko

ifneq ($(KERNELRELEASE),)
	keystone-driver-y := \
		keystone.o \
		keystone-page.o \
		keystone-ioctl.o \
		keystone-enclave.o \
		keystone-uintr.o \
	  keystone-sbi.o
	obj-m += keystone-driver.o
else

PWD := $(shell pwd)
LINUXSRC := $(PWD)/../linux
# LINUXSRC := /root/patch-keystone/linux

default:
	$(MAKE) CFLAGS_MODULE=-fno-pic -C $(LINUXSRC) ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- M=$(PWD)

endif

clean:
	rm -rvf *.o *.ko *.order *.symvers *.mod.c .tmp_versions .*o.cmd

```

## gcc -v
### 12.2.0
```log
riscv64-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/opt/riscv-gnu-toolchain/libexec/gcc/riscv64-unknown-linux-gnu/12.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /root/riscv-gnu-toolchain/build/../gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv-gnu-toolchain --with-sysroot=/opt/riscv-gnu-toolchain/sysroot --with-pkgversion=g --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=/root/riscv-gnu-toolchain/gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (g) 
```
### 13.2.0
```log
riscv64-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/opt/riscv-gnu-toolchain/libexec/gcc/riscv64-unknown-linux-gnu/12.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /root/riscv-gnu-toolchain/build/../gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv-gnu-toolchain --with-sysroot=/opt/riscv-gnu-toolchain/sysroot --with-pkgversion=g --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=/root/riscv-gnu-toolchain/gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (g) 
```

TommyMurphyTM1234, 2023-11-09:
You don't seem to have provided full instructions on how to reproduce the issue? In particular, what `git clone`, `configure`, `make` etc. commands used to compile the keystone driver (and the Linux kernel itself?) and details of where you obtained or how you built the RISC-V toolchain. Please review your issue and ensure that all information needed to reproduce the error is provided. 

The following also doesn't make sense:

> everything was normal when I compiled it with version 12.2.0 of Linux, but when I used version 13.2.0 ...

I presume you're referring to the version of the RISC-V Linux toolchain here and not the Linux kernel?

> **13.2.0**
> ...
>```
> riscv64-unknown-linux-gnu-gcc -v
> ...
> gcc version 12.2.0 (g)
> ```

This is output from the GCC 12 toolchain in spite of the caption.

wangzhankun, 2023-11-09:
@TommyMurphyTM1234 hello, sir, I have updated my issue

wangzhankun, 2023-11-09:
UPDATE! I HAVE FORCED THE GCC 13.2.0 ISA SPEC TO `2.2`!

TommyMurphyTM1234, 2023-11-09:
> UPDATE! I HAVE FORCED THE GCC 13.2.0 ISA SPEC TO `2.2`!

Why?

TommyMurphyTM1234, 2023-11-09:
> @TommyMurphyTM1234 hello, sir, I have updated my issue

Thanks but this is a problem.

> driver: my private driver that I cannot provide.

That being the case you'll need to create a small self-contained reproducible test case that you can share. Otherwise nobody here can reproduce your problem or comment meaningfully.

wangzhankun, 2023-11-09:
> > UPDATE! I HAVE FORCED THE GCC 13.2.0 ISA SPEC TO `2.2`!
> 
> Why?

All my code is developed under gcc12.2.0 version, of which  sbi spec only supports 2.2. I don’t want to upgrade sbi spec after upgrading gcc to 13.2.0 to avoid development problems caused by inconsistent spec versions. . The reason why I want to upgrade the gcc version is because 12.2.0 does not support `std::atomic<uint16_t>::is_always_lock_free`.

I will give a sample code as soon as possible.

kito-cheng, 2023-11-09:
@wangzhankun I guess you need this:

https://lore.kernel.org/lkml/20231101-module_relocations-v9-2-8dfa3483c400@rivosinc.com/#r

TommyMurphyTM1234, 2023-11-16:
> @wangzhankun I guess you need this:
> 
> https://lore.kernel.org/lkml/20231101-module_relocations-v9-2-8dfa3483c400@rivosinc.com/#r

@wangzhankun - did this work for you?

TommyMurphyTM1234, 2023-11-21:
Closing this since there is no update and it looks like the upstream patch is what is needed.