riscv-collab/riscv-gnu-toolchain #1364: use vector enabled rv64 toolchain for non-vector ISA

yf13, 2023-11-09:
I got a rv64 toolchain supporting vector extensions as below:

```
Using built-in specs.
COLLECT_GCC=/opt/riscv/vector/bin/riscv64-unknown-linux-musl-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/vector/bin/../libexec/gcc/riscv64-unknown-linux-musl/12.0.1/lto-wrapper
Target: riscv64-unknown-linux-musl
Configured with: /builds/riscv-toolchain/gcc/configure --target=riscv64-unknown-linux-musl --prefix=/builds/riscv-toolchain/install-native/ --with-sysroot=/builds/riscv-toolchain/install-native//riscv64-unknown-linux-musl --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=/builds/riscv-toolchain/gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdcv --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medany -march=rv64imafdcv -mabi=lp64d' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medany -march=rv64imafdcv -mabi=lp64d'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 12.0.1 20220505 (prerelease) (GCC) 
```

I am wondering if I can use this toolchain for non-vector targets like `rv64g`? 

I tried building Linux with `-march=rv64g`  option but got error like:

```
during GIMPLE pass: vect
/home/src/.../linux/fs/pipe.c: In function 'pipe_ioctl':
/home/src/.../linux/fs/pipe.c:594:13: internal compiler error: in get_len_gather_scatter_mode, at optabs-query.cc:638
  594 | static long pipe_ioctl(struct file *filp, unsigned int cmd, unsigned long arg)
      |             ^~~~~~~~~~
0x1e4087e internal_error(char const*, ...)
0x1e4087e internal_error(char const*, ...)
	???:0
0x87b7f4 fancy_abort(char const*, int, char const*)
	???:0
0x1dffde2 vect_analyze_stmt(vec_info*, _stmt_vec_info*, bool*, _slp_tree*, _slp_instance*, vec<stmt_info_for_cost, va_heap, vl_ptr>*)
	???:0
0x107b36b vect_analyze_loop(loop*, vec_info_shared*)
	???:0
```

Is my usage correct or it is a bug in older versions?

TommyMurphyTM1234, 2023-11-09:
Please clarify exactly how you are building your toolchain.
E.g. including the `git clone`, `configure` and `make` commands used and the repo(s) and branches used.

yf13, 2023-11-10:
@TommyMurphyTM1234 thanks a lot for the comments! actually I didn't build it myself, I got it from a chip vendor. 

If you confirm that my usage is correct, then I can report to  the chip vendor.
 
As that vendor one seems old, maybe I can start with a newer version and see how it behaves.  can you teach which version is proper?

TommyMurphyTM1234, 2023-11-10:
> As that vendor one seems old, maybe I can start with a newer version and see how it behaves. can you teach which version is proper?

Have you tried this?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=... --with-arch=rv64imafdcv --with-abi=lp64d --enable-multilib
make musl
```
Or maybe the `riscv-gcc` repo (https://github.com/riscv-collab/riscv-gcc) has more recent developmental/work in progress vector support than upstream GCC (it's always been confusing to me as to where the latest/best vector support lives):

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1301

> Is my usage correct or it is a bug in older versions?

I have no idea because I don't know what vendor or toolchain you are referring to.

TommyMurphyTM1234, 2023-11-10:
> Have you tried this?
> 
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain
> cd riscv-gnu-toolchain
> ./configure --prefix=... --with-arch=rv64imafdcv --with-abi=lp64d --enable-multilib
> make musl
> ```

Oh - that doesn't work - I forgot that `--enable-multilib` doesn't work with the Linux/musl toolchain.
Maybe try the Linux/Glibc toolchain instead?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=... --with-arch=rv64imafdcv --with-abi=lp64d --enable-multilib
make linux
```

TommyMurphyTM1234, 2023-11-10:
Just to clarify - if the toolchain obtained from the third party vendor is a Linux/Musl toolchain then it almost certainly is not a multilib toolchain - `riscv64-uknown-linux-musl-gcc -print-multi-lib` will clarify. If the only output is `.;` then it is not a multilib toolchain. Assuming that to be the case and, as per your first post, the toolchain default `arch/abi` is `rv64imafdcv/lp64d` you should still be able to compile for `rv64gc/lp64d` but you will either get link errors due to the lack of `rv64gc/lp64d` multilibs/startup code etc. or - worse still - it may link the default libs compiled for `rv64imafdcv/lp64d` and thus potentially include vector instructions while targeting a non vector enabled target.

yf13, 2023-11-11:
@TommyMurphyTM1234 thank you so much!

Indeed the vendor Linux/Musl toolchain just outputs `.;` when used with `-print-multi-lib` option. so it doesn't support `rv64g` variation as you said.

TommyMurphyTM1234, 2023-11-11:
> Indeed the vendor Linux/Musl toolchain just outputs .; when used with -print-multi-lib option. so it doesn't support rv64g variation as you said.

Just to be clear, it **can** generate code for, say, `rv64g/lp64d`, but it will not have multilib libraries/startup object files for that `arch/abi` to link to and will fall back to trying to link the toolchain default `arch/abi` ones - which may lead to link time or run time problems.

You can compile and link programs for, say, `rv64g/lp64d`, that are "standalone/self-contained" and don't depend on standard library/startup code, but that's probably unlikely to be practical especially when compiling Linux user mode programs. (It may be more feasible in a bare-metal/embedded scenario).

At the moment, there doesn't seem to be a way to build a multilib Linux/Musl toolchain. As such, you may need to build individual Linux/Musl toolchains for the specific `arch/abi`s that you need to target, or else use a multilib enabled Linux/Glibc toolchain as sketched out here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1364#issuecomment-1806062580

yf13, 2023-11-12:
@TommyMurphyTM1234 , many thanks! do you know if the LLVM submodule of this repo can do multi-lib for Musl or not?

TommyMurphyTM1234, 2023-11-12:
> I saw `README.md` says below for LLVM tool:
> 
> > Note, that a combination of `--enable-llvm` and multilib configuration flags is not supported.
> 
> Does this mean that Linux/Glibc multilib may likely work for LLVM later but not now?

I don't really understand what you mean.
Linux/Glibc and LLVM are completely different toolchains.
As far as I can see the LLVM toolchain uses Glibc only.
At least the CI builds are LLVM/Glibc only right now:

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2023.11.08
 
> I also noticed `README.md` says below about Musl:
> 
> > The musl compiler (riscv64-unknown-linux-musl-) will only be able to target 64-bit systems due to limitations in the upstream musl architecture support. The `--enable-multilib` flag therefore does not actually enable multilib support for musl libc.
> 
> Does this mean it is still possible to have Musl multilib for 64bit only variants like "rv64gc", "rv64gcv" etc?

No - at least, not as things stand.
I don't know what the issues are in allowing multilib for LLVM and/or Musl toolchains.

Why don't you just build separate toolchains for each `arch/abi` that you need to target as I suggested earlier?

yf13, 2023-11-13:
@TommyMurphyTM1234 thanks again for the explanations.

> Why don't you just build separate toolchains for each arch/abi that you need to target as I suggested earlier?

Yes I am trying to build `linux-glibc` for `-march=rv64gc_zicsr_zifencei` but not done yet.  The SoC at hand has two  CPU cores, one with vector and one without. Having multiple toolchains for one chip is really complex thus I am wondering if things will be simpler one day.

thank you so much for all the comments, I think this can be closed for now.

closed 2023-11-13 by yf13