riscv-collab/riscv-gnu-toolchain #1365: How to run 'make report-*' multiple times

muxiff, 2023-11-10:
Hi,

It seems that only once ‘make report-*’ is supported after the toolchain is built, because when I add some new gcc/binutils testsuits, ‘make report-*’ again will not refresh the test result, which causes me to have to rebuild the toolchain and then run 'make report-*'. So I was wondering is there a right way to avoid repeatedly building the toolchain while testing.

Thanks!

kito-cheng, 2023-11-10:
My workaround is manually remove corresponding stamp in the `<build>/stamps` folder...:P

TommyMurphyTM1234, 2023-11-10:
Might also be useful to review this issue as it's in a similar vein:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1196

cmuellner, 2023-11-10:
`rm -f stamps/check-gcc*`

muxiff, 2023-11-13:
Thank you for your reply. It really saved a lot of time.

closed 2023-11-13 by muxiff