riscv-collab/riscv-gnu-toolchain #1366: "Impossible Constriant" for well defined instruction

HakamAtassi, 2023-11-11:
Hello!

Hope you are well. I'm currently trying to add custom instruction to a soft riscv core. The function of the instruction is not relavent. 

The Mask and Match for the instruction is: 

[image]

[image]
[image]

Note that the first instruction is encoded as: 

[image]

Note that there is 1 available bit (MSB) in the instruction. I have been attempting to use a larger IMM in the instruction, yet, when I attempt to use 7 as my IMM 

[image]

Despite the fact that there is indeed an available bit. Not much is provided by the compiler to help debug this issue. Any ideas as to why the compiler is claiming the instruction is "impossible"? Thanks.

cmuellner, 2023-11-12:
The machine contraints are defined here: https://gcc.gnu.org/onlinedocs/gcc/Machine-Constraints.html
`I` is defined as:

`3<<9` has bit 10 as MSB.
`7<<9` has bit

cmuellner, 2023-11-12:
The machine constraints are defined here: https://gcc.gnu.org/onlinedocs/gcc/Machine-Constraints.html
I is defined as:
> An I-type 12-bit signed immediate.

`3<<9` has bit 10 as MSB.
`7<<9` has bit 11 as MSB (i.e., is a 12-bit unsigned immediate).

So your immediate does not fit into the used constraint.

closed 2023-11-12 by cmuellner

HakamAtassi, 2023-11-12:
Using -512 worked, thank you and apologies for the oversight!

DIGI1919, 2024-04-12:
> The machine constraints are defined here: https://gcc.gnu.org/onlinedocs/gcc/Machine-Constraints.html I is defined as:
> 
> > An I-type 12-bit signed immediate.
> 
> `3<<9` has bit 10 as MSB. `7<<9` has bit 11 as MSB (i.e., is a 12-bit unsigned immediate).
> 
> So your immediate does not fit into the used constraint. Your immediate `7<<9` is `0b1110_0000_0000`, which can also be expressed as `-512` in (s+11)-bit arithmetic.

hi，if i want build a function to deliver parameters ,how can i do that?such as:
`void write_csr(uint32_t csr_addr, uint32_t value) {
    asm volatile ("csrw %0, %1" : : "i"(csr_addr), "r"(value));
}`
but the addr of csr should be 12bit,so it can not pass the compiler.i have tested int8 int16 char,but can not pass at all.thank you.

cmuellner, 2024-04-12:
> hi，if i want build a function to deliver parameters ,how can i do that?such as: `void write_csr(uint32_t csr_addr, uint32_t value) { asm volatile ("csrw %0, %1" : : "i"(csr_addr), "r"(value)); }` [...]

The 12-bit address is encode into the instruction. So it must be a compile-time known constant and you cannot make it a run-time variable.

DIGI1919, 2024-04-12:
> The 12-bit address is encode into the instruction. So it must be a compile-time known constant and you cannot make it a run-time variable.

thanks  a lot,so is it means that the only way i can do is  to write many function to read different csr? perhaps, if it possible for me to use some c langrage skills to avoid it ?

cmuellner, 2024-04-12:
> thanks a lot,so is it means that the only way i can do is to write many function to read different csr? perhaps, if it possible for me to use some c langrage skills to avoid it ?

It depends on what you need, because I don't know where a non-inlined wrapper-function around `csrw` would be useful. So, one solution could be to use `__attribute__((always_inline))` for such a wrapper (assuming that you know the CSR at the call-site).

Or, if you like terrible code, you could have a big switch-case of if-else structure where you match the values to compile-time constants (`if (addr == 0x1) foo(0x1); else if (addr == 0x2) foo(0x2) ...`). Note: even if this works, this is wrong.

DIGI1919, 2024-04-12:
> It depends on what you need, because I don't know where a non-inlined wrapper-function around `csrw` would be useful. So, one solution could be to use `__attribute__((always_inline))` for such a [...]

thank you very much ☺