riscv-collab/riscv-gnu-toolchain #1372: A zifencei-related error occurs when running RISC-V build results with spikes.

YJHcc, 2023-11-23:
ubuntu ver : 18.04
I want to install riscv-gnu-toolchain, riscv-isa-sim, and riscv-pk and run RISC-V build results as spikes as in the process below.
```
Compiling and Running a Simple C Program
Install spike (see Build Steps), riscv-gnu-toolchain, and riscv-pk.
Write a short C program and name it hello.c. Then, compile it into a RISC-V ELF binary named hello:
$ riscv64-unknown-elf-gcc -o hello hello.c

Now you can simulate the program atop the proxy kernel:
$ spike pk hello
```

At this time, the following error occurs during the process of installing and building riscv-pk.

../machine/flush_icache.c: Assembler messages:
../machine/flush_icache.c:4: Error: unrecognized opcode `fence.i', extension `zifencei' required
Makefile:336: recipe for target 'flush_icache.o' failed
make: *** [flush_icache.o] Error 1

To solve this problem, we tried to solve it by executing the following command in the riscv-gnu-toolchain/build path to expand zifencei.
```
../configure --prefix=/home/yujihoon/riscv-demo --with-arch=rv64gc_zifencei
```

Afterwards, I tried make again, but the error still remained.

Is it true that zifencei is supported in the corresponding riscv-gnu-toolchain version?

If correct, please tell me the solution.

TommyMurphyTM1234, 2023-11-23:
Are you sure that you're using the latest sources?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1282
* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1283

Please clarify exactly how you are building the toolchain and simulator - i.e. all `git clone`, `configure` and `make` commands and build logs for the toolchain and Spike.

TommyMurphyTM1234, 2023-11-23:
This built without errors for me:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1372
cd issue-1372
./configure --prefix=`pwd`/installed-tools --with-sim=spike
make 2>&1 | tee build-toolchain.log
make build-sim 2>&1 | tee build-sim.log
```

- `build-toolchain.log`: [build-toolchain.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13449493/build-toolchain.zip)
- `build-sim.log`: [build-sim.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13449496/build-sim.zip)

TommyMurphyTM1234, 2023-11-23:
> This built without errors for me:
>
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1372
> cd issue-1372
> ./configure --prefix=`pwd`/installed-tools --with-sim=spike
> make 2>&1 | [...]

And this too:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1372
cd issue-1372
./configure --prefix=`pwd`/installed-tools --with-sim=spike --enable-multilib
make 2>&1 | tee build-toolchain.log
make build-sim 2>&1 | tee build-sim.log
```
* `build-toolchain.log`: [build-toolchain.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13449931/build-toolchain.zip)
* `build-sim.log`: [build-sim.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13449944/build-sim.zip)

TommyMurphyTM1234, 2023-11-24:
Any update @YJHcc?

TommyMurphyTM1234, 2023-11-25:
No update and I cannot reproduce the issue so I am closing it.
If you have an update and, more specifically, complete instructions for reproducing the issue then please post them and reopen the issue.

closed 2023-11-25 by TommyMurphyTM1234