riscv-collab/riscv-gnu-toolchain #1382: Failing to build the toolchain on macOS

bpervan, 2023-11-30:
Hi all,

I'm trying (and failing) to build the toolchain on macOS (M2 Pro, Sonoma 14. 1. 1.). More specifically, it seems that the `configure` step of `build-gcc-newlib-stage1` fails with:

```
checking size of long long... configure: error: in `/Volumes/rvbuild/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc':
configure: error: cannot compute sizeof (long long)
```

This is the output of `configure`:

```
configure: creating cache ./config.cache
checking build system type... aarch64-apple-darwin23.1.0
checking host system type... aarch64-apple-darwin23.1.0
checking target system type... aarch64-apple-darwin23.1.0
checking LIBRARY_PATH variable... ok
checking GCC_EXEC_PREFIX variable... ok
checking whether to place generated files in the source directory... no
checking whether a default linker was specified... no
checking whether a default dsymutil was specified... no
checking whether a default assembler was specified... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for gnatbind... no
checking for gnatmake... no
checking whether compiler driver understands Ada and is recent enough... no
checking for gdc... no
checking whether the D compiler works... no
checking how to run the C++ preprocessor... g++ -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... no
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking how to run the C preprocessor... gcc -E
checking for inline... inline
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking size of ino_t... 0
checking size of dev_t... 0
checking size of void *... 0
checking size of short... 0
checking size of int... 0
checking size of long... 0
checking for long long... yes
checking size of long long... configure: error: in `/Volumes/rvbuild/riscv-gnu-toolchain/gcc/gcc':
configure: error: cannot compute sizeof (long long)
See `config.log' for more details
```

What's also particularly interesting is that I get 0 for sizes of types that obviously should not be 0.

Any clues?

TommyMurphyTM1234, 2023-11-30:
Please compress and attach your full configure/build logs.

TommyMurphyTM1234, 2023-11-30:
BTW - if a pre-built toolchain will meet your needs then maybe the xPack version will do?

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#releases
* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/releases/
* https://xpack.github.io/blog/2023/09/25/riscv-none-elf-gcc-v13-2-0-2-released/

TommyMurphyTM1234, 2023-11-30:
> Here is the full [config.log](https://gist.github.com/bpervan/d2a29f6d1efc7b6bc24143a347663ff7)

Are you sure that that's the correct log file?
This suggests not:

* https://gist.github.com/bpervan/d2a29f6d1efc7b6bc24143a347663ff7#file-config-log-L1286-L1288

bpervan, 2023-12-01:
> Or maybe the failure is rooted in this earlier issue?
> 
> * https://gist.github.com/bpervan/d2a29f6d1efc7b6bc24143a347663ff7#file-config-log-L1183-L1190

Indeed it was, thanks for pointing that out, I missed it completely! Anyways, getting the installed `libc++` out of the way from `/usr/local/lib`, as suggested [here](https://forums.swift.org/t/macos-dyld-library-path-issue-library-not-loaded-rpath-libc-1-dylib/68567/11) did the trick.

closed 2023-12-01 by bpervan