riscv-collab/riscv-gnu-toolchain #1385: Build Errors in RISC-V Toolchain Installation - Implicit Declaration in libm/complex

gyuminb, 2023-12-06:
Hello,

I am experiencing build errors during the installation and updating of the RISC-V toolchain using the **`install.sh`** and **`update.sh`** scripts. The issue arises during the compilation of the **`newlib`** library, particularly within the **`libm/complex`** directory. The error is related to implicit declarations in math and complex headers, affecting functions like **`coshl`**, **`cosl`**, **`sinhl`**, and **`sinl`**. This results in compilation failures in the **`libm/complex`** part of **`newlib`**.

Here are the contents of the **`install.sh`** and **`update.sh`** scripts I am using:

**`install.sh`**:

```bash
#!/bin/sh

current_path=$PWD

# Check if script is run as root
if [ "$(id -u)" != "0" ]; then
    echo "This script requires superuser access for some operations."
    echo "Please run with 'sudo -E' when prompted."
    exit 1
fi

if [ ! -z "$SUDO_USER" ]; then
    USER_HOME=$(getent passwd "$SUDO_USER" | cut -d: -f6)
else
    USER_HOME=$HOME
fi

install_riscv_gcc() {
  INSTALL_DIR=$USER_HOME/riscv

  git clone https://github.com/riscv/riscv-gnu-toolchain || { echo "Failed to clone riscv-gnu-toolchain"; return 1; }
  cd riscv-gnu-toolchain || { echo "Failed to enter riscv-gnu-toolchain directory"; return 1; }
  git submodule update --init gcc

  cd gcc && git checkout trunk || { echo "Failed to checkout gcc trunk"; return 1; }
  cd "$current_path"

  cd riscv-gnu-toolchain
  mkdir gcc-build && cd gcc-build
  ../configure --prefix=$INSTALL_DIR || { echo "Configure failed"; return 1; }
  make -j 2 || { echo "Make failed"; return 1; }
  make install || { echo "Make install failed"; return 1; }
  cd "$current_path"

  ln -s $INSTALL_DIR/bin/riscv64-unknown-elf-gcc $current_path/riscv64-unknown-elf-gcc || { echo "Failed to create symbolic link for RISC-V GCC"; exit 1; }
  ln -s $INSTALL_DIR/bin/riscv64-unknown-elf-g++ $current_path/riscv64-unknown-elf-g++ || { echo "Failed to create symbolic link for RISC-V G++"; exit 1; }

}
```

**`update.sh`**:

```bash
#!/bin/sh

current_path=$PWD

# Check if script is run as root
if [ "$(id -u)" != "0" ]; then
    echo "This script requires superuser access for some operations."
    echo "Please run with 'sudo -E' when prompted."
    exit 1
fi

if [ ! -z "$SUDO_USER" ]; then
    USER_HOME=$(getent passwd "$SUDO_USER" | cut -d: -f6)
else
    USER_HOME=$HOME
fi

# Update and Rebuild RISC-V GCC
update_and_rebuild_riscv_gcc() {
  echo "Updating and rebuilding RISC-V GCC..."
  INSTALL_DIR=$USER_HOME/riscv
  if [ -d "$current_path/riscv-gnu-toolchain" ]; then
    cd "$current_path/riscv-gnu-toolchain" && git reset --hard && git clean -fdx && git pull && git submodule foreach --recursive git reset --hard && git submodule foreach --recursive git clean -fdx && git submodule update --init --recursive || { echo "Failed to update RISC-V GCC"; return 1; }
    cd gcc && git stash && git checkout trunk && git pull || { echo "Failed to update RISC-V GCC"; return 1; }
    cd ..
    if [ -d gcc-build ]; then
      sudo rm -rf gcc-build || { echo "Failed to delete existing RISC-V GCC build directory"; return 1; }
    fi

    if [ -d $INSTALL_DIR ]; then
      sudo rm -rf $INSTALL_DIR || { echo "Failed to delete existing RISC-V GCC installation directory"; return 1; }
    fi

    mkdir gcc-build && cd gcc-build || { echo "Failed to enter gcc-build directory"; return 1; }
    ../configure --prefix=$INSTALL_DIR || { echo "Configure failed"; return 1; }
    make -j 4 || { echo "Make failed"; return 1; }
    sudo make install || { echo "Make install failed"; return 1; }
    cd "$current_path"
    [ -L riscv64-unknown-elf-gcc ] && rm riscv64-unknown-elf-gcc
    [ -L riscv64-unknown-elf-g++ ] && rm riscv64-unknown-elf-g++
    ln -s $INSTALL_DIR/bin/riscv64-unknown-elf-gcc riscv64-unknown-elf-gcc || { echo "Failed to create symbolic link for RISC-V GCC"; return 1; }
    ln -s $INSTALL_DIR/bin/riscv64-unknown-elf-g++ riscv64-unknown-elf-g++ || { echo "Failed to create symbolic link for RISC-V G++"; return 1; }
    echo "RISC-V GCC successfully updated and rebuilt."
  else
    echo "RISC-V GCC directory does not exist."
    return 1
  fi
}
```

The error message encountered during the build process is as follows:

```bash
bashCopy code
  CC       libm/complex/libm_a-ccoshl.o
  ...
  /home/novice/riscv-gnu-toolchain/newlib/newlib/libm/complex/ccoshl.c: In function 'ccoshl':
  /home/novice/riscv-gnu-toolchain/newlib/newlib/libm/complex/ccoshl.c:43:13: error: implicit declaration of function 'coshl'; did you mean 'coshf'? [-Wimplicit-function-declaration]
  ...

```

I'm looking for guidance on resolving these build errors, which seem to be related to missing or incorrectly referenced library functions. Any help or insights you can provide would be greatly appreciated.

Thank you for your support.

kito-cheng, 2023-12-06:
Use newlib trunk as well, I already fixed that issue this week.

https://github.com/bminor/newlib/commit/04798b7bb69571452d2cfc7e0b052a9bbd3b619d
https://github.com/bminor/newlib/commit/bcd1883560b79b67f06fa1df6dea60d18fdd00c7

TommyMurphyTM1234, 2023-12-06:
> Here are the contents of the **`install.sh`** and **`update.sh`** scripts I am using:

If I run your scripts they do nothing.
Did you omit some step required to reproduce your issue?

gyuminb, 2023-12-06:
> > Here are the contents of the **`install.sh`** and **`update.sh`** scripts I am using:
> 
> If I run your scripts they do nothing. Did you omit some step required to reproduce your issue?
Yes, I realized that I only included the function definitions in the scripts and omitted the execution part. I have now updated the scripts to include the execution of the defined functions. You should be able to reproduce the issue now.

gyuminb, 2023-12-06:
> Use newlib trunk as well, I already fixed that issue this week.
> 
> [bminor/newlib@04798b7](https://github.com/bminor/newlib/commit/04798b7bb69571452d2cfc7e0b052a9bbd3b619d) [bminor/newlib@bcd1883](https://github.com/bminor/newlib/commit/bcd1883560b79b67f06fa1df6dea60d18fdd00c7)

Thank you for your guidance. I've applied the updates from the newlib trunk as you suggested, and I'm happy to report that it worked perfectly. I really appreciate your help.