riscv-collab/riscv-gnu-toolchain #1388: About build native gcc

Xxfore, 2023-12-08:
Hi,

   I have successfully build native toolchain by using  `make linux-native`, but i cant find `libcc1.so` for native toolchain under installed folder, 
   In order to generate libcc1.so, am i missing any steps for compiling?

Thanks a lot.

TommyMurphyTM1234, 2023-12-08:
Can you capture, compress and attach your full build log?
E.g.
```
make linux-native 2>&1 | tee build.log
```

Xxfore, 2023-12-08:
> Can you capture, compress and attach your full build log? E.g.
> 
> ```
> make linux-native 2>&1 | tee build.log
> ```

Hi, 
Please refer attached log, 

[build-1.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13612635/build-1.log)
[build-2.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13612641/build-2.log)

By using follow command , there's no libcc1.so under native folder , just have the host one.
`
find _install/riscv-toolchain/ | grep libcc1.so
_install/riscv-toolchain/lib/libcc1.so.0.0.0
_install/riscv-toolchain/lib/libcc1.so
_install/riscv-toolchain/lib/libcc1.so.0
`

Thanks a lot.

TommyMurphyTM1234, 2023-12-12:
Apologies for the delay in replying.
Unfortunately I can't really help here as I'm not familiar enough with the `make linux-native` option and don't have a setup in which I can exercise such a build.
I presume that it could be done on something like QEMU running a RISC-V Linux installation but I don't have the time to set this up right now.
I'm a bit confused by the fact that the log file contains a lot of the following:
```
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... riscv64-unknown-linux-gnu
```
which suggests that it's building on x86_64/Linux to run on x86_64/Linux to ultimately target RISC-V/Linux.
And in other places I see this which seems even more confusing...
```
checking host system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking target system type... riscv64-unknown-linux-gnu
checking target system type... riscv64-unknown-linux-gnu
checking target system type... riscv64-unknown-linux-gnu
```

TommyMurphyTM1234, 2024-03-31:
According to this:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1427#issuecomment-1997827479

> Building a native RISC-V toolchain is not a supported feature.

So I suspect that this issue can be closed?

TommyMurphyTM1234, 2024-06-06:
> According to this:
> 
> * [How to build RISC-V native toolchain? #1427 (comment)](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1427#issuecomment-1997827479)
> 
> > Building a native RISC-V toolchain is not a supported feature.
> 
> So I suspect that this issue can be closed?

Closing this issue now.

closed 2024-06-06 by TommyMurphyTM1234