riscv-collab/riscv-gnu-toolchain #1389: Oops, you commited a warcrime

Sainan, 2023-12-10:
1. `$` is not a valid command, drop that shit for easy copy-paste
2. Add `-j` to your make commands holy fuck

TommyMurphyTM1234, 2023-12-10:
No idea what this is meant to be but closing it because it seems pointless.

closed 2023-12-10 by TommyMurphyTM1234

Sainan, 2023-12-10:
I agree, `-j` should be the defaut, but alas.

cmuellner, 2023-12-11:
The instructions are not here to serve as a copy-and-paste list of instructions that fit all possible use cases. We could remove them, but having them will help beginners to understand what is needed to build a toolchain.

If you need help building this project, please take a look at the CI files. E.g., https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/workflows/build.yaml. If that's not enough, you can file a bug report stating what you typed, what you expected, and what you observed.

Adding `-j` might also not be the best idea (let me quote the man page: `If the -j option is given without an argument, make will not limit the number of jobs that can run simultaneously.`).

Using `-j$(nproc)` might be a good choice in many cases, but it is not essential to use it.

Using a `$` or `#` prefix before instructions is a good practice because it communicates if the command is expected to be run as a user or root. Of course, we could use this more consistently in the readme, but this would not provide any real value.

The only thing that clearly needs improvement is your language. But that is out of the scope of this project.