riscv-collab/riscv-gnu-toolchain #1391: fopen(filename, "w") of newlib toolchain results in segfault when run under qemu-riscv32

RamNalamothu, 2023-12-13:
With the newlib baremetal toolchain, the following simple program built with semihosting runs fine under Qemu.

$ cat file.c 
#include <stdio.h>
#include <stdlib.h>

int main(void) {
	FILE *fp = NULL;
	fp = fopen ("log.txt", "w");
	if (fp != NULL) {
	  printf("File opened successfully\n");
	  fclose(fp);
	} else {
	  printf("File opening failed\n");
	}
	exit(0);
}

$ qemu-riscv32 a.out
File opened successfully

But, however, in a perenniel test the _fopen(filename, "w")_ is running into segfault in the __malloc_r()_ function along the call chain.
$ qemu-riscv32 -strace P50002
--- SIGSEGV {si_signo=SIGSEGV, si_code=1, si_addr=0x00027144} ---

Are the file operations not expected to work properly even in semihosting?

@kito-cheng @jim-wilson

TommyMurphyTM1234, 2023-12-13:
Your post seems contradictory:

> With the newlib baremetal toolchain, the following simple program built with semihosting runs fine under Qemu.

> But, however, in a perenniel test the _fopen(filename, "w")_ is running into segfault in the __malloc_r()_ function along the call chain. $ qemu-riscv32 -strace P50002 --- SIGSEGV {si_signo=SIGSEGV, si_code=1, si_addr=0x00027144} ---

> Are the file operations not expected to work properly even in semihosting?

Bare metal requires something to provide low level support for things like system calls - e.g. hardware abstraction layer, board support package, [RT]OS. Without such support such calls will probably fail.

RamNalamothu, 2023-12-13:
I have built the newlib from tip of git://sourceware.org/git/newlib-cygwin.git using the LLVM toolchain and using that newlib as C library in another copy of LLVM toolchain for all the below experiments.

Build steps for newlib:

$ CC_FOR_TARGET=clang LD_FOR_TARGET=ld.lld RANLIB_FOR_TARGET=llvm-ranlib \
AR_FOR_TARGET=llvm-ar NM_FOR_TARGET=llvm-nm OBJDUMP_FOR_TARGET=llvm-objdump \
READELF_FOR_TARGET=llvm-readelf \
CFLAGS_FOR_TARGET="-Wno-implicit-function-declaration -Wno-int-conversion -target riscv32-unknown-elf -march=rv32imac -mabi=ilp32 -Os -fuse-ld=lld" \
../newlib-cygwin/configure --prefix=/path/to/llvm/install --target=riscv32-unknown-elf --disable-libstdcxx --disable-shared --enable-static --disable-multilib
$ make all && make install

Build steps for the C tests:

$ clang -target riscv32-unknown-elf -march=rv32imac -mabi=ilp32 -static \ 
**-nostdinc** -isystem /path/to/llvm/install/bin/../riscv32-unknown-elf/include -isystem /path/to/llvm/install/lib/clang/18/include \ **-nostdlib** -L/path/to/llvm/install/bin/../riscv32-unknown-elf/lib -L/path/to/llvm/install/lib/clang/18/lib/baremetal \ /path/to/llvm/install/bin/../riscv32-unknown-elf/lib/crt0.o -Wl,--start-group -lc **-lsemihost** -lclang_rt.builtins-riscv32 -Wl,--end-group -g -O0 file.c

I have built the Qemu for RISC-V from https://github.com/qemu/qemu of v7.2 stable branch which has semihosting support for RISC-V in both user and system emulation mode (https://www.qemu.org/docs/master/about/emulation.html#id3).

With that, the above simple test with fopen() works fine when run under _qemu-riscv32_.

> > Are the file operations not expected to work properly even in semihosting?
> 
> You're saying that the semihosting test works but then seem to be contradicting yourself?

Yes, it works for the above simple test with fopen() but not for other tests e.g. SPEC2000/bzip2, Perenniel etc which call fopen().

> What exactly do you mean by "perennial" here?

The validation test suites for C/C++ compilers available at https://www.peren.com.

> > Are the file operations not expected to work properly even in semihosting?

> Bare metal targets require something to provide low level support for things like system calls - e.g. HAL (Hardware Abstraction Layer), BSP (Board Support Package), [Real-Time] Operating System ([RT]OS) etc. Bare metal toolchains normally don't provide such support. Without this support such calls will probably fail unless, for example, they are intercepted by semihosting support.
> 
> For example, on a bare metal target with no filesystem a priori what exactly do you expect `fopen(filename, "w")` to do?

As both newlib and Qemu has semihosting support for RISC-V, I expect the fopen() call to happen through the host (Linux in my case) by semihosting mechanism.

TommyMurphyTM1234, 2023-12-13:
Sounds to me like a question to be posed in a newlib or QEMU forum?
I don't see how it's a `riscv-gnu-toolchain` issue.

RamNalamothu, 2023-12-14:
Ah, yes. Sorry for that.

However, what I was hoping to get here is how newlib toolchain and Qemu are supposed to be used in semihosting mode to get the file I/O and console I/O operations working?

From _riscv-gnu-toolchain_'s readme I see that Qemu is being (or can be) used for running newlib toolchain testsuite and I think people on this forum would have gone through the issues that I am running into before coming up with a working setup.

@kito-cheng @keith-packard

TommyMurphyTM1234, 2023-12-17:
> As both newlib and Qemu has semihosting support for RISC-V, I expect the fopen() call to happen through the host (Linux in my case) by semihosting mechanism.

Sounds like the failing case is not using semihosting for some reason.
You probably need to dig into that and maybe ask on the relevant forum for advice.
This is not an issue with the `riscv-gnu-toolchain` itself.

RamNalamothu, 2023-12-18:
Thanks for the comments @TommyMurphyTM1234.

I figured the root cause, in my case, is that the linker script I am using has not properly setup the heap area.

closed 2023-12-18 by RamNalamothu