riscv-collab/riscv-gnu-toolchain #1392: configure: error: cannot compute suffix of object files: cannot compile

Aaronyap2002, 2023-12-18:
thank me later:
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d--disable-float--disable-atomic

_Originally posted by @LakshmiKanth-77 in https://github.com/riscv-collab/riscv-gnu-toolchain/issues/134#issuecomment-808724060_

TommyMurphyTM1234, 2023-12-18:
> ```
> configure: error: cannot compute suffix of object files: cannot compile
> See `config.log' for more details
> ```

And what does `/home/aaron/riscv-gnu-toolchain/build-glibc-linux-rv32gc-ilp32d--disable-float--disable-atomic/config.log` say?

Aaronyap2002, 2023-12-18:
I don't exactly understand what i'm reading in the log file. I will attach both logs file with and without `--disable-float--disable-atomic`
without: 
[config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13701693/config.log)
with:
[config.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13701700/config.log)

TommyMurphyTM1234, 2023-12-18:
> I tried adding `--disable-float--disable-atomic` to be back as suggested by someone

Who is "someone"?
What do you mean by "to be back"?

> `--with-abi=ilp32d--disable-float--disable-atomic`

This is not a valid ABI string/specifier - in particular the `--disable-float--disable-atomic` bit.

TommyMurphyTM1234, 2023-12-18:
Seems to me like your compiler cannot be found when trying to configure/build GLIBC:
```
configure:3423: riscv32-unknown-linux-gnu-gcc  --version >&5
/home/aaron/riscv-gnu-toolchain/glibc/configure: line 3425: riscv32-unknown-linux-gnu-gcc: command not found
configure:3434: $? = 127
configure:3423: riscv32-unknown-linux-gnu-gcc  -v >&5
/home/aaron/riscv-gnu-toolchain/glibc/configure: line 3425: riscv32-unknown-linux-gnu-gcc: command not found
configure:3434: $? = 127
configure:3423: riscv32-unknown-linux-gnu-gcc  -V >&5
/home/aaron/riscv-gnu-toolchain/glibc/configure: line 3425: riscv32-unknown-linux-gnu-gcc: command not found
configure:3434: $? = 127
configure:3423: riscv32-unknown-linux-gnu-gcc  -qversion >&5
/home/aaron/riscv-gnu-toolchain/glibc/configure: line 3425: riscv32-unknown-linux-gnu-gcc: command not found
configure:3434: $? = 127
configure:3423: riscv32-unknown-linux-gnu-gcc  -version >&5
/home/aaron/riscv-gnu-toolchain/glibc/configure: line 3425: riscv32-unknown-linux-gnu-gcc: command not found
configure:3434: $? = 127
configure:3439: checking for suffix of object files
configure:3462: riscv32-unknown-linux-gnu-gcc  -c    -mcmodel=medlow -O2   conftest.c >&5
/home/aaron/riscv-gnu-toolchain/glibc/configure: line 3464: riscv32-unknown-linux-gnu-gcc: command not found
configure:3466: $? = 127
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "GNU C Library"
| #define PACKAGE_TARNAME "glibc"
| #define PACKAGE_VERSION "(see version.h)"
| #define PACKAGE_STRING "GNU C Library (see version.h)"
| #define PACKAGE_BUGREPORT "https://sourceware.org/bugzilla/"
| #define PACKAGE_URL "https://www.gnu.org/software/glibc/"
| #define PKGVERSION "(GNU libc) "
| #define REPORT_BUGS_TO "<https://www.gnu.org/software/libc/bugs.html>"
| /* end confdefs.h.  */
| 
| int
| main (void)
| {
| 
|   ;
|   return 0;
| }
configure:3481: error: in `/home/aaron/riscv-gnu-toolchain/build-glibc-linux-rv32gc-ilp32d':
configure:3483: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
```
Please clarify exactly how you are building your toolchain including all `git clone`, `configure` and `make` commands as well as any other steps that you take.

Aaronyap2002, 2023-12-18:
Here's the list of commands I have typed. 
```
aaron@aaron-TravelMate-P238-M:~/Downloads/ibex$ cd ..
aaron@aaron-TravelMate-P238-M:~/Downloads$ cd ..
aaron@aaron-TravelMate-P238-M:~$ cd riscv-gnu-toolchain/
aaron@aaron-TravelMate-P238-M:~/riscv-gnu-toolchain$ cd new^C
aaron@aaron-TravelMate-P238-M:~/riscv-gnu-toolchain$ sudo ./configure --prefix=/home/aaron/new/./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
[sudo] password for aaron: 
aaron@aaron-TravelMate-P238-M:~/riscv-gnu-toolchain$ sudo ./configure --prefix=/home/aaron/new/ --with-arch=rv32gc --with-abi=ilp32d
[sudo] password for aaron: 
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
aaron@aaron-TravelMate-P238-M:~/riscv-gnu-toolchain$ sudo make linux
rm -rf stamps/build-glibc-linux-rv32gc-ilp32d build-glibc-linux-rv32gc-ilp32d
mkdir build-glibc-linux-rv32gc-ilp32d
cd build-glibc-linux-rv32gc-ilp32d && \
	CC="riscv32-unknown-linux-gnu-gcc " \
	CXX="this-is-not-the-compiler-youre-looking-for" \
	CFLAGS="   -mcmodel=medlow -O2 " \
	CXXFLAGS="   -mcmodel=medlow -O2 " \
	ASFLAGS="  -mcmodel=medlow " \
	/home/aaron/riscv-gnu-toolchain/glibc/configure \
	--host=riscv32-unknown-linux-gnu \
	--prefix=/usr \
	--disable-werror \
	--enable-shared \
	--enable-obsolete-rpc \
	--with-headers=/home/aaron/riscv-gnu-toolchain/linux-headers/include \
	--disable-multilib \
	--enable-kernel=3.0.0 \
	 \
	--libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for riscv32-unknown-linux-gnu-gcc... riscv32-unknown-linux-gnu-gcc 
checking for suffix of object files... configure: error: in `/home/aaron/riscv-gnu-toolchain/build-glibc-linux-rv32gc-ilp32d':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make: *** [Makefile:404: stamps/build-glibc-linux-rv32gc-ilp32d] Error 1
aaron@aaron-TravelMate-P238-M:~/riscv-gnu-toolchain$ cd ..
aaron@aaron-TravelMate-P238-M:~$ mkdir new1
aaron@aaron-TravelMate-P238-M:~$ cd riscv-gnu-toolchain/
aaron@aaron-TravelMate-P238-M:~/riscv-gnu-toolchain$ sudo ./configure --prefix=/home/aaron/new1/ --with-arch=rv32gc --with-abi=ilp32d
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
aaron@aaron-TravelMate-P238-M:~/riscv-gnu-toolchain$ sudo make linux
rm -rf stamps/build-glibc-linux-rv32gc-ilp32d build-glibc-linux-rv32gc-ilp32d
mkdir build-glibc-linux-rv32gc-ilp32d
cd build-glibc-linux-rv32gc-ilp32d && \
	CC="riscv32-unknown-linux-gnu-gcc " \
	CXX="this-is-not-the-compiler-youre-looking-for" \
	CFLAGS="   -mcmodel=medlow -O2 " \
	CXXFLAGS="   -mcmodel=medlow -O2 " \
	ASFLAGS="  -mcmodel=medlow " \
	/home/aaron/riscv-gnu-toolchain/glibc/configure \
	--host=riscv32-unknown-linux-gnu \
	--prefix=/usr \
	--disable-werror \
	--enable-shared \
	--enable-obsolete-rpc \
	--with-headers=/home/aaron/riscv-gnu-toolchain/linux-headers/include \
	--disable-multilib \
	--enable-kernel=3.0.0 \
	 \
	--libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for riscv32-unknown-linux-gnu-gcc... riscv32-unknown-linux-gnu-gcc 
checking for suffix of object files... configure: error: in `/home/aaron/riscv-gnu-toolchain/build-glibc-linux-rv32gc-ilp32d':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make: *** [Makefile:404: stamps/build-glibc-linux-rv32gc-ilp32d] Error 1
```

TommyMurphyTM1234, 2023-12-18:
> Here's the list of commands I have typed.

There's no `git clone` there.

This worked fine for me using Ubuntu 22.04.3 LTS running on WSL2:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1392
cd issue-1392
./configure --prefix=`pwd`/installed-tools --with-arch=rv32gc --with-abi=ilp32d
make linux 2>&1 | tee build.log
```
Here is the zipped `build.log` in case it helps to compare with your results: 

* [build.log.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13702780/build.log.zip)

I can only assume that your problem is down to some local configuration/environmental issue.

Aaronyap2002, 2023-12-18:
okay! I shall try it again

Derrick0415, 2023-12-18:
Just encountered a similar problem when "make linux -j", config is "../configure --prefix=$RISCV --with-arch=rv64gcv_zfh_zvfh0p1 --with-abi=lp64d --with-tune=rocket --with-sim=spike --enable-llvm --enable-linux"
```shell
.....
mkdir -p -- riscv64-unknown-linux-gnu/libgcc
Configuring in riscv64-unknown-linux-gnu/libgcc
configure: creating cache ./config.cache
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking for --enable-version-specific-runtime-libs... no
checking for a BSD-compatible install... /usr/bin/install -c
checking for gawk... /usr/bin/gawk
checking for riscv64-unknown-linux-gnu-ar... /home/derrick/riscv-linux-gcc/riscv64-unknown-linux-gnu/bin/ar
checking for riscv64-unknown-linux-gnu-lipo... riscv64-unknown-linux-gnu-lipo
checking for riscv64-unknown-linux-gnu-nm... /home/derrick/riscv-gnu-toolchain/build/build-gcc-linux-stage1/./gcc/nm
checking for riscv64-unknown-linux-gnu-ranlib... /home/derrick/riscv-linux-gcc/riscv64-unknown-linux-gnu/bin/ranlib
checking for riscv64-unknown-linux-gnu-strip... /home/derrick/riscv-linux-gcc/riscv64-unknown-linux-gnu/bin/strip
checking whether ln -s works... yes
checking for riscv64-unknown-linux-gnu-gcc...  /home/derrick/riscv-gnu-toolchain/build/build-gcc-linux-stage1/./gcc/xgcc -B/home/derrick/riscv-gnu-toolchain/build/build-gcc-linux-stage1/./gcc/ -B/home/derrick/riscv-linux-gcc/riscv64-unknown-linux-gnu/bin/ -B/home/derrick/riscv-linux-gcc/riscv64-unknown-linux-gnu/lib/ -isystem /home/derrick/riscv-linux-gcc/riscv64-unknown-linux-gnu/include -isystem /home/derrick/riscv-linux-gcc/riscv64-unknown-linux-gnu/sys-include   
checking for suffix of object files... configure: error: in `/home/derrick/riscv-gnu-toolchain/build/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[1]: *** [Makefile:12811: configure-target-libgcc] Error 1
make[1]: Leaving directory '/home/derrick/riscv-gnu-toolchain/build/build-gcc-linux-stage1'
make: *** [Makefile:434: stamps/build-gcc-linux-stage1] Error 2
```

TommyMurphyTM1234, 2023-12-18:
> > Just encountered a similar problem when "make linux -j", config is "../configure --prefix=$RISCV --with-arch=rv64gcv_zfh_zvfh0p1 --with-abi=lp64d --with-tune=rocket --with-sim=spike --enable-llvm --enable-linux"
> 
> What if you do a clean build without `-j` parallelism?
> 
> ```
> rm -rf $RISCV
> make distclean
> make linux 2>&1 | tee build.log
> ```

Never mind - that's a red herring.
I tried without `-j` and got the same problem.
Looking at `build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc/config.log` I see this which seems to be the cause of your problem:
```
Assembler messages:
Error: rv64imafdcv_zicsr_zifencei_zfh_zfhmin_zve32f_zve32x_zve64d_zve64f_zve64x_zvfh0p1_zvl128b_zvl32b_zvl64b: unknown prefixed ISA extension `zvfh'
/tmp/ccQrl8Ko.s:3: Error: rv64i2p1_m2p0_a2p1_f2p2_d2p2_c2p0_v1p0_zicsr2p0_zifencei2p0_zfh1p0_zfhmin1p0_zve32f1p0_zve32x1p0_zve64d1p0_zve64f1p0_zve64x1p0_zvfh0p1_zvl128b1p0_zvl32b1p0_zvl64b1p0: unknown prefixed ISA extension `zvfh'
```
Are you sure that your architecture string/specifier is correct?
```
--with-arch=rv64gcv_zfh_zvfh0p1
```

TommyMurphyTM1234, 2023-12-18:
> Are you sure that your architecture string/specifier is correct?
> 
> ```
> --with-arch=rv64gcv_zfh_zvfh0p1
> ```

I can only see one version of the `Zvfh` extension ratified which I would assume is version 1.0?

* https://wiki.riscv.org/display/HOME/Recently+Ratified+Extensions#:~:text=%E2%80%9CZvfh/Zvfhmin%3A%E2%80%9D%20Vector%20Extension%20for%20Half%2DPrecision%20Floating%2DPoint%20Arithmetic/Vector%20Extension%20for%20Minimal%20Half%2D%0APrecision%20Floating%2DPoint%20Arithmetic

what is the rationale for specifying `zvfh0p1` aka version 0.1?

TommyMurphyTM1234, 2023-12-18:
Just looking at the upstream GCC patch mailing list and I wonder if `Zvfh` (and related?) extension support is still only being worked on on the `master` branch so maybe you need to use that? E.g.:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc -b master gcc-master
./configure --prefix=$RISCV --with-arch=rv64gcv_zfh_zvfh0p1 --with-abi=lp64d --with-tune=rocket --with-sim=spike --enable-llvm --enable-linux --with-gcc-src=`pwd`/gcc-master
make -j linux
```

TommyMurphyTM1234, 2023-12-18:
> Thx, I will try other isa spec. p.s. i'm trying to build a linux clang for Ara.

In that case shouldn't you be doing this?

* https://github.com/pulp-platform/ara?tab=readme-ov-file#toolchain

Derrick0415, 2023-12-18:
> > Thx, I will try other isa spec. p.s. i'm trying to build a linux clang for Ara.
> 
> In that case shouldn't you be doing this?
> 
> * https://github.com/pulp-platform/ara?tab=readme-ov-file#toolchain

already done, just trying to learn the process, thank you for your support and concern。:)

TommyMurphyTM1234, 2023-12-18:
> Just looking at the upstream GCC patch mailing list and I wonder if `Zvfh` (and related?) extension support is still only being worked on on the `master` branch so maybe you need to use that? E.g.: [...]

FYI that doesn't help.
Maybe GLIBC upstream master is needed?
Or maybe I'm on another wild goose chase... :-(

Derrick0415, 2023-12-19:
> FYI that doesn't help. Maybe GLIBC upstream master is needed? Or maybe I'm on another wild goose chase... :-(

Removing ```zfh_zvfh0p1``` is working, you were right.
I'll find out which one of ```zfh``` and ```zvfh0p1``` is causing the problem.
Thx again.

TommyMurphyTM1234, 2024-03-30:
Closing this since the issue seems to have been worked around.
If there is still some outstanding issue then it should probably be reported in the relevant upstream project (GCC, GLIBC, Binutils etc.).

closed 2024-03-30 by TommyMurphyTM1234