riscv-collab/riscv-gnu-toolchain #1396: Is there a way to clone the latest 12.2 version?

jesseopdenbrouw, 2023-12-23:
Hi all,

I would like to clone the latest 12.2 version of the toolchain. What git command should I use>

Greets

Jesse

jesseopdenbrouw, 2023-12-23:
Of course In mean with GCC 12.2 as accompanying tools.

TommyMurphyTM1234, 2023-12-23:
What about this?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc -b releases/gcc-12 gcc-12
./configure --prefix=... --with-gcc-src=`pwd`/gcc-12
make ...
```

jesseopdenbrouw, 2023-12-23:
I meant gcc 12.2 and the tools (binutils etc). I will test your proposal first.

TommyMurphyTM1234, 2023-12-23:
Another option might be to pull from the commit immediately prior to this one:

* https://github.com/riscv-collab/riscv-gnu-toolchain/commit/1d9d2ce4058faadc56922746da7b41f557efafd1

TommyMurphyTM1234, 2023-12-23:
> Another option might be to pull from the commit immediately prior to this one:
> 
> * https://github.com/riscv-collab/riscv-gnu-toolchain/commit/1d9d2ce4058faadc56922746da7b41f557efafd1

Which seems to be this commit: f7186fa52b4011684408858987840950e3e93417
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1396
cd issue-1396
git log

...

commit 1d9d2ce4058faadc56922746da7b41f557efafd1
Author: Christoph Müllner <christoph.muellner@vrull.eu>
Date:   Sun Oct 8 17:39:52 2023 +0200

    Bump GCC to 13.2.0

    This commit bumps GCC from 12.2.0 to 13.2.0 and adjusts the allowlist
    accordingly (for glibc and newlib multilib builds).

    Signed-off-by: Christoph Müllner <christoph.muellner@vrull.eu>

commit f7186fa52b4011684408858987840950e3e93417
Author: Tommy Murphy <tommy_murphy@hotmail.com>
Date:   Sun Oct 8 20:18:06 2023 +0100

    Bump riscv-gnu-toolchain spike submodule to latest master commit d1efcdf

...
```
In which case this might do what you want:
```
git checkout f7186fa52b4011684408858987840950e3e93417
./configure ...
make ...
```

jesseopdenbrouw, 2023-12-24:
@TommyMurphyTM1234 This works. thank you.

jesseopdenbrouw, 2023-12-24:
Closed this issue

closed 2023-12-24 by jesseopdenbrouw

TommyMurphyTM1234, 2023-12-24:
> @TommyMurphyTM1234 This works. thank you.

You're welcome.