riscv-collab/riscv-gnu-toolchain #1397: Build toolchain with rv32imafc_zba_zbb_zbc_zbs  newlib error： unrecognizable insn

naibiaozhou, 2023-12-26:
I build toolchain with config：
 `  ./configure --prefix=~/opt/riscv --with-arch=rv32imafc_zba_zbb_zbc_zbs --with-abi=ilp32f  --enable-debug-info  --enable-qemu-system --enable-llvm --with-sim=qemu`

build failed with message:
   /home/xxx/workspace/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c:1413:1: error: unrecognizable insn:
```
 1413 | }
      | ^
(insn 440 439 441 48 (set (reg:SI 205)
        (const_int 2048 [0x800])) "/home/xxx/workspace/riscv-gnu-toolchain/newlib/newlib/libc/stdlib/ldtoa.c":1304:15 -1
     (nil))
```
during RTL pass: vregs

Is this issue cause by B instruction?

Thanks

TommyMurphyTM1234, 2023-12-26:
Are you using old sources by any chance?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1236

I cannot reproduce this using the latest sources from the repo:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1397
cd issue-1397
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imafc_zba_zbb_zbc_zbs --with-abi=ilp32f  --enable-debug-info  --enable-qemu-system --enable-llvm --with-sim=qemu
make 2>&1 | tee build.log

...

./installed-tools/bin/riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./installed-tools/bin/riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/issue-1397/installed-tools/libexec/gcc/riscv32-unknown-elf/13.2.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/user/Downloads/issue-1397/gcc/configure --target=riscv32-unknown-elf --prefix=/home/user/Downloads/issue-1397/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gc891d8dc23e --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/issue-1397/installed-tools/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=ilp32f --with-arch=rv32imafc_zba_zbb_zbc_zbs --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os  -g  -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os  -g  -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (gc891d8dc23e)
```
Please clarify your full build steps - i.e. `git clone`, `configure` and `make` commands used.
And also capture, compress and attach your full config and build logs.

naibiaozhou, 2023-12-26:
Thanks   
The problem solved!

closed 2023-12-26 by naibiaozhou

TommyMurphyTM1234, 2023-12-26:
> Thanks The problem solved!

What was it and how did you solve it?
Such information may help others to avoid similar issues in the future.