riscv-collab/riscv-gnu-toolchain #1398: when install dir is changed in Makefile, `make` is not working

omerguzelelectronicguy, 2023-12-26:
When I compile the toolchain it is directly installed in the given directory. I wanted to move the executables to another directory. For example, I can do it in riscv-isa-sim by changing makefile and running `make install` however I couldn't make it in this repo. Could anyone help me with how to make it?

I thought directly move the binaries however I afaraid the corruption because of the paths. I also don't know what I need to move exactly ( executables and libraries others?).

TommyMurphyTM1234, 2023-12-26:
> When I compile the toolchain it is directly installed in the given directory. I wanted to move the executables to another directory. For example, I can do it in riscv-isa-sim by changing makefile [...]

Specify your desired install directory using `configure --prefix=<install-dir?`?

> I thought directly move the binaries however I afaraid the corruption because of the paths. 

What corruption are you referring to here?

Copying the toolchain top level directory works fine for me - e.g.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1398
cd issue-1398
./configure --prefix=`pwd`/installed-tools
make

cp -r installed-tools installed-tools-copy
```
Please be more specifc about what actual problems you face when copying/moving the toolchain.

TommyMurphyTM1234, 2023-12-26:
This worked for me:

Create a simple test program:
```
cd ~/Downloads
vi hello.c 

...

// hello.c
#include <stdio.h>

int main(int argc, char *argv[])
{
    printf("Hello, World!\n");
    return 0;
}

```
Build the toolchain:
```
cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1398
cd issue-1398
./configure --prefix=`pwd`/installed-tools
make
```
Compile the test program:
```
./installed-tools/bin/riscv64-unknown-elf-gcc -v ~/Downloads/hello.c -o hello
Using built-in specs.
COLLECT_GCC=./installed-tools/bin/riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/issue-1398/installed-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/Downloads/issue-1398/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/Downloads/issue-1398/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gc891d8dc23e --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/issue-1398/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (gc891d8dc23e)
COLLECT_GCC_OPTIONS='-v' '-o' 'hello' '-mtune=rocket' '-march=rv64imafdc_zicsr' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr'
 /home/user/Downloads/issue-1398/installed-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/cc1 -quiet -v -imultilib . /home/user/Downloads/hello.c -quiet -dumpbase hello.c -dumpbase-ext .c -mtune=rocket -march=rv64imafdc_zicsr -mabi=lp64d -misa-spec=20191213 -march=rv64imafdc_zicsr -version -o /tmp/cciVLsle.s
GNU C17 (gc891d8dc23e) version 13.2.0 (riscv64-unknown-elf)
        compiled by GNU C version 11.4.0, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.1, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/home/user/Downloads/issue-1398/installed-tools/riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/home/user/Downloads/issue-1398/installed-tools/riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/include
 /home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/include-fixed
 /home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/include
End of search list.
Compiler executable checksum: 3a8d6ab0324e42b9adcd636501ec419b
COLLECT_GCC_OPTIONS='-v' '-o' 'hello' '-mtune=rocket' '-march=rv64imafdc_zicsr' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr'
 /home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/bin/as -v --traditional-format -march=rv64imafdc_zicsr -march=rv64imafdc_zicsr -mabi=lp64d -misa-spec=20191213 -o /tmp/ccKkLrnz.o /tmp/cciVLsle.s
GNU assembler version 2.41 (riscv64-unknown-elf) using BFD version (GNU Binutils) 2.41
COMPILER_PATH=/home/user/Downloads/issue-1398/installed-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/:/home/user/Downloads/issue-1398/installed-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/:/home/user/Downloads/issue-1398/installed-tools/libexec/gcc/riscv64-unknown-elf/:/home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/:/home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/:/home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/bin/
LIBRARY_PATH=/home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/:/home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/lib/:/home/user/Downloads/issue-1398/installed-tools/riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'hello' '-mtune=rocket' '-march=rv64imafdc_zicsr' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr' '-dumpdir' 'hello.'
 /home/user/Downloads/issue-1398/installed-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/collect2 -plugin /home/user/Downloads/issue-1398/installed-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/liblto_plugin.so -plugin-opt=/home/user/Downloads/issue-1398/installed-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccXnltaB.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgloss -plugin-opt=-pass-through=-lgcc --sysroot=/home/user/Downloads/issue-1398/installed-tools/riscv64-unknown-elf -melf64lriscv -o hello /home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/lib/crt0.o /home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/crtbegin.o -L/home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0 -L/home/user/Downloads/issue-1398/installed-tools/lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/lib -L/home/user/Downloads/issue-1398/inst [... cut ...] ed-tools-copy/bin/../riscv64-unknown-elf/lib /tmp/ccx5S5bt.o -lgcc --start-group -lc -lgloss --end-group -lgcc /home/user/Downloads/installed-tools-copy/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/crtend.o
COLLECT_GCC_OPTIONS='-v' '-o' 'hello' '-mtune=rocket' '-march=rv64imafdc_zicsr' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr' '-dumpdir' 'hello.'

file hello
hello: ELF 64-bit LSB executable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), statically linked, not stripped
```

TommyMurphyTM1234, 2023-12-26:
The issue title:

> when install dir is changed in Makefile, make is not working

doesn't really correlate to the issue description:

> When I compile the toolchain it is directly installed in the given directory. I wanted to move the executables to another directory. For example, I can do it in riscv-isa-sim by changing makefile [...]

omerguzelelectronicguy, 2023-12-27:
I'm sorry for not getting back to you sooner. let me explain my problem in more detail.

I am working in a directory called `my_dir`. I installed the toolchain to `my_dir` by giving `--prefix=/home/omer/mydir`. Checked it. It is working fine. Then I wanted to install the toolchain to `/opt/riscv` I changed the install dir in the makefile from `/home/omer/mydir` to `/opt/riscv` and run make again. It returned "make: Nothing to be done for `all'". So, couldn't install the compiled toolchain to `/opt/riscv` in this way. (It worked when I installed spike again to a different directory)

I understand that `my_dir` is replaceable and can work anywhere when I copy it. So, I can copy the insiders of `my_dir` to `/opt/riscv/.`. However, I already installed the spike to `/opt/riscv` so there are already `bin`, `lib`, `include` directories that created by spike installation. I was afraid to harm them if I do it manually(I didn't try).

omerguzelelectronicguy, 2023-12-27:
When I was installing spike, `make install` was worked when I changed the install dir inside the `Makefile`. The compiled bin lib and include dirs are added to /opt/riscv/.` It doesn't work in the same in this repo. That is why I wrote the title like that. Excuse me if I make you confused.

TommyMurphyTM1234, 2023-12-27:
> I am working in a directory called `my_dir`. I installed the toolchain to `my_dir` by giving `--prefix=/home/omer/mydir`. Checked it. It is working fine. Then I wanted to install the toolchain to [...]

It's better not to manually edit the generated `Makefile`.
It's better to use `configure --prefix=<preferred-install-dir>`.

> It returned "make: Nothing to be done for `all'". 

You need to do `make distclean` first to clean out the previously built artifacts.
But if you're only changing the target install ("prefix") directory then you can just copy the toolchain as I mentioned earlier.
If you are copying to a "system" directory such as `/opt/riscv` which may be shared by multiple users then you might also want to change the file attributes (e.g. owner and group).

TommyMurphyTM1234, 2023-12-27:
Closing as there is no problem identified here with the `riscv-gnu-toolchain`.

TommyMurphyTM1234, 2023-12-27:
See also this issue:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1196

closed 2023-12-27 by TommyMurphyTM1234

TommyMurphyTM1234, 2023-12-27:
> When I was installing spike, `make install` was worked when I changed the install dir inside the `Makefile`. The compiled bin lib and include dirs are added to /opt/riscv/.` It doesn't work in the same in this repo. That is why I wrote the title like that. Excuse me if I make you confused.

I guess that the Spike `Makefile` has something that detects a change to the target install directory as an indication that it needs to do something (specifically do the "install" copying of build artefacts) but the `riscv-gnu-toolchain` one does not. Maybe some or most other projects also have that capabilitity but I supect that because `riscv-gnu-toolchain` is basically an integration of multiple other (mostly) upstream projects (GCC, GDB, Bimutils, glibc, Newlib etc.) it does not do this.

But, in any case, an already built/installed `riscv-gnu-toolchain` can safely be copied to another install directory on the same machine as mentioned earlier.