riscv-collab/riscv-gnu-toolchain #1400: I can "make command" for pk

drkome, 2023-12-31:
I made own processor with verilog and I need to test it with model processor. So I need to install spike pk and riscv32 gnu toolchain. But I cant install pk. if I try to write "make", you see that the following outputs.

gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized argument in option ‘-mabi=ilp32’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
make: *** [Makefile:336: file.o] Error 1

I want to exlainp detaily. First I builded gcc with the following commands.

./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
make
make linux

then I have installed spike with

../configure --prefix=/opt/riscv
make

then I have tried  pk with

 ../configure --prefix=/opt/riscv --with-arch=rv32i
make

but I just saw error I mention above. How can I fix this?

TommyMurphyTM1234, 2023-12-31:
What if you simply do the following?
With the commands below you should end up with all of the executables for the RISC-V bare-metal/Newlib, RISC-V Linux/Glibc and Spike/pk in `/opt/riscv/bin`.
The `--with-sim` option is described here: 

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#test-suite

```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1400
cd issue-1400
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d --with-sim=spike
make
make linux
make build-sim
```
> I made own processor with verilog and I need to test it with model processor. So I need to install spike pk and riscv32 gnu toolchain. 
> ...
> I want to exlainp detaily. First I builded gcc with the following commands.
> ```
> ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d 
> make 
> make linux
> ```

Why are you building both the bare-metal/Newlib **and** Linux/Glibc toolchains? 
Do you actually need both?
Usually one only needs one other the other.

> then I have installed spike with
> ```
> ../configure --prefix=/opt/riscv
> make
> ```
> then I have tried pk with
> ```
> ../configure --prefix=/opt/riscv --with-arch=rv32i 
> make
> ```
It's unclear what directory the above commands are being executed.
You need to explain in detail ALL commands that you use and clarify any directory changes along the way.
It's also unclear why you're using `--with-arch=rv32i` here when your target and tools seem to me `rv32gc`?

TommyMurphyTM1234, 2023-12-31:
This suggests that when building something (it's not clear what due to the lack of a clear/complete command transcript) it's passing RISC-V GCC options but invoking the wrong compiler - likely the native (x86_64?) one rather than the RISC-V cross compiler.
> ```
> gcc: error: unrecognized argument in option ‘-mcmodel=medany’
> gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
> gcc: error: unrecognized argument in option ‘-mabi=ilp32’
> gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
> make: *** [Makefile:336: file.o] Error 1
> ```

drkome, 2023-12-31:
I did now. I installed spike and pk now.  but now if  Itry to spike pk  I saw  following output.
"Error: cannot execute 32-bit program on RV64 hart."
  and I still reinstall spike pk and gcc  like you said. 
  Now I try to fix this "Error: cannot execute 32-bit program on RV64 hart.". 
and sorry dude. I am so rookie about this topic. I feel that I asking so basic things. ssorry for this.

drkome, 2023-12-31:
> This suggests that when building something (it's not clear what due to the lack of a clear/complete command transcript) it's passing RISC-V GCC options but invoking the wrong compiler - likely the [...]

I understand this now.  Because I forgot add opt/riscv/bin to add PATH. So I have this error. I reinstall this tool a lot of times.

TommyMurphyTM1234, 2023-12-31:
> but now if Itry to spike pk I saw following output. "Error: cannot execute 32-bit program on RV64 hart." and I still reinstall spike pk and gcc like you said. Now I try to fix this "Error: cannot execute 32-bit program on RV64 hart.". 

Spike defaults to an RV64 target unless told otherwise:

* https://github.com/riscv-software-src/riscv-isa-sim/blob/master/configure#L6576

You probably need to tell it that you want RV32 - e.g.
```
spike --isa=RV32IMAFDC pk ...
```

> I understand this now. Because I forgot add opt/riscv/bin to add PATH. So I have this error. I reinstall this tool a lot of times.

I don't understand what you mean - are you saying that you still have a problem?

Note that this did not work for me:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1400
cd issue-1400
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d --with-sim=spike
make
make linux
make build-sim
```
because when building Spike I got this:
```
riscv32-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv32gc -mabi=ilp32f -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I/home/user/Downloads/riscv-gnu-toolchain/pk/pk -I/home/user/Downloads/riscv-gnu-toolchain/pk/bbl -I/home/user/Downloads/riscv-gnu-toolchain/pk/softfloat -I/home/user/Downloads/riscv-gnu-toolchain/pk/dummy_payload -I/home/user/Downloads/riscv-gnu-toolchain/pk/machine -I/home/user/Downloads/riscv-gnu-toolchain/pk/util -c /home/user/Downloads/riscv-gnu-toolchain/pk/pk/pk.c
riscv32-unknown-elf-gcc -Wl,--build-id=none -nostartfiles -nostdlib -static  -march=rv32gc -mabi=ilp32f -fno-stack-protector -o pk pk.o -L.  -lpk  -lmachine  -lsoftfloat  -lutil -lgcc -Wl,--defsym=MEM_START=0x80000000,-T,/home/user/Downloads/riscv-gnu-toolchain/pk/pk/pk.lds
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_divdi3.o): can't link double-float modules with single-float modules
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_divdi3.o)
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_moddi3.o): can't link double-float modules with single-float modules
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_moddi3.o)
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_udivdi3.o): can't link double-float modules with single-float modules
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_udivdi3.o)
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_umoddi3.o): can't link double-float modules with single-float modules
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_umoddi3.o)
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_clz.o): can't link double-float modules with single-float modules
/home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/user/Downloads/riscv-gnu-toolchain/installed-tools/lib/gcc/riscv32-unknown-elf/13.2.0/libgcc.a(_clz.o)
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:336: pk] Error 1
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-pk32'
make: *** [Makefile:887: stamps/build-pk32] Error 2
```
I think I would need to change the configure command to:
```
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32f --with-sim=spike
```
or maybe build a multilib bare-metal/Newlib toolchain.

drkome, 2023-12-31:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1400
cd issue-1400
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d --with-sim=spike
make
make linux
make build-sim
```

I dont know how. But I worked for me. I did this now.  But now cant install pk idk why.
I just saw this following outputs.

```
/opt/riscv/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: ./libpk.a(file.o): can't link soft-float modules with double-float modules
/opt/riscv/lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file ./libpk.a(file.o)
collect2: error: ld returned 1 exit status
make: *** [Makefile:336: pk] Error 1
kamer@DESKTOP-7ULN6M1:~/riscv-gnu-toolcha
```

**I don't understand what you mean - are you saying that you still have a problem?** <= No No,  I just try to say how I fixed this.

**Note that this did not work for me:**   <= Worked for me. But sometimes I dont use sudo . And I have get error bcs they dont write opt's folder.   Even if this is not the error, could something like this happen? Or maybe it can't be installed because you didn't specify the file in the path? I was having problems because I used "sudo make" while installing pk. Because riscv32-unknown-elf-gcc was not working with sudo. I made such strange mistakes. But still somewhere pk causes problems.

TommyMurphyTM1234, 2023-12-31:
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1400
> cd issue-1400
> ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d --with-sim=spike
> make
> make linux
> make [...]
That's the same issue that I encountered.
The problem is that it's trying to build pk using `-march=rv32gc -mabi=ilp32f` but the toolchain was built with `--with-arch=rv32gc --with-abi=ilp32d` and the different ABIs result in this incompatibility.
I'm trying a multilib build to see if that helps.

> **Note that this did not work for me:** <= Worked for me. But sometimes I dont use sudo . And I have get error bcs they dont write opt's folder. Even if this is not the error, could something like [...]

The issue has nothing to do with `sudo`. I don't use it because I am installing to a local rather than a system directory:
```
./configure --prefix=`pwd`/installed-tools
make
```

You never addressed this question - maybe you clarify if/why you need both toolchains (bare-metal/Newlib and Linux/Glibc)?

> Why are you building both the bare-metal/Newlib and Linux/Glibc toolchains?
> Do you actually need both?
> Usually one only needs one other the other.

TommyMurphyTM1234, 2024-01-01:
This worked for me:
```
// hello.c
#include <stdio.h>

int main(int argc, char *argv[])
{
  printf("hello, world!\n");
  return 0;
}
```
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1400
cd issue-1400
./configure --prefix=`pwd`/installed-tools --with-sim=spike --enable-multilib
make linux && make build-sim

./installed-tools/bin/riscv64-unknown-linux-gnu-gcc -march=rv32gc -mabi=ilp32d -static hello.c -o hello

./installed-tools/bin/spike --isa=rv32gc pk32 hello
bbl loader
hello, world!
```

TommyMurphyTM1234, 2024-01-01:
I presume that my previous post addresses the issues of how to build, install and use (for `rv32gc/ilp32d`) Spike/pk so I am closing this issue.

closed 2024-01-01 by TommyMurphyTM1234