riscv-collab/riscv-gnu-toolchain #1401: How to write a clang config file equals to nano.spec ?

naibiaozhou, 2024-01-02:
I want to use newlib_nano for riscv32 ,. I add -lc_nano -lg_nano -lm_nano to cflags, but when link the lib -lc is append by clang, this makes the elf file big than gcc .
the link command:
` "/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld" -m elf32lriscv -X --gc-sections -lgcc -lc_nano -lgcc --start-group -lgcc -lc_nano -lgloss_nano --end-group --static --noinhibit-exec -T /home/xxx/workspace/aaa/build/aaa.ld -Map aaa.map CMakeFiles/EMPS.elf.dir/benchmark/main.c.obj CMakeFiles/aaa.elf.dir/aaaVerInfo.c.obj lib/libbbb.a lib/libbasic.a lib/libstubs.a lib/libos.a lib/libdrivers.a lib/liblog.a lib/libbenchmark.a lib/libbasic.a lib/libdrivers.a lib/libbbb.a lib/libos.a -lbasic -L/home/xxx/workspace/emps/3rd/lib -L/home/xxx/workspace/emps/build/lib -L/home/xxx/opt/riscv_temp/bin/../lib/gcc/riscv32-unknown-elf/13.2.0 -L/home/xxx/opt/riscv_temp/bin/../lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/lib _--start-group -lc -lgloss --end-group -lgcc_ -o aaa.elf`

after I add -nostdlib option the lib is right,bug many symbol like printf can not found :

` "/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld" -m elf32lriscv -X --gc-sections -lgcc -lc_nano -lgcc --start-group -lgcc -lc_nano -lgloss_nano --end-group --static --noinhibit-exec -T /home/xxx/workspace/emps/build/aaa.ld -Map aaa.map CMakeFiles/aaa.elf.dir/benchmark/main.c.obj CMakeFiles/aaa.elf.dir/aaaVerInfo.c.obj lib/libbbb.a lib/libbasic.a lib/libstubs.a lib/libos.a lib/libdrivers.a lib/liblog.a lib/libbenchmark.a lib/libbasic.a lib/libdrivers.a lib/libbbb.a lib/libos.a -lbasic -L/home/xxx/workspace/emps/3rd/lib -L/home/xxx/workspace/emps/build/lib -L/home/xxx/opt/riscv_temp/bin/../lib/gcc/riscv32-unknown-elf/13.2.0 -L/home/xxx/opt/riscv_temp/bin/../lib/gcc/riscv32-unknown-elf/13.2.0/../../../../riscv32-unknown-elf/lib -o aaa.elf

error message:
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: /home/xxx/workspace/aaa/platform/basic/src/basic_intr.c:177: undefined reference to `printf'
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: lib/libbasic.a(basic_intr.c.obj): in function `basic_intr_enable_irq':
/home/xxx/workspace/emps/platform/basic/src/basic_intr.c:269: undefined reference to `printf'
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: lib/libdrivers.a(uart_driver.c.obj):/home/xxx/workspace/aaa/drivers/src/uart/uart_driver.c:158: more undefined references to `printf' follow
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: lib/libbenchmark.a(core_main.c.obj): in function `CoreMarkMain':
/home/xxx/workspace/aaa/benchmark/coremark/src/core_main.c:250: undefined reference to `__ltdf2'
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: /home/xxx/workspace/aaa/benchmark/coremark/src/core_main.c:260: undefined reference to `__fixunsdfsi'
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: /home/xxx/workspace/aaa/benchmark/coremark/src/core_main.c:405: undefined reference to `__gtdf2'
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: /home/xxx/workspace/aaa/benchmark/coremark/src/core_main.c:407: undefined reference to `__floatunsidf'
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: /home/xxx/workspace/aaa/benchmark/coremark/src/core_main.c:408: undefined reference to `__divdf3'
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: /home/xxx/workspace/aaa/benchmark/coremark/src/core_main.c:416: undefined reference to `__ltdf2'
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: /home/xxx/workspace/aaa/benchmark/coremark/src/core_main.c:455: undefined reference to `__floatunsidf'
/home/xxx/opt/riscv_temp/bin/riscv32-unknown-elf-ld: /home/xxx/workspace/aaa/benchmark/coremark/src/core_main.c:456: undefined reference to `__divdf3'
`
Anybody can help me?
Thanks

TommyMurphyTM1234, 2024-01-02:
This is a question for a Clang/LLVM forum, not here.

* https://discourse.llvm.org/c/clang/6

closed 2024-01-02 by TommyMurphyTM1234

naibiaozhou, 2024-01-03:
OK thanks.