riscv-collab/riscv-gnu-toolchain #1402: When E is enabled, the CSRs in "CSR_CLASS_I" will be disassembled into .insn.

muxiff, 2024-01-03:
Hi!
I found that when the E extension is enabled, all operations on the CSRs in `CSR_CLASS_I` can be compiled normally, but during disassembly or gdb debugging, they will be disassemmbled into .insn, which can easily be misleading. 

Therefore, do we have a way to make disassembly execute correctly, or consider the compatibility of these CSRs between i and e?

cmuellner, 2024-01-03:
What would you expect as "execute correctly"? And what would be the use-case of that?

The idea here is that if you have instructions that can't be disassembled, you get at least the opcode as `.insn`.
This is, e.g., valuable in the case of disassembling binaries with new instructions with an old version of Binutils.

muxiff, 2024-01-03:
I used the latest released toolchain.

The case can be like :
```
csrw mstatus, 0x5
```
Complied with `-march=rv32e -mabi=ilp32e`, and disassembled to:
```
00000000 <.text>:
   0:   3002d073                .insn   4, 0x3002d073
```

Complied with `-march=rv32i -mabi=ilp32`, disassembled to:
```
00000000 <.text>:
0:   3002d073                csrw    mstatus,5
```
This is what I want.

cmuellner, 2024-01-03:
Note, that `csrw` is part of Zicsr, so `rv32e` would not be enough.

I just checked upstream, and I believe there we have the correct behaviour:

```
$ riscv64-unknown-linux-gnu-as -march=rv32e -mabi=ilp32e foo.s -o foo.o ; ./install/bin/riscv64-unknown-linux-gnu-readelf -A foo.o
foo.s: Assembler messages:
foo.s:1: Error: unrecognized opcode `csrw mstatus,0x5', extension `zicsr' required
readelf: Error: 'foo.o': No such file
```
```
$ riscv64-unknown-linux-gnu-as -march=rv32e_zicsr -mabi=ilp32e foo.s -o foo.o ; ./install/bin/riscv64-unknown-linux-gnu-readelf -A foo.o
Attribute Section: riscv
File Attributes
  Tag_RISCV_arch: "rv32e1p9_zicsr2p0"
  Tag_RISCV_priv_spec: 1
  Tag_RISCV_priv_spec_minor: 11
```
```
$ riscv64-unknown-linux-gnu-objdump -d foo.o

foo.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <.text>:
   0:   3002d073                csrwi   mstatus,5
```

TommyMurphyTM1234, 2024-01-03:
> I just checked upstream, and I believe there we have the correct behaviour:

Not just upstream - the latest CI release build exhibits the same (correct) behaviour as you describe in your previous post.

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases

Closing this as the issue has been explained.

closed 2024-01-03 by TommyMurphyTM1234

fanghuaqi, 2024-01-04:
Hi @cmuellner @TommyMurphyTM1234 , we are compiling the toolchain `--with-isa-spec=2.2` which the `zicsr` is part of `I` or `E`, so we are expecting when we compile code using `rv32ic` or `rv32ie` it will default include `zicsr` extension, and I see this commit https://github.com/bminor/binutils-gdb/commit/7671eff8f08de314d8c9837225eba95ed5ea053b add an imply rule for `I` extension to include `zicsr` but for `E` extension, it didn't handle it.

[image]

So we add a implicit rule for `E` extension to include `Zicsr`, then it fixed this issue reported.

[image]

cmuellner, 2024-01-04:
> Hi @cmuellner @TommyMurphyTM1234 , we are compiling the toolchain `--with-isa-spec=2.2` which the `zicsr` is part of `I` or `E`

RV32E and RV64E were both ratified in 2023 (see https://wiki.riscv.org/display/HOME/Specification+Status).

With ISA spec 2.2 you most likely refer to a 2019 release (https://github.com/riscv/riscv-isa-manual/releases/tag/archive) of the unpriv spec (https://github.com/riscv/riscv-isa-manual/releases/download/archive/riscv-spec-v2.2.pdf).
When looking at this document, page 3 clearly states that RV32E is not frozen.

So, I can't see a ratified specification that claims that RV32E includes Zicsr instructions.
Therefore, I also don't expect (and see) this implemented in Binutils.

Of course, you can patch Binutils to include Zicsr for RV32E.
But I doubt that this will be accepted upstream (there is no ratified specification).
This project here will not accept anything beyond branches to upstream repos.

Why not simply use a recent toolchain and set `-march=rv32e_zicsr` to get what you need?

fanghuaqi, 2024-01-05:
> Why not simply use a recent toolchain and set `-march=rv32e_zicsr` to get what you need?

Hi @cmuellner ， thanks , we just want to be back compatible with previous release, thank you.

TommyMurphyTM1234, 2024-01-07:
> > Why not simply use a recent toolchain and set `-march=rv32e_zicsr` to get what you need?
> 
> Hi @cmuellner ， thanks , we just want to be back compatible with previous release, thank you.

A complicating factor here is that you're trying to leverage earlier versions of the tools to support an extension (`E`) that was still draft and not ratified at the time.