riscv-collab/riscv-gnu-toolchain #1403: Compile functions from standard library for running on bare-metal RISCV processor.

DiamantisPa, 2024-01-03:
I am trying to compile with the `riscv-gnu-toolchain` some C++ code containing different functions from the standard  library like malloc(), printf(), memset() etc, using my own linker script and some code for bootstraping. My goal is to run the produced assembly on a bare-metal RISCV processor.

The code compiles okay ( i think ) but when i try to run the binary (in order to verify that its valid) i get the error `core: 4 byte read to unmapped address 0xfffffff4 at 0x465fc`.

This is my custom linker script (lscript):
```
MEMORY
{
ram : ORIGIN = 0x0000, LENGTH = 0x1000000
stack : ORIGIN = ORIGIN(ram) + LENGTH(ram), LENGTH = 0x00200000
}
SECTIONS
{
.text : { *(.text*) } > ram
.bss : { *(.bss*) } > ram
}

__stack_top = ORIGIN(ram) + LENGTH(ram);

_HeapStart = .;
	__HeapStart = .;
	. +=0x00200000 ;

_end = .; PROVIDE (end = .);
```

This the the boostrap code i use (bootstrap.s):
```
.globl _start
_start:
li a0,1000000
mv sp,a0
jal main
hang: j hang
```

To compile the code i run:
`riscv32-unknown-elf-g++ -march=rv32im -mabi=ilp32 -T lscript bootstrap.s -nostartfiles -fno-common -static -fno-builtin-printf -specs=nosys.specs $(INC_PARAMS) dcnn_lenet_tb.cpp -o dcnn_lenet_tb.o`

This is the compilation output:

[image]

When i try to ran the binary file `dcnn_lenet_tb.o` with `riscv32-unknown-elf-run -t --trace-file run.log --trace-register --trace-memory dcnn_lenet_tb.o`, i get following output

> core: 4 byte read to unmapped address 0xfffffff4 at 0x465fc
program stopped with signal 11 (Segmentation fault).

I think it has something to do with the stack definition in the linker script (?), although not sure. Could anyone provide some guidance ?

Thanks a lot.

TommyMurphyTM1234, 2024-01-03:
No reproducible test case provided and, in any case, this is a question about using the toolchain as opposed to reporting a suspected problem/bug.
Such queries are better suited to other forums such as the RISC-V mailing list, StackOverflow etc.

TommyMurphyTM1234, 2024-01-03:
BTW - the `riscvXX-unknown-elf-run` simulator has various issues/limitations - you are probably better off using something like Spike or QEMU instead.

closed 2024-01-03 by TommyMurphyTM1234

TommyMurphyTM1234, 2024-01-03:
Similar issue posted here but the original poster never followed up:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1083

DiamantisPa, 2024-01-03:
> No reproducible test case provided and, in any case, this is a question about using the toolchain as opposed to reporting a suspected problem/bug. Such queries are better suited to other forums such as the RISC-V mailing list, StackOverflow etc.

I added the test case here https://github.com/DiamantisPa/riscv-toolchain-test , if anyone is interested in looking at it. I will try another forum, thanks.

> BTW - the `riscvXX-unknown-elf-run` simulator has various issues/limitations - you are probably better off using something like Spike or QEMU instead.

Sure, but the simulation from `riscvXX-unknown-elf-run` is the closest to running the produced assembly on bare-metal. I don't know about `QEMU` but `Spike` adds its own logic, for example for exception handling, etc.

> Similar issue posted here but the original poster never followed up:
> 
> * [riscv64-unknown-linux-gnu-run Segmentation fault？ #1083](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1083)
> 
> And a similar issue here:
> 
> * https://www.reddit.com/r/Zig/comments/rjpxe6/how_to_build_riscv64imacunknownnoneelf_target_and/

Thanks. Had a look at these  posts before posting here, but the info in them didn't help much.

TommyMurphyTM1234, 2024-01-03:
Why don't you debug your code to see what's happening and where/how/why things go wrong?

DiamantisPa, 2024-01-04:
But it's not about the code is about the compilation with the `riscv-gnu-toolchain` for bare metal. The code works fine but it creates a `lw` instruction with a target address of  `0xfffffff4` which is huge. Spike will throw a page_fault exception and will handle it by itself, but that's not possible in bare-metal.

I don't know if this is a issue with the toolchain or its by design (See: https://groups.google.com/a/groups.riscv.org/g/isa-dev/c/xmDTqT_P8Bo/m/zkheMZ7nAQAJ?pli=1) and if it's possible somehow to overcome with different linker script or some other way (on the compilation level) in order to be able to use it with bare metal.

Anyways, thanks for your time.