riscv-collab/riscv-gnu-toolchain #1404: symtab.h:307:23: Compilation fails on Mac M3 and Intel (error: expected unqualified-id style = split_style::DOT;)

fschuetz, 2024-01-05:
Hi

I am trying to compile the riscv-toolchain on a mac M3 and an older Intel machine. I installed the prerequisites with homebrew as described in the readme,  except python3, as I am using python 3.12.0 from pyenv. A while in, on both machines, compilation fails with the follwoing error:

```
In file included from /Volumes/Development/toolchains/riscv-gnu-toolchain/gdb/gdb/m2-exp.y:42:
In file included from /Volumes/Development/toolchains/riscv-gnu-toolchain/gdb/gdb/language.h:26:
In file included from /Volumes/Development/toolchains/riscv-gnu-toolchain/gdb/gdb/symtab.h:39:
/Volumes/Development/toolchains/riscv-gnu-toolchain/gdb/gdb/split-name.h:34:3: error: expected identifier
  DOT,
  ^
m2-exp.c:163:13: note: expanded from macro 'DOT'
#define DOT 302
            ^
In file included from /Volumes/Development/toolchains/riscv-gnu-toolchain/gdb/gdb/m2-exp.y:42:
In file included from /Volumes/Development/toolchains/riscv-gnu-toolchain/gdb/gdb/language.h:26:
/Volumes/Development/toolchains/riscv-gnu-toolchain/gdb/gdb/symtab.h:307:23: error: expected unqualified-id
        style = split_style::DOT;
                             ^
m2-exp.c:163:13: note: expanded from macro 'DOT'
#define DOT 302
            ^
2 errors generated.
make[3]: *** [m2-exp.o] Error 1
make[2]: *** [all-gdb] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-gdb-newlib] Error 2
```

Before starting `make`, I configured the build with `./configure --prefix=/opt/riscv --enable-multilib` and also set permissions to /opt/riscv accordingly.

Can you point me where I might go wrong?

Thanks

TommyMurphyTM1234, 2024-01-05:
> I am trying to compile the riscv-toolchain on a mac M3 and an older Intel machine. 

What OS is the Intel machine running? 

I've never seen this issue when building on x86_64 Linux.

TommyMurphyTM1234, 2024-01-05:
> ```
> /Volumes/Development/toolchains/riscv-gnu-toolchain/gdb/gdb/split-name.h:34:3: error: expected identifier
>   DOT,
>   ^
> m2-exp.c:163:13: note: expanded from macro 'DOT'
> #define DOT 302
>             ^
> ...
> /Volumes/Development/toolchains/riscv-gnu-toolchain/gdb/gdb/symtab.h:307:23: error: expected unqualified-id
>         style = split_style::DOT;
>                              ^
> m2-exp.c:163:13: note: expanded from macro 'DOT'
> #define DOT 302
>             ^
> ```
Similar issue here?

* https://www.mail-archive.com/bug-binutils@gnu.org/msg41767.html

TommyMurphyTM1234, 2024-01-05:
What version of the sources are you building? In other words - what is your `git clone https://github.com/riscv-collab/riscv-gnu-toolchain` command? Are you building from the head of the repo or some older branch/tag/commit?

Please capture, compress and attach your build log - e.g.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1404
cd issue-1404
./configure ...
make 2>&1 | tee build.log
```

fschuetz, 2024-01-05:
> > I am trying to compile the riscv-toolchain on a mac M3 and an older Intel machine.
> 
> What OS is the Intel machine running?
> 
> I've never seen this issue when building on x86_64 Linux.

It is also a mac (older with intel processor, not arm), running macOS Sonoma 14.2.1. 

As for the similarity of the issue: I am not sure. To me this looks a colliding define with a type. The issue looks like a type definition on standard types issue at first glance, but it might be related. I need to investigate. Any further advice apreciated.

TommyMurphyTM1234, 2024-01-05:
> As for the similarity of the issue: I am not sure. 

The exact same errors as you get occur here:

* https://www.mail-archive.com/bug-binutils@gnu.org/msg41767.html

```
and the following on macOS(Intel) Catalina 10.15.7,

  CXX    m2-exp.o
In file included from m2-exp.y:42:
In file included from ./language.h:26:
In file included from ./symtab.h:39:
./split-name.h:34:3: error: expected identifier
  DOT,
  ^
m2-exp.c:163:13: note: expanded from macro 'DOT'
#define DOT 302
            ^
In file included from m2-exp.y:42:
In file included from ./language.h:26:
./symtab.h:307:23: error: expected unqualified-id
        style = split_style::DOT;
                             ^
m2-exp.c:163:13: note: expanded from macro 'DOT'
#define DOT 302
            ^
2 errors generated.
make[2]: *** [m2-exp.o] Error 1
```

fschuetz, 2024-01-05:
Oh, I overread this. I will look into it. I attached the build log to this post.
[build.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13847220/build.log)

TommyMurphyTM1234, 2024-01-05:
Maybe try this to see if it still happens [with the latest GDB master branch sources](https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#use-source-tree-other-than-riscv-gnu-toolchain)?
```
cd ~/Downloads
git clone https://sourceware.org/git/binutils-gdb.git gdb-master
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1404
cd issue-1404
./configure ... --with-gdb-src=$HOME/Downloads/gdb-master
make
```

fschuetz, 2024-01-07:
This fixes the build. Thank you.

closed 2024-01-07 by fschuetz

TommyMurphyTM1234, 2024-01-07:
> This fixes the build. Thank you.

You mean building like this fixes things?

> Maybe try this to see if it still happens [with the latest GDB master branch [...]

fschuetz, 2024-01-08:
> You mean building like this fixes things?

Yes (and adding libmpfr (brew install mpfr, and setting LDFLAGS and CPPFLAGS accordingly as a dependency for  binutils-gdb and doing all of that in a case sensitive file system of course). Compilation goes through without error (while compiling without binutils-gdb still fails, I cross checked).

What I noted is that the build supports the following architectures:
```
rv32i/ilp32;@march=rv32i@mabi=ilp32
rv32im/ilp32;@march=rv32im@mabi=ilp32
rv32iac/ilp32;@march=rv32iac@mabi=ilp32
rv32imac/ilp32;@march=rv32imac@mabi=ilp32
rv32imafc/ilp32f;@march=rv32imafc@mabi=ilp32f
rv64imac/lp64;@march=rv64imac@mabi=lp64
rv64imafdc/lp64d;@march=rv64imafdc@mabi=lp64d
```
I assume this is the default multilib setting and additional architectures / abis can be configured (I need rv32ec architecture with lp32e abi).

TommyMurphyTM1234, 2024-01-09:
> while compiling without binutils-gdb still fails

This doesn't make sense to me. The toolchain requires binutils to be built. GDB is optional and can be omitted using `configure ... --disable-gdb` if preferred.

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/8c969a9efe68a811cf524174d25255632029f3d3/configure#L1341
 
> What I noted is that the build supports the following architectures:
>
> ```
> rv32i/ilp32;@march=rv32i@mabi=ilp32
> rv32im/ilp32;@march=rv32im@mabi=ilp32
> rv32iac/ilp32;@march=rv32iac@mabi=ilp32 [...]

I presume that you used `configure ... --enable-multilib` in which case you got the default set of multilibs for the bare metal toolchain?

> and additional architectures / abis can be configured (I need rv32ec architecture with lp32e abi).

For the bare metal toolchain you can use `configure ... --with-multilib-generator=...` to specify a custom set of multilibs.

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#build-with-customized-multi-lib-configure

But if you need to target only `rv32ec/ilp32e` (I presume that `lp32e` above was a typo on your part?) then it might be simpler to build the toolchain for that specific `arch/abi`:

* `configure ... --with-arch=rv32ec --with-abi=ilp32e`