riscv-collab/riscv-gnu-toolchain #1407: /usr/lib/riscv64-unknown-elf/bin/ld donot having lm in it.

Vinayakamk, 2024-01-13:
**ERROR IS:/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lm**

i have installed riscv gnu tool chain completely.as the photo says list of things in /opt/riscv/bin
as im in need of using this in my tool,
this happened,saying -lm is not there .

kindly help how can i add that missing package (lm) to this path /usr/lib/riscv64-unknown-elf/bin/ld

/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lm
[image]

im very new to this,sorry if you already fixed in any issues.i  expect commands to fix this error.
thank you very much in advance

TommyMurphyTM1234, 2024-01-13:
Please explain in detail how you obtained/built your toolchain and provide a small self contained reproducible test case - i.e. code and `Makefile` or GCC commands used to compile/link it.

Vinayakamk, 2024-01-13:
building my tool chain:
git clone https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain
sudo apt-get install autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev

then
./configure --prefix=/opt/riscv
make ,hre make threw me permisssion denied error ,so i did sudo make.it went well
then i did this .export PATH=$PATH:/opt/riscv/bin

for linux 

./configure --prefix=/opt/riscv
sudo make linux.

and when i get inside my required directory which has this make file.
[Makefile.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13928833/Makefile.txt)
threw me that error

TommyMurphyTM1234, 2024-01-13:
This works fine for me:
```
cd ~/Downloads

# Get/extract the latest CI build of the tools (I'm using WSL2 Ububtu 22.04.1)
# https://github.com/riscv-collab/riscv-gnu-toolchain/releases
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2023.12.20/riscv64-elf-ubuntu-22.04-gcc-nightly-2023.12.20-nightly.tar.gz
tar -xzvf riscv64-elf-ubuntu-22.04-gcc-nightly-2023.12.20-nightly.tar.gz

# Compile test code
export PATH=$HOME/Downloads/riscv/bin:$PATH
riscv64-unknown-elf-gcc test.c -o test -lm
```
```
// test.c
#include <stdio.h>

int main()
{
    float f;
    double d;

    f = 1.02;
    d = 3.14;

    d = d * f;

    printf("test = %f\n", d);

    return 0;
}
```

TommyMurphyTM1234, 2024-01-13:
> then for newlib ./configure --prefix=/opt/riscv make ,here make threw me permisssion denied error ,so i did sudo make.

That's because you generally need `root/sudo` privileges to write to a "system" directory such as `/opt`.

> for linux
> 
> ./configure --prefix=/opt/riscv sudo make linux.

Why are you building the Linux toolchain when your `Makefile` only uses the bare-metal Newlib toochain?

> then i did this .export PATH=$PATH:/opt/riscv/bin

You should probably put your toolchain at the **start** of the `PATH` in case there are other RISC-V toolchains already installed:
```
export PATH=/opt/riscv/bin:$PATH
```
What happens if you do this (replace `<your-ld-script>` with your custom linker script):
```
riscv64-unknown-elf-gcc -march=rv64imafdc -mabi=lp64 -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -fno-common -fno-builtin-printf -static -nostdlib -nostartfiles -lm -lgcc -T <your-ld-script> test.c -o test
```
Where `test.c` is:
```
// test.c
#include <stdio.h>

int main()
{
    float f;
    double d;

    f = 1.02;
    d = 3.14;

    d = d * f;

    // printf("test = %f\n", d);

    return 0;
}
```

TommyMurphyTM1234, 2024-01-13:
I think that this is simply a `PATH` issue.
You are installing your toolchain in `/opt/riscv` but the error refers to `/usr/lib/riscv64-unknown-elf/bin/ld`.
That suggests that there's another toolchain already installed that is being picked up here.
As I said above you should probably do this to ensure that only your `/opt/riscv` toolchain gets picked up:
```
export PATH=/opt/riscv/bin:$PATH
```

Vinayakamk, 2024-01-13:
yes is it worked for me.
im using riscv isa spike simulator.
but now spike is not running ,gnu tool error got fixed .
[image]

any suggestions.

thanks for your great help

TommyMurphyTM1234, 2024-01-13:
The error message is literally telling you what you're doing wrong - running Spike in 32-bit mode (`--isa=rv32`) but passing it a 64-bit (`-march=rv64imafdc -mabi=lp64`) program.

closed 2024-01-13 by TommyMurphyTM1234

Vinayakamk, 2024-01-13:
sure sir thank you very much...