riscv-collab/riscv-gnu-toolchain #1408: Question about rvv 1.0 support in objdump?

LWenH, 2024-01-15:
Hi,  riscv-gnu-toolchain developers, I have a question regarding an issue with rvv 1.0 instructions not decoding properly in objdump, but I'm not sure if this is the right place to ask this question.

I have a program that contains some rvv 1.0 instructions. When I use the riscv64-linux-gnu-objdump tool for disassembly, it disassembles and displays correctly:

[image]

However, when I directly use `objdump -d ./myprom` for disassembly, it seems unable to properly disassemble:

[image]

The objdump command I'm using is being run directly on a RISC-V Fedora operating system. I've even updated to the latest binutils tools to try disassembling, but it still fails to work properly. Is this an issue with binutils? Am I missing any parameters in objdump that are necessary for disassembling RVV code? Also, what is the relationship between riscv64-linux-gnu-objdump and objdump?  Thank you for taking the time to look into this issue.

Here are some parameters of my running environment:
| Software | Version |
|-------|-------|
| OS | Fedora Linux 38 (Thirty Eight) riscv64 |
| Kernel | 6.5.0-0.rc4.30.3.riscv64.fc39.riscv64 |
| Binutils(objdump) | 2.41-5.fc40 |
| riscv64-linux-gnu-objdump | GNU objdump version 2.39-4.fc38 |

LWenH, 2024-01-15:
This issue occurred when I was using perf to analyze the hotspots in my program, as it seems that the Linux perf tool directly uses the **objdump** utility to implement the disassembly function：

[image]

pz9115, 2024-01-15:
I think the reason is binutils version 2.39 is low for rvv support, so you may need to update your  binutils version higher than 2.40.

LWenH, 2024-01-15:
> I think the reason is binutils version 2.39 is low for rvv support, so you may need to update your binutils version higher than 2.40.

Thank you for taking the time to look at this issue. However, I have already updated my objdump to version 2.41. Could this be an issue with binutils?

pz9115, 2024-01-15:
It seems to be an issue with fedora distribution. Can you use as to compile a assemble file with rvv instruction in your environment?

like
```
vsetvli  a0, a1, e64, mf8, tu, ma
```

use as with `-c` option to assemble it and see if can get an obj file

LWenH, 2024-01-15:
Yeah, pretty weird question, I tried the following command:
```bash
vim test.s

#write this instruction to the test.s and save
vsetvli  a0, a1, e64, mf8, tu, ma

as -c ./test.s -march=rv64gcv -o test.o

objdump -d ./test.o | less
```
and I got following correct output:
[image]

Do you know what is the issue about it? Or do you have a link refer to this fedora issue? Thank you.

TommyMurphyTM1234, 2024-01-15:
You don't explain in your original post how exactly you're compiling your code and invoking `objdump` - i.e. the commands/command line options used in both cases. A small self-contained reproducible test case (code) would also help. Please use logs rather than screenshots as not everybody can easily read the latter and the former are more easily searched/processed.

LWenH, 2024-01-15:
> You don't explain in your original post how exactly you're compiling your code and invoking `objdump` - i.e. the commands/command line options used in both cases. A small self-contained reproducible [...]

Your suggestion is right, as this is an screenshot I just cropped from a benchmark, so it is not easy to reproduce or read :-)

Here is a reproducible example to reflect this issue.  Thank you for your patience in reading this far.

Suppose there is a simple rvv intrinsic program [rvv_saxpy.c](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/main/examples/rvv_saxpy.c), and I use clang/llvm to compile it:
```sh
clang ./rvv_saxpy.c -O3 -march=rv64gcv -o main
```
and I tried to use the Binutils objdump to disassemble the elf file:
```sh
objdump -d ./main | less
```
I got the incorrect disassemble output like following:
```asm
11154:       006707b3                add     a5,a4,t1
11158:       0287e507                .insn   4, 0x0287e507
1115c:       0286e587                .insn   4, 0x0286e587
11160:       006687b3                add     a5,a3,t1
11164:       0287e607                .insn   4, 0x0287e607
11168:       b29415d7                .insn   4, 0xb29415d7
1116c:       b2a41657                .insn   4, 0xb2a41657
11170:       028685a7                .insn   4, 0x028685a7
11174:       02878627                .insn   4, 0x02878627
```

However, when I use the riscv64-linux-gnu-objdump to disassemble the same elf:
```sh
riscv64-linux-gnu-objdump -d ./main | less
```
I got the correct output:
```asm
11154:       006707b3                add     a5,a4,t1
11158:       0287e507                vl1re32.v       v10,(a5)
1115c:       0286e587                vl1re32.v       v11,(a3)
11160:       006687b3                add     a5,a3,t1
11164:       0287e607                vl1re32.v       v12,(a5)
11168:       b29415d7                vfmacc.vv       v11,v8,v9
1116c:       b2a41657                vfmacc.vv       v12,v8,v10
11170:       028685a7                vs1r.v  v11,(a3)
11174:       02878627                vs1r.v  v12,(a5)
```

LWenH, 2024-01-15:
> It's better to use -xD instead of -d. I never use -d.

Thank you folks, I also tried the -xD option, but the issue still exists.

zhongjuzhe, 2024-01-15:
I just tried. It works well.

0000000000010398 <saxpy_golden>:
   10398:       cd39                    beqz    a0,103f6 <saxpy_golden+0x5e>
   1039a:       00458793                addi    a5,a1,4
   1039e:       c2202773                csrr    a4,vlenb
   103a2:       40f607b3                sub     a5,a2,a5
   103a6:       1761                    addi    a4,a4,-8
   103a8:       02f77963                bgeu    a4,a5,103da <saxpy_golden+0x42>
   103ac:       8732                    mv      a4,a2
   103ae:       0d0077d7                vsetvli a5,zero,e32,m1,ta,ma
   103b2:       5e0551d7                vfmv.v.f        v3,fa0
   103b6:       0d0577d7                vsetvli a5,a0,e32,m1,ta,ma
   103ba:       00279693                slli    a3,a5,0x2
   103be:       0205e087                vle32.v v1,(a1)
   103c2:       02066107                vle32.v v2,(a2)
   103c6:       8d1d                    sub     a0,a0,a5
   103c8:       a22190d7                vfmadd.vv       v1,v3,v2
   103cc:       020760a7                vse32.v v1,(a4)
   103d0:       95b6                    add     a1,a1,a3
   103d2:       9636                    add     a2,a2,a3
   103d4:       9736                    add     a4,a4,a3
   103d6:       f165                    bnez    a0,103b6 <saxpy_golden+0x1e>
   103d8:       8082                    ret
   103da:       050a                    slli    a0,a0,0x2
   103dc:       9532                    add     a0,a0,a2
   103de:       0005a787                flw     fa5,0(a1)
   103e2:       00062707                flw     fa4,0(a2)
   103e6:       0611                    addi    a2,a2,4
   103e8:       0591                    addi    a1,a1,4
   103ea:       70f577c3                fmadd.s fa5,fa0,fa5,fa4
   103ee:       fef62e27                fsw     fa5,-4(a2)
   103f2:       fea616e3                bne     a2,a0,103de <saxpy_golden+0x46>
   103f6:       8082                    ret

00000000000103f8 <saxpy_vec>:
   103f8:       c10d                    beqz    a0,1041a <saxpy_vec+0x22>
   103fa:       0d3577d7                vsetvli a5,a0,e32,m8,ta,ma
   103fe:       0205e807                vle32.v v16,(a1)
   10402:       02066407                vle32.v v8,(a2)
   10406:       00279713                slli    a4,a5,0x2
   1040a:       b3055457                vfmacc.vf       v8,fa0,v16
   1040e:       8d1d                    sub     a0,a0,a5
   10410:       02066427                vse32.v v8,(a2)
   10414:       95ba                    add     a1,a1,a4
   10416:       963a                    add     a2,a2,a4
   10418:       f16d                    bnez    a0,103fa <saxpy_vec+0x2>
   1041a:       8082                    ret

riscv64-unknown-elf-gcc -O3 auto.c
riscv64-unknown-elf-objdump a.out -d | less

I guess it's operation system issue. Seems that objdump is not pointed to correct source.

LWenH, 2024-01-15:
I think a solution might be create a soft link to make objdump to refer the riscv64-unknown-elf-objdump :-)

LWenH, 2024-01-15:
I fixed this issues by making a soft link objdump to the riscv64-linux-gnu-objdump binary, and place it to
the /usr/bin:
```bash
sudo cp /usr/bin/objdump /usr/bin/objdump_bak
sudo ln -s /usr/bin/riscv64-linux-gnu-objdump /usr/bin/objdump
```
Now, Linux perf can correctly disassemble the elf either. T

Thank you for you guys helping with me. I will close this issue.

closed 2024-01-15 by LWenH

TommyMurphyTM1234, 2024-01-15:
> It's better to use -xD instead of -d. I never use -d.

Why? `-x` is perfectly fine in my experience,