riscv-collab/riscv-gnu-toolchain #1409: Rebuilding Compiler

rtorre32, 2024-01-16:
If I want to rebuild the compiler for 32 bit can I just run the following command

./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
make linux

Or do I have to recreate the directory where I want it? (Where I currently have the 64 bit compiler)

Also, how can I configure the starting address of the stack?

TommyMurphyTM1234, 2024-01-17:
> If I want to rebuild the compiler for 32 bit can I just run the following command
> 
> ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d make linux
> 
> Or do I have to recreate the directory where I want it? (Where I currently have the 64 bit compiler)

You mean with both builds having the same install directory (`--prefix=/opt/riscv`)? I'm not sure if mixing the two builds like that is safe. You might want to do them separately first and then comparing the two install directories to check if there are any file clashes.

Why don't you just use `--enable-multilib` to build a single multilib enabled toolchain?

> Also, how can I configure the starting address of the stack?

Using a custom linker script.

rtorre32, 2024-01-17:
> Using a custom linker script.

Is there a linker script in here that I can reference? If not, in what directory should it go and would I have to rebuild the compiler once I create the linker script?

TommyMurphyTM1234, 2024-01-17:
> Is there a linker script in here that I can reference? If not, in what directory should it go

There are lots of examples available online.
E.g.

* https://five-embeddev.com/quickref/stack.html
* https://www.shincbm.com/embedded/2021/05/03/riscv-and-modern-c++-part1-3.html

You should look at other examples and read the `ld` documentation:

* https://ftp.gnu.org/old-gnu/Manuals/ld-2.9.1/html_mono/ld.html#SEC6
* https://mcyoung.xyz/2021/06/01/linker-script/

> and would I have to rebuild the compiler once I create the linker script?

No. The linker script for your target is passed to the cross compiler/linker when creating your RISC-V program for your target.

TommyMurphyTM1234, 2024-01-17:
Closing this as it doesn't identify any issue/bug with the `riscv-gnu-toolchain`. Please use other forums such as the mailing lists, StackOverflow etc. for general support questions.

closed 2024-01-17 by TommyMurphyTM1234