riscv-collab/riscv-gnu-toolchain #1411: make musl fails to build

Lovett1991, 2024-01-21:
## System info
`Linux Lovetts-MacBookPro 6.2.0-39-generic #40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

## Issue

Clean clone,
Installed dependencies listed in readme,
then ran
```
./configure --prefix=/opt/riscv
...
alex@Lovetts-MacBookPro:~/workspace/riscv-gnu-toolchain$ make musl
...
cc1: error: no include path in which to search for stdc-predef.h
...
The directory that should contain system headers does not exist:
  /opt/riscv/sysroot/usr/include
make[3]: *** [Makefile:3305: stmp-fixinc] Error 1
make[3]: Leaving directory '/home/alex/workspace/riscv-gnu-toolchain/build-gcc-musl-stage2/gcc'
make[2]: *** [Makefile:4621: all-gcc] Error 2
make[2]: Leaving directory '/home/alex/workspace/riscv-gnu-toolchain/build-gcc-musl-stage2'
make[1]: *** [Makefile:1039: all] Error 2
make[1]: Leaving directory '/home/alex/workspace/riscv-gnu-toolchain/build-gcc-musl-stage2'
make: *** [Makefile:844: stamps/build-gcc-musl-stage2] Error 2
```

`stdc-prefef.h` does appear to be present
```
alex@Lovetts-MacBookPro:~/workspace/riscv-gnu-toolchain$ find ./ -name 'stdc-predef.h'
./glibc/include/stdc-predef.h
./musl/include/stdc-predef.h
```

Have I missed a step here or is this a genuine problem?

TommyMurphyTM1234, 2024-01-21:
If you do a `make distclean` (and maybe also `rm -rf /opt/riscv`) before `make musl` does it make any difference?

Lovett1991, 2024-01-22:
That did it, thanks

closed 2024-01-22 by Lovett1991