riscv-collab/riscv-gnu-toolchain #1412: while debuggung with gdb, custom instructions not showing

RohanAhmed1, 2024-01-21:
I added the custom instructions in toolchain. The toolchain works perfectly and my new instructions in also showing in dump file(disasmebly) with their proper naming convention but when I use gdb to debug, the hex is showing instead of the custom instruction name

TommyMurphyTM1234, 2024-01-21:
You'll have to provide some detail about how exactly you modified the toolchain to add support for your custom instructions.

RohanAhmed1, 2024-01-21:
Adding custom instructions compilation support, to RISCV ... https://medium.com/@viveksgt/adding-custom-instructions-compilation-support-to-riscv-toolchain-78ce1b6efcf4

I followed these steps

TommyMurphyTM1234, 2024-01-21:
> Adding custom instructions compilation support, to RISCV ... https://medium.com/@viveksgt/adding-custom-instructions-compilation-support-to-riscv-toolchain-78ce1b6efcf4
> 
> I followed these steps

Did you do [this](https://medium.com/@viveksgt/adding-custom-instructions-compilation-support-to-riscv-toolchain-78ce1b6efcf4#:~:text=optinally%20same%20changes%20can%20be%20done%20in%0Ariscv%2Dgnu%2Dtoolchain/riscv%2Dgdb/opcodes/riscv%2Dopc.c%0Ariscv%2Dgnu%2Dtoolchain/riscv%2Dgdb/include/opcode/riscv%2Dopc.h)?

> optinally same changes can be done in
> `riscv-gnu-toolchain/riscv-gdb/opcodes/riscv-opc.c`
> `riscv-gnu-toolchain/riscv-gdb/include/opcode/riscv-opc.h`

TommyMurphyTM1234, 2024-01-21:
This may also be relevant:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837#issuecomment-772385619

kito-cheng, 2024-01-23:
Just remind that: gdb and binutils are shared same code base, but they are different folder in riscv-gnu-toolchain, so you need to apply those change for gdb folder as well, generally should be same change, but may facing some minor merge conflict due to the base version mismatch, because binutils and gdb had different release schedule.

TommyMurphyTM1234, 2024-01-23:
Thanks @kito-cheng - I always wondered why there were two different copies of `binutils-gdb` with different branches/tags pulled in by `riscv-gnu-toolchain`.

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/8c969a9efe68a811cf524174d25255632029f3d3/.gitmodules#L1-L4
* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/8c969a9efe68a811cf524174d25255632029f3d3/.gitmodules#L20-L23
* `binutils @ 49f6117`
* `gdb @ 662243d`

TommyMurphyTM1234, 2024-01-28:
No update from @RohanAhmed1 for over a week so I am assuming that the issue has been explained by the need to make the relevant changes in the two copies of the `binutils-gdb` sources and, as such, am closing the issue. If it's still a problem then reopen it with further info.

closed 2024-01-28 by TommyMurphyTM1234

RohanAhmed1, 2024-02-14:
Hi, apologies from my side for being late of trying suggestion of **adding custom instructions in the gdb folder as well**.

Let me give you all of what I have done and what issue I am facing.

### Procedure and Explanation
I am adding custom instruction(**vgmul.vv**) in the toolchain in following path.

_At gdb/opcodes/riscv-opc.c and riscv-binutils/opcodes/riscv-opc.c_
{"vgmul.vv", 0, INSN_CLASS_V, "Vd,Vt", MATCH_VGMULVV, MASK_VGMULVV, match_opcode, 0 },

_At riscv-binutils/include/opcode/riscv-opc.h and gdb/include/opcode/riscv-opc.h_
#define MATCH_VGMULVV    0xA208A077
2586 #define MASK_VGMULVV     0xFE0FF07F

Now, I am test this instrucion by making a simple test.
...
      li x11, 0x100000000
 29      li x12, 0x100000040
 30      li x10, 0x100000120
 31      li x9,  0x100000080
 32
 33      vle32.v  v2, (x11)
 34      vle32.v  v4, (x12)
 36      vgmul.vv v2, v4
 37
 38      vse32.v  v2, (x10)
 39      # test ends here
 
Now, I build this test with the help of obj_folder of updated toolchain with;
> riscv64-unknown-elf-gcc -march=rv64gv -O -g -mcmodel=medany -gdwarf-2 -c -o test.out
 
 and create disassmebly file with
 > riscv64-unknown-elf-objdump -D test.o > test.dump
 
 This dissassembly file has the vgmul.vv instruction.
 ...
 80000002c:   12050513            addi    a0,a0,288
 20    800000030:   0010049b            addiw   s1,zero,1
 21    800000034:   02049493            slli    s1,s1,0x20
 22    800000038:   08048493            addi    s1,s1,128
 23    80000003c:   0205e107            vle32.v v2,(a1)
 24    800000040:   02066207            vle32.v v4,(a2)
 25    800000044:   a248a177            vgmul.vv    v2,v4
 26    800000048:   02056127            vse32.v v2,(a0)
...

It means that, my instruction is identifed by the toolchain and my test is running as well.

### Issue
Now, I am debugging this with **gdb** of updated toolchain by:
> riscv64-unknown-elf-gdb test.out

and running this command to open the debugger window of dissambly by
> tui reg vector
> layout next

[image]

As you see in this, my custom instruction is showing with **.4byte** instead of **vgmul.vv**

Am I missing some files to update?

TommyMurphyTM1234, 2024-02-14:
> I am adding custom instruction(vgmul.vv) in the toolchain in following path.
> ...

After making the relevant code changes did you make sure to do a full/clean build?
E.g.
```
make distclean
rm -rf <toolchain-prefix-dir>
make
```
BTW - your screenshot is really difficult to read - please use logs rather than screenshots for ease of readability and analysis.

RohanAhmed1, 2024-02-15:
Yeah, I have followed this command to build except **make distclean** and use one more command. Let me write my steps to build. 

> make clean
> rm -rf <toolchain-prefix-dir>
> ./configure -with-gcc-src=<path-to-gcc> --prefix=<toolchain-prefix-dir> --with-cmodel=medany
> make.

_-with-gcc-src=<path-to-gcc>_ is used to save cloning time. Actually, It is a copy of a clone of gcc 13.0. Instead of using riscv RNU toolchain / gcc which I think was gcc 12.3 or something. 

I will try this command as well and let you know. Thanks

TommyMurphyTM1234, 2024-02-15:
> Yeah, I have followed this command to build except **make distclean** and use one more command. Let me write my steps to build.
> ```
> make clean
> rm -rf [toolchain-prefix-dir]
> ./configure -with-gcc-src=[path-to-gcc] --prefix=[toolchain-prefix-dir] --with-cmodel=medany
> make
> ```
I don't think that `make clean` is sufficient to do a clean build. 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1196

> _-with-gcc-src=_ is used to save cloning time. 

Cloning only happens the first time you do the following:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure ...
make
```
It doesn't happen thereafter.

> Actually, It is a copy of a clone of gcc 13.0. Instead of using riscv RNU toolchain / gcc which I think was gcc 12.3 or something.

I don't know what "riscv RNU toolchain" is.
If you mean this repo - `riscv-gnu-toolchain` - then it has been GCC 13.2.0 for a while now:

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/59ab58e8a4aed4ed8f711ebab307757a5ebaa1f5/.gitmodules#L5-L8
* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1341

RohanAhmed1, 2024-02-16:
Thanks you for your response. Yeah I mean riscv-gnu-toolchain repo. 

Well I tried it out with these steps now,
> make distclean
> rm -rf [toolchain-prefix-dir]
> ./configure -with-gcc-src=[path-to-gcc] --prefix=[toolchain-prefix-dir] --with-cmodel=medany
> make

The issue is still remain. What can be other thing I can try? can you suggest some?

TommyMurphyTM1234, 2024-02-16:
> The issue is still remain. What can be other thing I can try? can you suggest some?

Maybe ask about the issue on the GDB mailing list since this is not really a RISC-V specific issue?

* https://www.sourceware.org/gdb/mailing-lists/